mgonzs13/yolo_ros #1: ModuleNotFoundError: No module named 'ultralytics.tracker'

hcdiekmann, 2023-03-28:
Launching with 
```bash
ros2 launch yolov8_bringup yolov8.launch.py 
```
gives me an import error:
```bash
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [yolov8_node-1]: process started with pid [24716]
[yolov8_node-1] Traceback (most recent call last):
[yolov8_node-1]   File "/home/aaeon/interbotix_ws/install/yolov8_ros/lib/yolov8_ros/yolov8_node", line 11, in <module>
[yolov8_node-1]     load_entry_point('yolov8-ros==0.0.0', 'console_scripts', 'yolov8_node')()
[yolov8_node-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 490, in load_entry_point
[yolov8_node-1]     return get_distribution(dist).load_entry_point(group, name)
[yolov8_node-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
[yolov8_node-1]     return ep.load()
[yolov8_node-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2445, in load
[yolov8_node-1]     return self.resolve()
[yolov8_node-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2451, in resolve
[yolov8_node-1]     module = __import__(self.module_name, fromlist=['__name__'], level=0)
[yolov8_node-1]   File "/home/aaeon/interbotix_ws/install/yolov8_ros/lib/python3.8/site-packages/yolov8_ros/yolov8_node.py", line 13, in <module>
[yolov8_node-1]     from ultralytics.tracker.trackers.basetrack import BaseTrack
[yolov8_node-1] ModuleNotFoundError: No module named 'ultralytics.tracker'
[ERROR] [yolov8_node-1]: process has died [pid 24716, exit code 1, cmd '/home/aaeon/interbotix_ws/install/yolov8_ros/lib/yolov8_ros/yolov8_node --ros-args -r __node:=yolov8_node -r __ns:=/yolo --params-file /tmp/launch_params_mijvcv3d -r image_raw:=/camera/rgb/image_raw'].
```

Any help would be appreciated.

mgonzs13, 2023-03-28:
Hi @hcdiekmann, which version of ultralytics are you using? 
Last version, 8.0.58, should work.

mgonzs13, 2023-03-29:
Launching with ```ros2 launch yolov8_bringup yolov8.launch.py device:=cpu``` should use CPU instead of GPU.

hcdiekmann, 2023-03-29:
Thank you. This works.

closed 2023-03-29 by hcdiekmann