mgonzs13/yolo_ros #2: AttributeError: type object 'SetBool' has no attribute 'Request'

JINtaeung, 2023-06-27:
```
~/v8_ros2/src/yolov8_ros$  ros2 launch yolov8_bringup yolov8.launch.py device:=cpu
[INFO] [launch]: All log files can be found below /home/jtw/.ros/log/2023-06-27-09-51-51-294832-home-5348
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [yolov8_node-1]: process started with pid [5350]
[yolov8_node-1] Traceback (most recent call last):
[yolov8_node-1]   File "/home/jtw/v8_ros2/install/yolov8_ros/lib/yolov8_ros/yolov8_node", line 33, in <module>
[yolov8_node-1]     sys.exit(load_entry_point('yolov8-ros==0.0.0', 'console_scripts', 'yolov8_node')())
[yolov8_node-1]   File "/home/jtw/v8_ros2/install/yolov8_ros/lib/yolov8_ros/yolov8_node", line 25, in importlib_load_entry_point
[yolov8_node-1]     return next(matches).load()
[yolov8_node-1]   File "/usr/lib/python3.8/importlib/metadata.py", line 77, in load
[yolov8_node-1]     module = import_module(match.group('module'))
[yolov8_node-1]   File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
[yolov8_node-1]     return _bootstrap._gcd_import(name[level:], package, level)
[yolov8_node-1]   File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
[yolov8_node-1]   File "<frozen importlib._bootstrap>", line 991, in _find_and_load
[yolov8_node-1]   File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
[yolov8_node-1]   File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
[yolov8_node-1]   File "<frozen importlib._bootstrap_external>", line 848, in exec_module
[yolov8_node-1]   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
[yolov8_node-1]   File "/home/jtw/v8_ros2/install/yolov8_ros/lib/python3.8/site-packages/yolov8_ros/yolov8_node.py", line 40, in <module>
[yolov8_node-1]     class Yolov8Node(Node):
[yolov8_node-1]   File "/home/jtw/v8_ros2/install/yolov8_ros/lib/python3.8/site-packages/yolov8_ros/yolov8_node.py", line 98, in Yolov8Node
[yolov8_node-1]     req: SetBool.Request,
[yolov8_node-1] AttributeError: type object 'SetBool' has no attribute 'Request'
[ERROR] [yolov8_node-1]: process has died [pid 5350, exit code 1, cmd '/home/jtw/v8_ros2/install/yolov8_ros/lib/yolov8_ros/yolov8_node --ros-args -r __node:=yolov8_node -r __ns:=/yolo --params-file /tmp/launch_params_rqr5qbe3 -r image_raw:=/camera/rgb/image_raw'].
```

**what I should do to resolve the error**

mgonzs13, 2023-06-27:
Hi @JINtaeung, I'm not sure why you are getting that error. Which ROS distro do you use? Do you have any ROS 1 distro installed?

JINtaeung, 2023-06-28:
i'm using ROS foxy

closed 2023-07-08 by mgonzs13