mgonzs13/yolo_ros #3: how to set the live stream for detection?

flydragon2018, 2023-07-04:
I have started the camera and publish image, the default 
value 'input_image_topic' in launch is changed to  correct path.

however,  no message published in  /yolo/detections ?

flydragon2018, 2023-07-04:
now I input parameters in command line,  new error message:

[yolov8_node-1]     detection.bbox.center.position.x = float(box[0])
[yolov8_node-1] AttributeError: 'Pose2D' object has no attribute 'position'
[ERROR] [yolov8_node-1]: process has died [pid 11887, exit code 1, cmd '/home/zzl/ros2_ws/install/yolov8_ros/lib/yolov8_ros/yolov8_node --ros-args -r __node:=yolov8_node -r __ns:=/yolo --params-file /tmp/launch_params_h0adb14y -r image_raw:=/camera/color/image_raw'].

flydragon2018, 2023-07-04:
fixed one error, new comes!   
version problem?

AttributeError: 'Detection2D' object has no attribute 'id'

flydragon2018, 2023-07-04:
I fixed by changing a few places. 

from vision_msgs.msg import Detection2D related problem.

mgonzs13, 2023-07-04:
Wich ROS 2 distro are you using? This is updated for Humble.

flydragon2018, 2023-07-05:
I  am using Foxy.

mgonzs13, 2023-07-05:
That's the problem. vision_msgs has changed from Foxy to Humble.
For a quick fix you can clone the vision_msgs repo using the ros2 branch.

closed 2023-07-11 by mgonzs13