mgonzs13/yolo_ros #4: How do you include ultralytics library in ROS2?

Peanpepu, 2023-07-28:
I'm trying more or less to replicate your node but adding some other things and I have problems including the ultralytics library. Could you tell me how you did it?

mgonzs13, 2023-07-29:
Hi @Peanpepu, have you installed ultralytics with pip and imported it in the Python file?
Btw, why are you replicating my node instead of forking the repo and editing the code?

Peanpepu, 2023-07-29:
No, I'm just downloading the ultralytics repository with git and trying to incluye It in ros but I Will try as you say thank you. And I'm not using your repository directly because I found It after I has made It by myself. If I keep having problems I Will also try downloading yours and change It.
Thank you very mucho for the help.

closed 2023-07-29 by mgonzs13