mgonzs13/yolo_ros #9: [debug_node-3] AttributeError: module 'cv2' has no attribute 'Mat'

mzahana, 2023-08-25:
Hi @mgonzs13 
Thanks for the great repo.

I am using your latest main branch inside a docker container on Jetson Orin Nx with Ros2 container.

My Opncv version is `4.5.0`

The `debug_node` throws the following error.

`[debug_node-3] AttributeError: module 'cv2' has no attribute 'Mat'`

Any idea what could be the issue ?

mgonzs13, 2023-08-25:
Hi @mzahana, I am using OpenCV 4.5.4 and opencv-python 4.8.0.76. Verify your opencv-python version, upgrading might be necessary.

raniaa03, 2023-08-31:
I am running into the same issue and have opencv-python 4.8.0.76 installed. Was there a fix found?

mgonzs13, 2023-09-01:
I have replaced cv2.Mat with np.array. This should solve it.

closed 2023-11-16 by mgonzs13