mgonzs13/yolo_ros #10: Issues with launching the launch file

arun-venkat-23, 2023-09-29:
Executed the below command when in the directory where this file is present:
ros2 launch yolov8.launch.py

Error:
[INFO] [launch]: All log files can be found below /home/arunvenkat/.ros/log/2023-09-28-23-26-26-077037
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=PackageNotFoundError("package 'yolov8_ros' not found, searching: ['/opt/ros/foxy']")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/ament_index_python/packages.py", line 50, in get_package_prefix
    content, package_prefix = get_resource('packages', package_name)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ament_index_python/resources.py", line 48, in get_resource
    raise LookupError(
LookupError: Could not find the resource 'yolov8_ros' of type 'packages'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 453, in execute
    ret = super().execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 823, in execute
    self.__expand_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 668, in __expand_substitutions
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 668, in <listcomp>
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_context.py", line 232, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/substitutions/executable_in_package.py", line 76, in perform
    package_prefix = super().perform(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/substitutions/find_package.py", line 79, in perform
    result = self.find(package)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/substitutions/find_package.py", line 96, in find
    return get_package_prefix(package_name)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ament_index_python/packages.py", line 52, in get_package_prefix
    raise PackageNotFoundError(
ament_index_python.packages.PackageNotFoundError: "package 'yolov8_ros' not found, searching: ['/opt/ros/foxy']"

Could you please help with this? At the same time when using the "ros2 launch yolov8_bringup yolov8.launch.py" statement did not produce any output except for an error (Package 'yolov8_bringup' not found: "package 'yolov8_bringup' not found, searching: ['/opt/ros/foxy']")

mgonzs13, 2023-09-29:
Hi @arun-venkat-23, it seems you did not use colcon build followed by source ~/ros2_ws/install/setup.bash.

arun-venkat-23, 2023-09-29:
I'm experiencing the same issue even after building the workspace and sourcing it.

mgonzs13, 2023-09-29:
Which message are you getting when you source your workspace and launch the file in the same terminal? I the case of your previous message ```Package 'yolov8_bringup' not found: "package 'yolov8_bringup' not found, searching: ['/opt/ros/foxy']"```, I think you are not sourcing the workspace in the terminal where you try to run yolov8.launch.py.

closed 2023-11-16 by mgonzs13