mgonzs13/yolo_ros #15: 请问用什么指令能够运行track，运行的效果是什么样的？

zhang320, 2023-12-04:


KXRLYM, 2023-12-05:
Not the author but I am having the same problem as described in the title (lol)

- ROS2 Humble running on Ubuntu 22.04, using Azure Kinect
- nothing is being published to yolo/dbg_image or any topic really
- launching '**ros2 launch yolov8_bringup yolov8.launch.py device:=cpu input_image_topic:=/rgb/image_formatted**' and everything seems fine with one and only output '[yolov8_node-1] YOLOv8m summary (fused): 218 layers, 25886080 parameters, 0 gradients, 78.9 GFLOPs'
- /rbg/image_formatted seems normal, I can see the camera image on RVIZ2 image viewer

mgonzs13, 2023-12-05:
Hi @zhang320 and @KXRLYM, please, check if yolov8_node is subscribed to your topic and if the QoS is the same for both, yolov8_node and your publisher.

zhang320, 2023-12-05:
Hello, I am using the v4l2 that comes with ROS2_ Camera sent an image_ Raw message, but I haven't received any feedback. How can I obtain tracking results?

mgonzs13, 2023-12-05:
Tracking results are published in the /yolo/tracking topic. If you don't get any results, it may be due to the QoS of your image topic.  This package uses Best Effort for the image topics, the sensor QoS.

KXRLYM, 2023-12-05:
I have made the QoS consistent between my publisher and YOLO subscriber but still doesn't work :( 
Added some screenshots:

**No image**
[image]
I have both my publisher and YOLO sub open

**rqt_graph** yolo nodes subscribing to my node 
[image]

Publisher with 
~~~
self.publisher = self.create_publisher(Image, '/rgb/image_formatted', qos_profile_sensor_data)
~~~

zhang320, 2023-12-06:
Thank you for your response. I have got the result.

mgonzs13, 2023-12-06:
Yes, that should launch the normal version but in the rqt_graph image, you have the 3D node. In that case, no db images will be published till 3D data is obtained.

closed 2023-12-07 by mgonzs13

alt1-m8, 2023-12-11:
Hi,

I have a similar problem. I have the input_image_topic remapped to /camera/color/image_raw. Ros topic list shows 1 publisher and 3 subscribers. I have attempted to visualise with rviz2. The /yolo/dbg_image streams fine, but with no detections. Changing to /yolo/detections shows a 'No Image' error.

I'm assuming that upon launching within terminal with 'ros2 launch yolov8_bringup yolov8.launch.py model:=best4.pt input_image_topic:=/camera/color/image_raw' the stream should start up by itself with the bounding box detections, but I only see 'Model summary (fused): 168 layers , 1125971 parameters, 0 gradients, 28.4 GFLOPS'

Device: Nvidia Jetson Orin Nano
Camera: Realsense D455
ROS-distro: Foxy
model: custom trained (best.pt)
input_image_topic: /camera/color/image_raw

How can I get the image?

Thanks