mgonzs13/yolo_ros #17: How can I get image? part2

alt1-m8, 2023-12-11:
Hi,

I have a similar problem. I have the input_image_topic remapped to /camera/color/image_raw. Ros topic list shows 1 publisher and 3 subscribers. I have attempted to visualise with rviz2. The /yolo/dbg_image streams fine, but with no detections. Changing to /yolo/detections shows a 'No Image' error.

I'm assuming that upon launching within terminal with 'ros2 launch yolov8_bringup yolov8.launch.py model:=best4.pt input_image_topic:=/camera/color/image_raw' the stream should start up by itself with the bounding box detections, but I only see 'Model summary (fused): 168 layers , 1125971 parameters, 0 gradients, 28.4 GFLOPS'

Device: Nvidia Jetson Orin Nano
Camera: Realsense D455
ROS-distro: Foxy
model: custom trained (best.pt)
input_image_topic: /camera/color/image_raw

How can I get the image?

Thanks

mgonzs13, 2023-12-11:
Hi @alt1-m8, have you checked the QoS of your topics? This package uses the default QoS profile for sensor data.

alt1-m8, 2023-12-12:
Hi @mgonzs13 thanks for the quick reply. I have checked the Qos topics (see attached picture), 
[image]

This image shows the topic info for the image_raw topic published by the real sense node. It's 'RELIABLE'. If I understand correctly, 'BEST_EFFORT' is required in its place. Is there a way to specify this in the cli during launch of the realsense, or is this a setting that must be applied in the launch file, or within the ROS settings?

mgonzs13, 2023-12-12:
I have added some QoS parameters to the nodes and launch files to configure the reliability of the topics. So now you can change the QoS of the nodes of this package instead of the QoS of your camera. Take a look at the README. You can change the reliability by adding ```image_reliability:=1```, ```depth_image_reliability:=1``` and ```depth_info_reliability:=1``` when you run the launch ````0=system default, 1=Reliable, 2=Best Effort)```.

alt1-m8, 2023-12-13:
Thanks for the quick update. I have launched yolov8-ros with the new changes and the reliability adjustments on cli. The ros2 topic info --verbose shows yolov8-ros and the realsense as RELIABLE. However, I still haven't gotten a stream. Just to check, upon launching the yolov8-ros, should the stream with the detections show up as a pop-up window, or must RVIZ2 be launched separately? I have used RVIZ to visualise all the topics, and all streams show as no image with the exception of image_raw.

mgonzs13, 2023-12-13:
Yes, you have to open rviz2 to visualize the ```/yolo/dbg_image```. You can also echo the topics ```/yolo/detections``` and ```/yolo/tracking```.

mgonzs13, 2023-12-13:
@Unknown9190, in the examples of the README, I am using an RGB-D camera, that publishes color and depth images and the point cloud. The ```yolov8.launch.py``` launch only runs the nodes to apply YOLOv8, the tracking and the debug. For the 3D option, you have to use ```yolov8_3d.launch.py```, which runs the 3D node subscribing to the depth image and camera info. 

If you want depth images from monocular images, you can check solutions as the ones presented in [Papers With Code](https://paperswithcode.com/task/monocular-depth-estimation).

Unknown9190, 2023-12-15:
@mgonzs13 Understood, I'll try to implement it into my ros2 project, thanks for the help!

mgonzs13, 2024-02-28:
Hey @Unknown9190, how is this going?

Redhorde, 2024-09-03:
I had similar issue and in my case it turned out my gpu wasn't available, but the error message took a very long to show up for whatever reason. Luckily one time I left it running for more than 10 minutes and it finally did. No idea why it took so long to notify me, it starts working almost immediately when I switched the device to cpu.

Redhorde, 2024-12-20:
@besma14 Is there no image at all published on the topic or are there no detections marked on it? If it's the latter try 
```shell
ros2 topic echo /yolo/detections
``` 
to see if your model is even detecting anything, it should return some coordinates

besma14, 2024-12-20:
Using the command: ros2 topic echo /yolo/detections … I don’t have any output for the topic /yolo/detections, and no detections are marked on my image. However, I can see the image in RQT for the topic /yolo/dbg_image, but no detections are marked on it.

Redhorde, 2024-12-20:
I'd check if the path to model is correct because the node clearly started correctly and it has access to the image topic if it sends it to dbg_image. That leaves only a problem with yolo detection or the model you used