mgonzs13/yolo_ros #19: ZeroDivision error

GeraltRivskiy, 2023-12-29:
Hi, 
I'm trying to use Webots camera with ROS2 and yolov8.
I created node as a Webots controller, taking image and publish it in camera/rgb/image
`img = Image()
        img.height = self.__cam.getHeight()
        img.width = self.__cam.getWidth()
        img.encoding = 'bgr8'
        img.data = self.__cam.getImage()
        # print()
        self.image_pub.publish(img)`
But when I launch yolov8 node it gives me zero division error. When I commented publishing line (so camera/rgb/image_raw is empty), I have no errors so its probably issues with image I take, but I dont get it

`[yolov8_node-1]   File "/home/geraltrivskiy/.local/lib/python3.10/site-packages/ultralytics/data/augment.py", line 668, in __call__
[yolov8_node-1]     r = min(new_shape[0] / shape[0], new_shape[1] / shape[1])
[yolov8_node-1] ZeroDivisionError: division by zero
`

mgonzs13, 2023-12-29:
That's a bit strange. Can you check the image messages published in the image topic with ```ros2 echo```?

GeraltRivskiy, 2023-12-29:
It looks like this:

header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: ''
height: 320
width: 320
encoding: bgr8
is_bigendian: 0
step: 0
data:
- 103
- 120
- 140
- 255
- 103
- 120
- 140
- 255
- 103
- 120
- 139
- 255
- 104
- 120
- 139
- 255
- 104
- 120
- 140
- 255
- 103
- 119
- 139
- 255
- 103
- 120
- 139
- 255
- 102
- 119
- 139
- 255
- 103
- 119
- 140
- 255
- 103
- 119
- 140
- 255
- 102
- 119
- 139
- 255
- 100
- 118
- 138
- 255
- 101
- 117
- 138
- 255
- 101
- 118
- 138
- 255
- 101
- 117
- 138
- 255
- 100
- 117
- 138
- 255
- 100
- 117
- 138
- 255
- 99
- 116
- 138
- 255
- 99
- 116
- 137
- 255
- 98
- 115
- 137
- 255
- 98
- 115
- 137
- 255
- 99
- 116
- 137
- 255
- 98
- 115
- 137
- 255
- 98
- 116
- 137
- 255
- 98
- 116
- 137
- 255
- 98
- 115
- 137
- 255
- 98
- 115
- 136
- 255
- 97
- 114
- 136
- 255
- 97
- 114
- 136
- 255
- 97
- 114
- 136
- 255
- 97
- 114
- 137
- 255
- 98
- 115
- 137
- 255
- '...'
---

GeraltRivskiy, 2023-12-29:
I edited source code of your package by edit some parameters in model.predict: augment=False and imgsz=(320,320) so it wont transform images, but it still do

mgonzs13, 2023-12-29:
You should fill all the fields of the messages, I mean the frame_id, the stamp and the step.

GeraltRivskiy, 2023-12-29:
Frame_id and stamp is not fillable according to [ros docs](https://github.com/ros2/common_interfaces/blob/humble/sensor_msgs/msg/Image.msg)
I tried to fill step, but I dont actually get what it needs
In docs it described as "Full row length in bytes", and I tried to use len() or __sizeof__() functions on image (that I get by self.__cam.getImage()), but its not working
`[yolov8_node-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/cv_bridge/core.py", line 180, in imgmsg_to_cv2
[yolov8_node-1]     im = np.ndarray(shape=(img_msg.height, int(img_msg.step/dtype.itemsize/n_channels), n_channels),
[yolov8_node-1] TypeError: buffer is too small for requested array
`

GeraltRivskiy, 2023-12-30:
Thanks a lot!
It works fine for me with step, thanks for explanation, I got that stamp and id should be filled via std_msgs/Header, but it worked without it, so I decided to continue my work without it
It has issues with objects I detect, all of them has the same class_name, its strange because I tested model separately and it worked fine. But its not about issue I started, thanks a lot for your help!

closed 2023-12-30 by GeraltRivskiy