mgonzs13/yolo_ros #20: Is it possible to Deploying a Single Instance of yolov8_ros Node for Multiple Camera Detections

Ekanshh, 2023-12-29:
I am currently utilizing the yolov8_ros node for object detections and tracking on camera streams. However, the current setup involves deploying a separate yolov8_ros node for each camera, leading to performance issues. I am interested in exploring the possibility of deploying a single instance of the yolov8_ros node to handle detections on multiple cameras concurrently. 

Here is a brief overview of the current vs desired process:

- Current System: 
[image]

- Desired System:
[image]

mgonzs13, 2023-12-29:
Hi @Ekanshh, what about creating a node that subscribes to multiple camera topics and then publishes those images in the input topic of the yolov8_node?

Ekanshh, 2023-12-29:
Thank you for your quick response. I will give it a try and provide an update here to let you know how it goes.

RP-ND, 2024-02-21:
Did you ever manage to go further with this @Ekanshh?

Ekanshh, 2024-02-21:
> Did you ever manage to go further with this @Ekanshh?

No, I haven’t made any progress yet. I’m still sticking to separate nodes. Essentially, I’ve customized the bringup launch file to accommodate 6 nodes for two cameras, with 2 nodes dedicated to detection, 2 for tracking, and 2 for debugging purposes. However, it’s currently on my to-do list and not of high priority at the moment.

RP-ND, 2024-02-21:
> No, I haven’t made any progress yet. I’m still sticking to separate nodes. Essentially, I’ve customized the bringup launch file to accommodate 6 nodes for two cameras, with 2 nodes dedicated to [...]

Fair, I have an embedded front end with a Nvidia Orin Nano, I may go further with this since running two custom nodes, even if they are 8n, reaches the compute limit of the frontend memory wise.

Ekanshh, 2024-02-27:
> > > Did you ever manage to go further with this @Ekanshh?
> > 
> > 
> > No, I haven’t made any progress yet. I’m still sticking to separate nodes. Essentially, I’ve customized the bringup launch file to accommodate 6 nodes for two cameras, with 2 nodes dedicated to detection, 2 for tracking, and 2 for debugging purposes. However, it’s currently on my to-do list and not of high priority at the moment.
> 
> Fair, I have an embedded front end with an Orin Nano, I may go further with this since running two custom nodes, even if they are 8n, reaches the compute limit of the frontend memory wise.

If you do come up with a working solution or make further progress, let me know. :)

Ekanshh, 2024-03-04:
Closing this issue as I've successfully resolved it for my particular scenario by implementing the suggestion provided by @mgonzs13. My approach involved modifying the yolov8_ros node to subscribe to multiple camera topics synchronously and utilizing YOLO to batch predict on these images. Subsequently, I forwarded the predictions to multiple tracking topics corresponding to the camera topics.

Edwardius, 2025-07-20:
hey @mgonzs13 , would you be open to me helping out with integrating some sort of multi-camera handling for this?

Similar to what @Ekanshh here did with having a synchronization layer followed by a batch prediction process. If you have ideas of your own for how to implement, let me know.

mgonzs13, 2025-07-21:
Hi @Edwardius, do you want to synchronize all images from each camera and then apply YOLO?