mgonzs13/yolo_ros #21: Could not transform: "base_link" passed to lookupTransform argument target_frame does not exist

cncowboy, 2024-01-04:
Every things is ok when I launch with command below:
ros2 launch yolov8_bringup yolov8.launch.py input_image_topic:=/camera/color/image_raw 

But it could not work when I launch with command below:
ros2 launch yolov8_bringup yolov8_3d.launch.py input_image_topic:=/camera/color/image_raw input_depth_topic:=/camera/depth/image_raw

When I view yolo image in rvit ，I got error message:
Could not transform: "base_link" passed to lookupTransform argument target_frame does not exist

mgonzs13, 2024-01-04:
Hey @cncowboy, do you have a base_link frame in your tf tree? If not, you should change the target_frame when you launch the 3D version (```target_frame:=<your frame>```). Btw, you should also set the input_depth_info_topic.

cncowboy, 2024-01-05:
thank you for responsing my issue so quickly，i have found the base_link of camera in TF tree:camera_link，it seems work well,

the input_depth_info_topic's default value is same as  in camera, so it need not to change.

closed 2024-01-05 by cncowboy