mgonzs13/yolo_ros #24: How to create and train the dataset

cds4488, 2024-01-24:
Hello, may I ask how to use this model to train the dataset you have created? What should be the format and production of the RGB dataset and the RGBD dataset? Because what I want to do is to create a dataset of transparent objects in the laboratory. Thank you!

mgonzs13, 2024-01-24:
Hey @cds4488, to use your custom dataset, you have to [train YOLOv8 as normal](https://docs.ultralytics.com/modes/train/#usage-examples). Then, pass the new model path to the launch file using the ```model``` parameter.

For instance, you have the following example. Replace the [coco128.yaml](https://github.com/ultralytics/ultralytics/blob/main/ultralytics/cfg/datasets/coco128.yaml) with you YAML file and choose the base YOLOv8 model.
```python
from ultralytics import YOLO

# Load a model
model = YOLO('yolov8n.yaml')  # build a new model from YAML
model = YOLO('yolov8n.pt')  # load a pretrained model (recommended for training)
model = YOLO('yolov8n.yaml').load('yolov8n.pt')  # build from YAML and transfer weights

# Train the model
results = model.train(data='coco128.yaml', epochs=100, imgsz=640)
```

cds4488, 2024-01-24:
@ mgonzs13 Thanks for your answering, I will try later!

mgonzs13, 2024-02-28:
Hey @cds4488, how is this going?