mgonzs13/yolo_ros #26: base_link" passed to lookupTransform argument target_frame does not exist

RP-ND, 2024-02-13:
Hello there,

Trying to use a DepthAi Luxonis Pro Wide POE camera with a ROS setup. I am getting the disparity frame topic published to /oak/stereo/image_raw, but this error gets created which I do not understand.

> ros2 launch yolov8_bringup yolov8_3d.launch.py input_image_topic:=/oak/rgb/image_raw input_depth_topic:=/oak/stereo/image_raw input_depth_info_topic:=/oak/stereo/camera_info
> [INFO] [launch]: All log files can be found below /home/ndaimm/.ros/log/2024-02-13-18-45-10-639973-ndaimm-desktop-4676
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [yolov8_node-1]: process started with pid [4677]
> [INFO] [tracking_node-2]: process started with pid [4679]
> [INFO] [detect_3d_node-3]: process started with pid [4681]
> [INFO] [debug_node-4]: process started with pid [4683]
> [yolov8_node-1] /home/ndaimm/.local/lib/python3.10/site-packages/matplotlib/projections/__init__.py:63: UserWarning: Unable to import Axes3D. This may be due to multiple versions of Matplotlib being installed (e.g. as a system package and as a pip package). As a result, the 3D projection is not available.
> [yolov8_node-1]   warnings.warn("Unable to import Axes3D. This may be due to multiple versions of "
> [tracking_node-2] /home/ndaimm/.local/lib/python3.10/site-packages/matplotlib/projections/__init__.py:63: UserWarning: Unable to import Axes3D. This may be due to multiple versions of Matplotlib being installed (e.g. as a system package and as a pip package). As a result, the 3D projection is not available.
> [tracking_node-2]   warnings.warn("Unable to import Axes3D. This may be due to multiple versions of "
> [debug_node-4] /home/ndaimm/.local/lib/python3.10/site-packages/matplotlib/projections/__init__.py:63: UserWarning: Unable to import Axes3D. This may be due to multiple versions of Matplotlib being installed (e.g. as a system package and as a pip package). As a result, the 3D projection is not available.
> [debug_node-4]   warnings.warn("Unable to import Axes3D. This may be due to multiple versions of "
> [detect_3d_node-3] [ERROR] [1707867925.587148194] [yolo.detect_3d_node]: Could not transform: "base_link" passed to lookupTransform argument target_frame does not exist. 
> [detect_3d_node-3] [ERROR] [1707867925.805917716] [yolo.detect_3d_node]: Could not transform: "base_link" passed to lookupTransform argument target_frame does not exist. 
> [detect_3d_node-3] [ERROR] [1707867925.877574412] [yolo.detect_3d_node]: Could not transform: "base_link" passed to lookupTransform argument target_frame does not exist. 
> [detect_3d_node-3] [ERROR] [1707867926.300230844] [yolo.detect_3d_node]: Could not transform: "base_link" passed to lookupTransform argument target_frame does not exist. 
> [detect_3d_node-3] [ERROR] [1707867926.489552873] [yolo.detect_3d_node]: Could not transform: "base_link" passed to lookupTransform argument target_frame does not exist. 
> [detect_3d_node-3] [ERROR] [1707867926.595309086] [yolo.detect_3d_node]: Could not transform: "base_link" passed to lookupTransform argument target_frame does not exist. 
> [detect_3d_node-3] [ERROR] [1707867926.655508926] [yolo.detect_3d_node]: Could not transform: "base_link" passed to lookupTransform argument target_frame does not exist. 
> [detect_3d_node-3] [ERROR] [1707867926.697220906] [yolo.detect_3d_node]: Could not transform: "base_link" passed to lookupTransform argument target_frame does not exist. 
> [detect_3d_node-3] [ERROR] [1707867926.741311144] [yolo.detect_3d_node]: Could not transform: "base_link" passed to lookupTransform argument target_frame does not exist. 
> [detect_3d_node-3] [ERROR] [1707867926.820498681] [yolo.detect_3d_node]: Could not transform: "base_link" passed to lookupTransform argument target_frame does not exist. 
> [detect_3d_node-3] [ERROR] [1707867927.058484653] [yolo.detect_3d_node]: Could not transform: "base_link" passed to lookupTransform argument target_frame does not exist. 

I also get this package error, though I am not sure how to solve it:

>  /home/ndaimm/.local/lib/python3.10/site-packages/matplotlib/projections/__init__.py:63: UserWarning: Unable to import Axes3D. This may be due to multiple versions of Matplotlib being installed (e.g. as a system package and as a pip package). As a result, the 3D projection is not available.
> [yolov8_node-1]   warnings.warn("Unable to import Axes3D. This may be due to multiple versions of "
> [tracking_node-2] /home/ndaimm/.local/lib/python3.10/site-packages/matplotlib/projections/__init__.py:63: UserWarning: Unable to import Axes3D. This may be due to multiple versions of Matplotlib being installed (e.g. as a system package and as a pip package). As a result, the 3D projection is not available.
> [tracking_node-2]   warnings.warn("Unable to import Axes3D. This may be due to multiple versions of "
> [debug_node-4] /home/ndaimm/.local/lib/python3.10/site-packages/matplotlib/projections/__init__.py:63: UserWarning: Unable to import Axes3D. This may be due to multiple versions of Matplotlib being installed (e.g. as a system package and as a pip package). As a result, the 3D projection is not available.
> [debug_node-4]   warnings.warn("Unable to import Axes3D. This may be due to multiple versions of "
>  
 
 
Please let me know any steps I could take in order to get this working. Thank you.

mgonzs13, 2024-02-14:
Hi @RP-ND, that error means that the base_link, set in the target_frame param of the launch file, does not exist. You have to change it when you run the launch by ```target_frame:=<you frame>```, for instance with you camera base link.

RP-ND, 2024-02-21:
Thank you! This was exactly it, was confused when I couldn't pass the /tf parameter but then realized I just needed to call the base frame!

closed 2024-02-21 by RP-ND

lymcs, 2024-09-04:
excuse me,,why can not i do this?

lymcs, 2024-09-04:
> Thank you! This was exactly it, was confused when I couldn't pass the /tf parameter but then realized I just needed to call the base frame!

what is base frame????

mgonzs13, 2024-10-01:
It is the base frame in your tf tree, which may be base_frame, base_link, odom, map and so on.

BingPeng0, 2024-10-28:
So I am mounting my camera on a robot arm. In this case, the base frame should be the base of my arm or the base link of the camera

mgonzs13, 2024-10-28:
@BingPeng0 the target_frame is the reference frame to transform the 3D positions of the detected objects. If you use the base of the camera, 3D positions will be computed with the camera base as the reference. If you use the base of the robot, 3D positions will be computed with the robot base as the reference. I recommend you use the robot base.