mgonzs13/yolo_ros #27: detect_3d_node.py: on_detections not being called!

amiryanj, 2024-02-23:
Hello, 

I am debugging the repo on my Jetson Orin (with ROS2 Foxy) and I can't get the 3d detections.
I debugged the nodes and it seems the callback on_detections() is not being called, even though the topics corresponding to all the 3 subscribers are being published (depth_image, depth_info, and detections).
What might be the reasons?

Many thanks.

mgonzs13, 2024-02-23:
Hey @amiryanj, have you checked the QoS? By default, subscribers use best effort. You can also change them using launch params.

amiryanj, 2024-02-23:
Hey @mgonzs13 
Thanks for your quick feedback.
I am kinda new to ROS2. I just checked out the Reliability Policies, they sound to be matching.
There is only one policy that is not matched (the Durability of the publisher on /camera/depth/image_rect_raw.
Not sure if that can be the reason? and how to fix that possibly?

```
jetson@jetson-desktop:~$ ros2 topic info /camera/depth/image_rect_raw --verbose 
Type: sensor_msgs/msg/Image

Publisher count: 1

Node name: camera
Node namespace: /camera
Topic type: sensor_msgs/msg/Image
Endpoint type: PUBLISHER
GID: 01.0f.d9.91.6c.0d.35.52.01.00.00.00.00.00.14.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
  Durability: RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds

Subscription count: 1

Node name: detect_3d_node
Node namespace: /yolo
Topic type: sensor_msgs/msg/Image
Endpoint type: SUBSCRIPTION
GID: 01.0f.d9.91.07.2a.32.b4.01.00.00.00.00.00.14.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
  Durability: RMW_QOS_POLICY_DURABILITY_VOLATILE
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds

jetson@jetson-desktop:~$ ros2 topic info /camera/depth/camera_info --verbose 
Type: sensor_msgs/msg/CameraInfo

Publisher count: 1

Node name: camera
Node namespace: /camera
Topic type: sensor_msgs/msg/CameraInfo
Endpoint type: PUBLISHER
GID: 01.0f.d9.91.6c.0d.35.52.01.00.00.00.00.00.15.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
  Durability: RMW_QOS_POLICY_DURABILITY_VOLATILE
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds

Subscription count: 1

Node name: detect_3d_node
Node namespace: /yolo
Topic type: sensor_msgs/msg/CameraInfo
Endpoint type: SUBSCRIPTION
GID: 01.0f.d9.91.07.2a.32.b4.01.00.00.00.00.00.15.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
  Durability: RMW_QOS_POLICY_DURABILITY_VOLATILE
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds
```

mgonzs13, 2024-02-23:
Are you getting messages in the ````/yolo/detections```` topic? In the case of the 3D, you also have to set the ```target_frame``` to your base frame, for example, ```camera_link```.

mgonzs13, 2024-02-28:
Hey @amiryanj, how is this going?

closed 2024-03-18 by mgonzs13