mgonzs13/yolo_ros #28: 3D object detection Transformation issue

imdsafi09, 2024-02-27:
Hi,
I am trying to run the stack with only D435 camera. As I dont have the base_link, I change the target_frame into camera_link. However, I am still getting the following error.

"[detect_3d_node-3] [ERROR] [1709019322.086835673] [yolo.detect_3d_node]: Could not transform: "camera_link" passed to lookupTransform argument target_frame does not exist. 
"
[image]

mgonzs13, 2024-02-28:
Hey @imdsafi09, I couldn't reproduce your error. I may be a problem with your tf. Which ROS distro are you using?

imdsafi09, 2024-03-25:
Thank you for your response, @mgonzs13. The issue was caused by a mismatch in the input image channel, which was resolved when I converted mono16 to rgb8.

imdsafi09, 2024-04-30:
Hi @mgonzs13 , sorry for the late response. I resolved the previous issue, and it's working with the Ouster-lidar signal image topic. However, 3D bounding visualization is the same for all the classes. Would it be possible to have a different virtualization based on the class type, similar to a 2D one?

mgonzs13, 2024-04-30:
Hi @imdsafi09, do you mean changing the color of the 3D bounding box? I think that can be easily included.

mgonzs13, 2024-04-30:
Check out the [new version](https://github.com/mgonzs13/yolov8_ros/releases/tag/3.0.2) that includes the 3D bounding boxes with the same color as the 2D boxes.

imdsafi09, 2024-05-01:
Hi @mgonzs13,
 Thank you for your excellent work. I implemented it successfully, and the 2D and 3D versions work fine. In my case, I am integrating 3D object detection with 3D mapping to save the object's location within the map. I tried several approaches. However, they cause redundancy and make the overall process computationally more expensive. Could you please give any suggestions?
Regards,

mgonzs13, 2024-05-01:
I have never used semantic localization. Which works have you checked? From a fast search, I have found these [1](https://www.researchgate.net/profile/David-Chaves-4/publication/333625768_Integration_of_CNN_into_a_Robotic_Architecture_to_Build_Semantic_Maps_of_Indoor_Environments/links/5d402e134585153e592d3545/Integration-of-CNN-into-a-Robotic-Architecture-to-Build-Semantic-Maps-of-Indoor-Environments.pdf), [2](https://arxiv.org/pdf/2210.01456), [3](https://www-sop.inria.fr/members/Renato-Jose.Martins/files/preprint_semanticMapping.pdf). Do you have a public repo to take a look at your work?

imdsafi09, 2024-05-01:
I had no known issues with mapping or localization. I need help to keep/save the detected 3D object position on the map. I am attaching the script I am following for you to look over. It would be similar but different from this one. 
[1](https://github.com/verlab/3DSemanticMapping_JINT_2020/blob/master/auto/src/map_interceptor.py)
[2](https://github.com/roy2909/QuadrupedSearchandRescue/blob/main/frontier/frontier/frontier_detection.pyl)

mgonzs13, 2024-05-01:
So, If I understand the code in [1](https://github.com/verlab/3DSemanticMapping_JINT_2020/blob/master/auto/src/map_interceptor.py), markers from door detections are used to modify the occupancy grid map of the localization. As a result, you will have the detections included in the map. Or do you also want to visualize the detection markers? You can add a publisher to publish the markers included in the map so that the don't desappear.

imdsafi09, 2024-05-02:
Yes, you understood it correctly. They used markers from objects of interest to modify the occupancy grid map. Technically, the markers visualization does not add value to the overall system. Only the localization of the object within the map is essential for my case. However, for visual understanding, I am trying to add detected markers as well. Thank you for all your insightful comments. I will look into it as you suggested.

HHX0607, 2024-05-23:
> Hi, I am trying to run the stack with only D435 camera. As I dont have the base_link, I change the target_frame into camera_link. However, I am still getting the following error even though the camera_link frame exists.
> 
> "[detect_3d_node-3] [ERROR] [1709019322.086835673] [yolo.detect_3d_node]: Could not transform: "camera_link" passed to lookupTransform argument target_frame does not exist. " ![Screenshot from 2024-02-27 16-40-54](https://private-user-images.githubusercontent.com/61491400/308080322-bf848c04-7757-4ab3-911d-aa08480964fd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0NTg5NTgsIm5iZiI6MTcxNjQ1ODY1OCwicGF0aCI6Ii82MTQ5MTQwMC8zMDgwODAzMjItYmY4NDhjMDQtNzc1Ny00YWIzLTkxMWQtYWEwODQ4MDk2NGZkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIzVDEwMDQxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTljNjQ5NzIzNDFkMWFhZjcwNDdlYWMyMGU0ZTI1N2RjNWFhMjg5ZDM2NDVjMDVkNzQzODJkMDFjM2Q1NjljODgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.tGaE0VtD_Za27aDVGaXPdwEdAA8kPNBwne6YdpvKS0k)
> 
> ![Screenshot from 2024-02-27 17-12-33](https://private-user-images.githubusercontent.com/61491400/308088492-d3d0dda4-68e8-40ec-ac78-63f03bf4ff6b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0NTg5NTgsIm5iZiI6MTcxNjQ1ODY1OCwicGF0aCI6Ii82MTQ5MTQwMC8zMDgwODg0OTItZDNkMGRkYTQtNjhlOC00MGVjLWFjNzgtNjNmMDNiZjRmZjZiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIzVDEwMDQxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ4YzllNzcxZDdjNTU1ZDQ5YmQ5MTkwNjFkMGVkZWVkODkwYjIyZGM2MTI0ODFlMzA5ODU3N2RiNzlkNjYyNjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.GW8aoVbmVmr5v95-hCh3-TWLdg79Fw3zaRB9iz29H0w)

Hello！I am facing the same problem.Could you tell me how to create a camera_link and what is necessary to set. I just use D435i and i want to test 3d detection on yolov8_ros. Thanks!

HHX0607, 2024-05-23:
> Hi, I am trying to run the stack with only D435 camera. As I dont have the base_link, I change the target_frame into camera_link. However, I am still getting the following error even though the [...]

@imdsafi09 Hello！I am facing the same problem.Could you tell me how to create a camera_link and what is necessary to set. I just use D435i and i want to test 3d detection on yolov8_ros. Thanks!

imdsafi09, 2024-05-24:
Hi @HHX0607,

In my case, I changed the default value from  [default_value="base_link",](https://github.com/mgonzs13/yolov8_ros/blob/main/yolov8_bringup/launch/yolov8_3d.launch.py) to "camera_link"  and it worked fine. Hopefully, it will work for you as well. If you have already made changes in this file and are still facing the problem, then try checking the 'tf_tree'.

lymcs, 2024-09-04:
> 您好，感谢您的回复。这样做的原因是我的相机之前没有发布 /tf_static 帧。但是，我还有另一个问题，我正在使用 Ouster-lidar 信号图像主题 “/ouster/signal_image” 将其订阅为相机。该主题正在使用类似的基于 yolov8 的 ros 包[ultralytics_ros](https://github.com/Alpaca-zip/ultralytics_ros)但是当我尝试在此处订阅它时，它出现以下错误：
> 
> [yolov8_node-1]文件 “/home/imad/.local/lib/python3.10/site-packages/ultralytics/engine/predictor.py”，第 125 行，预处理中 [yolov8_node-1] im = im[...， ：：-1].transpose（（0， 3， 1， 2）） # BGR 到 RGB，BHWC 到 BCHW，（n， 3， h， w）
> 
> 您能否请告诉我可能的解决方案。
> 
> Ubuntu：22.04 ROS-简陋
> 
> 此致敬意

excuse me, my camera had previously published the/tf_static frame.I also dont have the base_link, and I change the target_frame into camera_link.however,"[detect_3d_node-3] [ERROR] [1709019322.086835673] [yolo.detect_3d_node]: Could not transform: "camera_link" passed to lookupTransform argument target_frame does not exist.
"
[image]
/camera/color/camera_info
/camera/color/image_raw
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/imu
/clicked_point
/goal_pose
/initialpose
/parameter_events
/rosout
/tf
/tf_static
/yolo/dbg_image

[image]

I wish your response.

lymcs, 2024-09-04:
> > 您好，感谢您的回复。这样做的原因是我的相机之前没有发布 /tf_static 帧。但是，我还有另一个问题，我正在使用 Ouster-lidar 信号图像主题 “/ouster/signal_image” 将其订阅为相机。该主题正在使用类似的基于 yolov8 的 ros 包[ultralytics_ros](https://github.com/Alpaca-zip/ultralytics_ros)但是当我尝试在此处订阅它时，它出现以下错误：
> > [yolov8_node-1]文件 “/home/imad/.local/lib/python3.10/site-packages/ultralytics/engine/predictor.py”，第 125 行，预处理中 [yolov8_node-1] im = im[...， ：：-1].transpose（（0， 3， 1， 2）） # BGR 到 RGB，BHWC 到 BCHW，（n， 3， h， w）
> > 您能否请告诉我可能的解决方案。
> > Ubuntu：22.04 ROS-简陋
> > 此致敬意
> 
> 对不起，我的相机之前已经发布了 /tf_static 帧。我也没有base_link，我将target_frame更改为 camera_link.但是，“[detect_3d_node-3] [错误] [1709019322.086835673] [yolo.detect_3d_node]：无法转换：”传递给 lookupTransform 参数的“camera_link”target_frame不存在。 “ ![2024-09-04 14-22-07屏幕截图](https://private-user-images.githubusercontent.com/102289088/364248314-e8986e92-8fb5-4dde-8bd7-bf8787052d16.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU0MzQ5NDcsIm5iZiI6MTcyNTQzNDY0NywicGF0aCI6Ii8xMDIyODkwODgvMzY0MjQ4MzE0LWU4OTg2ZTkyLThmYjUtNGRkZS04YmQ3LWJmODc4NzA1MmQxNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwNFQwNzI0MDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01YTE2NzJlNGU0MGNmMTMxNzNkNDQzM2VjMzQyNmQ3ZWZlYWFkYmI3YzM0NzZkZTc0MjFhNmI3MDUxMjFmMDdmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Jcpj33oy1PCaBbhLy1S1BL76DFWzBVJGs_a5PQ_lHh8) /camera/color/camera_info /camera/color/image_raw /camera/color/metadata /camera/depth/camera_info /camera/depth/image_rect_raw /camera/depth/metadata /camera/extrinsics/depth_to_color /camera/imu /clicked_point /goal_pose /initialpose /parameter_events /rosout /tf /tf_static /yolo/dbg_image
> 
> ![2024-09-04 14-25-09屏幕截图](https://private-user-images.githubusercontent.com/102289088/364248711-2ab647e7-3ceb-4384-a319-5d3dd54e6d0f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU0MzQ5NDcsIm5iZiI6MTcyNTQzNDY0NywicGF0aCI6Ii8xMDIyODkwODgvMzY0MjQ4NzExLTJhYjY0N2U3LTNjZWItNDM4NC1hMzE5LTVkM2RkNTRlNmQwZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwNFQwNzI0MDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YjhlN2MxMGRmNWU5YjcwMTQ3YThhMGMyYmYxZTNmZDQzNWUxNGY0ZWZjNTFlYjczY2I3MDA0NDFjMDNiYzMzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-hhF8WSMJtlhjPowvLo193DisHxTyxk1-q1o1kHYs9g)
> 
> 我希望得到您的回复。

[image]

imdsafi09, 2024-09-04:
Hi @lymcs,
If you change the base_link into camer_link correctly and you are running a real-time depth camera (d435, d435i, etc.), then try tf to check the tf tree or check in rviz if you have a camera_link ? as shown in the screenshot. If there is another error, then it should work.
[image]

fadai333, 2024-09-05:
> 对不起，我的相机之前已经发布了 /tf_static 帧。我也没有base_link，我将target_frame更改为 camera_link.但是，“[detect_3d_node-3] [错误] [1709019322.086835673] [yolo.detect_3d_node]：无法转换：”传递给 lookupTransform [...]

Hello, may I ask if this problem has been solved? This problem also occurred when I was running the program, but I found that the /tf_static topic has no result to output, I don't know whether it is caused by this error, I hope to get your reply. My version of ros2 is foxy

weilinhan, 2024-09-08:
hi,I have run the 3d detect node with no errors and I can already see the 2d detect box on rviz2, what else do I need to add to see the 3d detect box?

closed 2024-12-29 by github-actions[bot]