mgonzs13/yolo_ros #34: Lifecycle nodes

kongminseok, 2024-04-24:
Hi, I would like to measure the CPU usage per core as described in the readme. Could you please provide detailed guidance on how to do this? Additionally, is there a way to determine how many CPU cores are being used?

agonzc34, 2024-04-25:
Hi, to measure the CPU usage I used htop and bashtop. Those tools can get the CPU per process.
[image]

And to determine the number of CPUs I think there isn't any tool that can measure it. In this case, you can analyze the CPU usage graphs or you can read the code searching for multi threading.

I hope it helps :)

kongminseok, 2024-04-25:
@agonzc34 Thank you for your response:)