mgonzs13/yolo_ros #35: No image in Rviz2 using realsense camera!

user-redans, 2024-04-30:
Hello,

I've tried to launch 'yolov8_bringup/yolov8.launch.py' but I have encountered an issue on Rviz2 yolo topic vizualization, see image below:
[image]
 
There is not any camera image from topic /yolo/dbg_image...You can see below a screenshoot with log terminal information:
[image]
 
This is a list with all my ROS2-humble available topics:
[image]

And some info's about my Jetson AGX Xavier that I'm using, with Jetpack 5.1.3: 
[image]

I'm using a Intel RealSense camera, model D415.

I don't know what I'm missing? Do you have any suggestions?

mgonzs13, 2024-04-30:
Hi @user-redans, which camera topic are you using? In your image, there are two topics. You may be publishing your realsense images in the ```/camera/camera/color/image_raw```and yolov8 is subscribed to the ```/camera/rgb/image_raw/ topic. Check also the QoS of the topics. You can modify both the image topic and the QoS using the launch params, take a look at the README.

user-redans, 2024-04-30:
> Hi @user-redans, which camera topic are you using? In your image, there are two topics. You may be publishing your realsense images in the `/camera/camera/color/image_raw`and yolov8 is subscribed to [...]

I'm using default topic value, in launch .py script: "/camera/camera/color/image_raw" and image_reliability = 1

user-redans, 2024-04-30:
Topic "/camera/rgb/image_raw/" vizualization is also wrong, see image below:
[image]

user-redans, 2024-04-30:
This is my graph with all nodes and topics(rqt_graph), see image below: 

[image]

mgonzs13, 2024-04-30:
In the rqt_graph, you are using the topic /camera/rgb/image_raw instead of /camera/camera/color/image_raw. There are new changes in the repository; could you try to see if they work for you?

user-redans, 2024-04-30:
It didn't work!
See my new graph:
[image]

mgonzs13, 2024-04-30:
How are you launching it? It would be best if you were using something similar to this: ```ros2 launch yolov8_bringup yolov8.launch.py input_image_topic:=/camera/camera/color/image_raw image_reliability:=1```.

user-redans, 2024-05-01:
I've launched with the same command line as you told me, but didn't work!
In Rviz2 only this topic "/camera/camera/color/image_raw" works, see image below:
[image]

I've used this command line for camera launching: ros2 launch realsense2_camera rs_launch.py

mgonzs13, 2024-05-01:
Can you show me your rqt_graoh again after launching the command?

mgonzs13, 2024-05-08:
Hi @user-redans, how is this going?

closed 2024-05-20 by user-redans