mgonzs13/yolo_ros #37: custom yolo pretrained problem.

TheSmileXiao, 2024-05-06:
It has problem when using the custom model, it is a yolov8s-seg model trained by me with the pre-trained models, and it does not le me use it, however yolov9m-seg allowed me to use it.

I tryed ```ros2 launch yolov8_bringup yolov8.launch.py model:=yolov8s-seg.pt ```
And show below infos in terminal.

[INFO] [launch]: All log files can be found below /home/xiao/.ros/log/2024-05-07-00-15-06-915019-xiao-laptop-3712
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [yolov8_node-1]: process started with pid [3713]
[INFO] [tracking_node-2]: process started with pid [3715]
[INFO] [debug_node-3]: process started with pid [3717]
[tracking_node-2] /usr/lib/python3/dist-packages/scipy/__init__.py:146: UserWarning: A NumPy version >=1.17.3 and <1.25.0 is required for this version of SciPy (detected version 1.26.4
[tracking_node-2]   warnings.warn(f"A NumPy version >={np_minversion} and <{np_maxversion}"
[yolov8_node-1] [INFO] [1715033711.300011187] [yolo.yolov8_node]: Yolov8Node created
[yolov8_node-1] [INFO] [1715033711.300812167] [yolo.yolov8_node]: Configuring yolov8_node
[tracking_node-2] [INFO] [1715033711.372381450] [yolo.tracking_node]: Configuring tracking_node
[yolov8_node-1] [INFO] [1715033711.388569432] [yolo.yolov8_node]: Activating yolov8_node
[tracking_node-2] [INFO] [1715033711.464118411] [yolo.tracking_node]: Activating tracking_node
[debug_node-3] [INFO] [1715033711.501457219] [yolo.debug_node]: Debug node created
[debug_node-3] [INFO] [1715033711.502009061] [yolo.debug_node]: Configuring debug_node
[debug_node-3] [INFO] [1715033711.592135822] [yolo.debug_node]: Activating debug_node
[yolov8_node-1] Traceback (most recent call last):
[yolov8_node-1]   File "/home/xiao/ros2_ws/install/yolov8_ros/lib/yolov8_ros/yolov8_node", line 33, in <module>
[yolov8_node-1]     sys.exit(load_entry_point('yolov8-ros==0.0.0', 'console_scripts', 'yolov8_node')())
[yolov8_node-1]   File "/home/xiao/ros2_ws/install/yolov8_ros/lib/python3.10/site-packages/yolov8_ros/yolov8_node.py", line 293, in main
[yolov8_node-1]     rclpy.spin(node)
[yolov8_node-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 222, in spin
[yolov8_node-1]     executor.spin_once()
[yolov8_node-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 739, in spin_once
[yolov8_node-1]     self._spin_once_impl(timeout_sec)
[yolov8_node-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 736, in _spin_once_impl
[yolov8_node-1]     raise handler.exception()
[yolov8_node-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
[yolov8_node-1]     self._handler.send(None)
[yolov8_node-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 437, in handler
[yolov8_node-1]     await call_coroutine(entity, arg)
[yolov8_node-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 362, in _execute_subscription
[yolov8_node-1]     await await_or_execute(sub.callback, msg)
[yolov8_node-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
[yolov8_node-1]     return callback(*args)
[yolov8_node-1]   File "/home/xiao/ros2_ws/install/yolov8_ros/lib/python3.10/site-packages/yolov8_ros/yolov8_node.py", line 239, in image_cb
[yolov8_node-1]     results = self.yolo.predict(
[yolov8_node-1]   File "/home/xiao/.local/lib/python3.10/site-packages/ultralytics/engine/model.py", line 439, in predict
[yolov8_node-1]     return self.predictor.predict_cli(source=source) if is_cli else self.predictor(source=source, stream=stream)
[yolov8_node-1]   File "/home/xiao/.local/lib/python3.10/site-packages/ultralytics/engine/predictor.py", line 168, in __call__
[yolov8_node-1]     return list(self.stream_inference(source, model, *args, **kwargs))  # merge list of Result into one
[yolov8_node-1]   File "/home/xiao/.local/lib/python3.10/site-packages/torch/utils/_contextlib.py", line 35, in generator_context
[yolov8_node-1]     response = gen.send(None)
[yolov8_node-1]   File "/home/xiao/.local/lib/python3.10/site-packages/ultralytics/engine/predictor.py", line 228, in stream_inference
[yolov8_node-1]     self.model.warmup(imgsz=(1 if self.model.pt or self.model.triton else self.dataset.bs, 3, *self.imgsz))
[yolov8_node-1]   File "/home/xiao/.local/lib/python3.10/site-packages/ultralytics/nn/autobackend.py", line 596, in warmup
[yolov8_node-1]     self.forward(im)  # warmup
[yolov8_node-1]   File "/home/xiao/.local/lib/python3.10/site-packages/ultralytics/nn/autobackend.py", line 425, in forward
[yolov8_node-1]     y = self.model(im, augment=augment, visualize=visualize, embed=embed)
[yolov8_node-1]   File "/home/xiao/.local/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1532, in _wrapped_call_impl
[yolov8_node-1]     return self._call_impl(*args, **kwargs)
[yolov8_node-1]   File "/home/xiao/.local/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1541, in _call_impl
[yolov8_node-1]     return forward_call(*args, **kwargs)
[yolov8_node-1]   File "/home/xiao/.local/lib/python3.10/site-packages/ultralytics/nn/tasks.py", line 89, in forward
[yolov8_node-1]     return self.predict(x, *args, **kwargs)
[yolov8_node-1]   File "/home/xiao/.local/lib/python3.10/site-packages/ultralytics/nn/tasks.py", line 107, in predict
[yolov8_node-1]     return self._predict_once(x, profile, visualize, embed)
[yolov8_node-1]   File "/home/xiao/.local/lib/python3.10/site-packages/ultralytics/nn/tasks.py", line 128, in _predict_once
[yolov8_node-1]     x = m(x)  # run
[yolov8_node-1]   File "/home/xiao/.local/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1532, in _wrapped_call_impl
[yolov8_node-1]     return self._call_impl(*args, **kwargs)
[yolov8_node-1]   File "/home/xiao/.local/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1541, in _call_impl
[yolov8_node-1]     return forward_call(*args, **kwargs)
[yolov8_node-1]   File "/home/xiao/.local/lib/python3.10/site-packages/ultralytics/nn/modules/head.py", line 111, in forward
[yolov8_node-1]     x = self.detect(self, x)
[yolov8_node-1]   File "/home/xiao/.local/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1709, in __getattr__
[yolov8_node-1]     raise AttributeError(f"'{type(self).__name__}' object has no attribute '{name}'")
[yolov8_node-1] AttributeError: 'Segment' object has no attribute 'detect'
[ERROR] [yolov8_node-1]: process has died [pid 3713, exit code 1, cmd '/home/xiao/ros2_ws/install/yolov8_ros/lib/yolov8_ros/yolov8_node --ros-args -r __node:=yolov8_node -r __ns:=/yolo --params-file /tmp/launch_params_zms6h5zx -r image_raw:=/camera/rgb/image_raw'].

[image]

mgonzs13, 2024-05-08:
Hi @TheSmileXiao, I have launched yolov8_ros with the pretrained yolov8s-seg.pt and it works fine. Are you using the versions from the [requirements.txt](https://github.com/mgonzs13/yolov8_ros/blob/main/requirements.txt)? Is your trained model public available to try it?

TheSmileXiao, 2024-05-08:
> Hi @TheSmileXiao, I have launched yolov8_ros with the pretrained yolov8s-seg.pt and it works fine. Are you using the versions from the [requirements.txt](https://github.com/mgonzs13/yolov8_ros/blob/main/requirements.txt)? Is your trained model public available to try it?

Hi, I am also able to start yolov8m-seg working, but have problems with customed model.Have you free time to try this model?

Thank you very much.
 
[yolov8_xiao-seg.zip](https://github.com/mgonzs13/yolov8_ros/files/15250196/yolov8_xiao-seg.zip)

mgonzs13, 2024-05-09:
Hi @TheSmileXiao, I have just tested you model and I am not getting any error. Please, check the version of the requirements. Which version or commit of this package are you using?

TheSmileXiao, 2024-05-09:
Thanks, I will check my packes.

closed 2024-05-09 by TheSmileXiao

TheSmileXiao, 2024-05-13:
The problem occurs due to a mismatch between the model version and the ultralytics version.
If anyone have the same problems.

https://github.com/ultralytics/ultralytics/issues/11727