mgonzs13/yolo_ros #38: ros2 version

fatemeh-mohseni-AI, 2024-05-08:
I had a question about the version of ROS2 which is compatible with this repository. 
can I use foxy and humble for example ?
doesn't matter which version is installed?

mgonzs13, 2024-05-09:
Hey @fatemeh-mohseni-AI, this repository is fully compatible with Humble. I suppose it also works in Foxy since the yolo nodes use the same code as Humble.

mgonzs13, 2024-05-14:
Hi @fatemeh-mohseni-AI, how you try it in Foxy?

fatemeh-mohseni-AI, 2024-05-20:
hello . sorry for being late .
I tried the installation process on ubuntu 20 desktop + ROS foxy . that was successful , (I just installed the package and didn't run it )
But in dockrizing process there was an error with this content : 

**173.0 #All required rosdeps installed successfully
173.7 Starting >>> yolov8_ros
175.2 Finished <<< yolov8_ros [1.54s]
175.2 Starting >>> yolov8_msgs
175.8 --- stderr: yolov8_msgs
175.8 CMake Error at CMakeLists.txt:5 (find_package):
175.8   By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
175.8   has asked CMake to find a package configuration file provided by
175.8   "ament_cmake", but CMake did not find one.
175.8 
175.8   Could not find a package configuration file provided by "ament_cmake" with
175.8   any of the following names:
175.8 
175.8     ament_cmakeConfig.cmake
175.8     ament_cmake-config.cmake
175.8 
175.8   Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
175.8   "ament_cmake_DIR" to a directory containing one of the above files.  If
175.8   "ament_cmake" provides a separate development package or SDK, be sure it
175.8   has been installed.
175.8 
175.8 
175.8 ---
175.8 Failed   <<< yolov8_msgs [0.64s, exited with code 1]
175.9** 

And this is the Dockerfile : 

```
FROM --platform=linux/amd64 ros:foxy-ros-base

ENV repo=/root/ros2_ws/src/Ros2_Yolov8
ENV wd=/root/ros2_ws
ENV ROS_DISTRO=foxy

RUN mkdir -p $repo
WORKDIR $wd

COPY . $repo

RUN apt-get update &&  \
    apt-get install -y --no-install-recommends  \
      ros-foxy-ament-cmake python3-pip python3-rosdep ros-foxy-rclpy ros-foxy-sensor-msgs

RUN cd $repo && \
    pip3 install --no-cache-dir -r requirements.txt && \
    pip3 install --no-cache-dir -U rosdep colcon-common-extensions && \
    cd $wd && \
    rosdep update && \
    rosdep install --from-paths src --ignore-src -r -y && \
    colcon build

CMD ["ros2", "launch", "yolov8_bringup", "yolov8.launch.py", "model:=yolov8m-seg.pt"]
```

also I tried build it on ubuntu 22 , both server and desktop

fatemeh-mohseni-AI, 2024-05-20:
surprisingly I just solved it . 
this is the new Dockerfile : 

 
```
FROM --platform=linux/amd64 ros:foxy-ros-base

ENV repo=/root/ros2_ws/src/Ros2_Yolov8
ENV wd=/root/ros2_ws
ENV ROS_DISTRO=foxy

RUN mkdir -p $repo
WORKDIR $wd

COPY . $repo

RUN apt-get update &&  \
    apt-get install -y --no-install-recommends  \
      ros-foxy-ament-cmake \
      python3-pip \
      python3-rosdep \
      ros-foxy-rclpy \
      ros-foxy-sensor-msgs && \
    rm -rf /var/lib/apt/lists/*

RUN cd $repo && \
    pip3 install --no-cache-dir -r requirements.txt && \
    pip3 install --no-cache-dir -U rosdep colcon-common-extensions

RUN rosdep update && \
    rosdep install --from-paths src --ignore-src -r -y

RUN /bin/bash -c "source /opt/ros/$ROS_DISTRO/setup.bash && colcon build"
CMD ["ros2", "launch", "yolov8_bringup", "yolov8.launch.py", "model:=yolov8m-seg.pt"]
```

this line was the solution :

**/bin/bash -c "source /opt/ros/$ROS_DISTRO/setup.bash** 

I have just created the Dockerfile for both **amd** and **arm** systems like Jetson families 
I can share it if you want .

mgonzs13, 2024-05-21:
Yeah, Python lifecycle is only available from Humble. A foxy branch can be created to remove them.

fatemeh-mohseni-AI, 2024-05-21:
> Yeah, Python lifecycle is only available from Humble. A foxy branch can be created to remove them.

So it can not be installed in foxy totally

mgonzs13, 2024-05-21:
You can clone the humble rclpy in your docker to include the lifecycle or you can edit yolov8_ros to replace lifecycle with nodes.

richardp4, 2024-05-26:
Hi,
I got the same problem.
When i launched yolov8_bringup yolov8.launch.py, I got this reply.
My ROS version is foxy not humble., where can i get the rclpy.lifecycle?

$ ros2 launch yolov8_bringup yolov8.launch.py

[yolov8_node-1]   File "/home/user/robot_ws/install/yolov8_ros/lib/yolov8_ros/yolov8_node", line 11, in <module>
[yolov8_node-1]     load_entry_point('yolov8-ros==0.0.0', 'console_scripts', 'yolov8_node')()
[yolov8_node-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 490, in load_entry_point
[yolov8_node-1]     return get_distribution(dist).load_entry_point(group, name)
[yolov8_node-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
[yolov8_node-1]     return ep.load()
[yolov8_node-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2445, in load
[yolov8_node-1]     return self.resolve()
[yolov8_node-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2451, in resolve
[yolov8_node-1]     module = __import__(self.module_name, fromlist=['__name__'], level=0)
[yolov8_node-1]   File "/home/user/robot_ws/install/yolov8_ros/lib/python3.8/site-packages/yolov8_ros/yolov8_node.py", line 24, in <module>
[yolov8_node-1]     from rclpy.lifecycle import LifecycleNode
[yolov8_node-1] ModuleNotFoundError: No module named 'rclpy.lifecycle'
[ERROR] [yolov8_node-1]: process has died [pid 55357, exit code 1, cmd '/home/user/robot_ws/install/yolov8_ros/lib/yolov8_ros/yolov8_node --ros-args -r __node:=yolov8_node -r __ns:=/yolo --params-file /tmp/launch_params_830zfy0j -r image_raw:=/camera/rgb/image_raw'].
[tracking_node-2] Traceback (most recent call last):
[tracking_node-2]   File "/home/user/robot_ws/install/yolov8_ros/lib/yolov8_ros/tracking_node", line 11, in <module>
[tracking_node-2]     load_entry_point('yolov8-ros==0.0.0', 'console_scripts', 'tracking_node')()
[tracking_node-2]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 490, in load_entry_point
[tracking_node-2]     return get_distribution(dist).load_entry_point(group, name)
[tracking_node-2]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
[tracking_node-2]     return ep.load()
[tracking_node-2]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2445, in load
[tracking_node-2]     return self.resolve()
[tracking_node-2]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2451, in resolve
[tracking_node-2]     module = __import__(self.module_name, fromlist=['__name__'], level=0)
[tracking_node-2]   File "/home/user/robot_ws/install/yolov8_ros/lib/python3.8/site-packages/yolov8_ros/tracking_node.py", line 25, in <module>
[tracking_node-2]     from rclpy.lifecycle import LifecycleNode
[tracking_node-2] ModuleNotFoundError: No module named 'rclpy.lifecycle'
[ERROR] [tracking_node-2]: process has died [pid 55359, exit code 1, cmd '/home/user/robot_ws/install/yolov8_ros/lib/yolov8_ros/tracking_node --ros-args -r __node:=tracking_node -r __ns:=/yolo --params-file /tmp/launch_params_nqpq0dp3 -r image_raw:=/camera/rgb/image_raw'].
[debug_node-3] Traceback (most recent call last):
[debug_node-3]   File "/home/user/robot_ws/install/yolov8_ros/lib/yolov8_ros/debug_node", line 11, in <module>
[debug_node-3]     load_entry_point('yolov8-ros==0.0.0', 'console_scripts', 'debug_node')()
[debug_node-3]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 490, in load_entry_point
[debug_node-3]     return get_distribution(dist).load_entry_point(group, name)
[debug_node-3]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
[debug_node-3]     return ep.load()
[debug_node-3]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2445, in load
[debug_node-3]     return self.resolve()
[debug_node-3]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2451, in resolve
[debug_node-3]     module = __import__(self.module_name, fromlist=['__name__'], level=0)
[debug_node-3]   File "/home/user/robot_ws/install/yolov8_ros/lib/python3.8/site-packages/yolov8_ros/debug_node.py", line 29, in <module>
[debug_node-3]     from rclpy.lifecycle import LifecycleNode
[debug_node-3] ModuleNotFoundError: No module named 'rclpy.lifecycle'
[ERROR] [debug_node-3]: process has died [pid 55361, exit code 1, cmd '/home/user/robot_ws/install/yolov8_ros/lib/yolov8_ros/debug_node --ros-args -r __node:=debug_node -r __ns:=/yolo --params-file /tmp/launch_params__emd98zc -r image_raw:=/camera/rgb/image_raw -r detections:=tracking'].

mgonzs13, 2024-05-26:
For those who want to use lifecycles in Foxy, since it is deprecated, you have to get the updated code from [Humble](https://github.com/ros2/rclpy/blob/humble/rclpy/rclpy/lifecycle/__init__.py) or [Rolling](https://github.com/ros2/rclpy/blob/rolling/rclpy/rclpy/lifecycle/__init__.py). But if you don't want to build al the rclpy, you can use an old version of yolov8_ros that does not use lifecycles, like [2.2.2](https://github.com/mgonzs13/yolov8_ros/releases/tag/2.2.2). I hope this can help you.

closed 2024-10-01 by mgonzs13