mgonzs13/yolo_ros #39: Encoding error by ultralytics prediction

SamuelHafner, 2024-05-13:
Hi @mgonzs13, 

thank you for your great work!!! 

I have an error when when running the yolov8_node the self.yolo.predict cause an error (tried on 2 different cameras):

[yolov8_node-1] RuntimeError: Given groups=1, weight of size [48, 3, 3, 3], expected input[1, 4, 384, 640] to have 3 channels, but got 4 channels instead

I could fix this after i changed the desired encoding on the cv2_bridge:
from: 
```python
self.cv_bridge.imgmsg_to_cv2(msg)
```
to 
```python
self.cv_bridge.imgmsg_to_cv2(msg, desired_encoding='bgr8')
```

should we maybe add this permanent or is there some reason this encoding is empty?

mgonzs13, 2024-05-14:
Hey @SamuelHafner, thanks for the issue. A better solution could be as follows. Can you try it?

```cv_image = self.cv_bridge.imgmsg_to_cv2(
                msg,
                desired_encoding=msg.encoding
            )```

SamuelHafner, 2024-05-14:
Hey @mgonzs13, thanks for your fast response. It is also not working because in my case I have an image with the encoding `bgra8` and therefore an cv_image  shape of `(360, 640, 4)`. We have the alpha value too, 

If I change the desired encoding for the image to `bgr8` it converts the cv_image with the shape `(360, 480, 3)`.

And I think that the yolo package can only handle images without alpha

mgonzs13, 2024-05-14:
If you don't want to modify yolov8_ros, you can try to add a node to remove the alpha channel and republish the image as bgr8.

SamuelHafner, 2024-05-14:
Yeah i could. But I added this desired_endcoding permanent now.
Would you like this change too? Should I do an merge request or just let it local. 
Because normally you always would need an bgr8 encoding. 

Btw. these recordings are from robocup@home 2023 right? We also were there. Which team? :)

closed 2024-06-20 by mgonzs13

kylenietfeld, 2024-07-17:
For me, I was using ros-gst-bridge and I simply added the line: "! video/x-raw,format=BGR" to my pipeline so the image being published was bgr format instead of bgra. Then yolo worked as expected.