mgonzs13/yolo_ros #41: Operating system

godcapable, 2024-06-04:
Hi, I recently installed ROS2 Humble on windows 10 operating system.
I'd like to know if your YOLOV8 X ROS2 package can be installed and run on windows 10 operating system.

Actually I tried to install them today as your instruction, but was not really successful:
Git clone and pip install was done well, but rosdep command said there is no available package for windows OS.I ignored the error from rosdep step, and proceeded to colcon build. 

The /yolo/detections topic was created well, but the debugged image was not visualized in RVIZ, and there was no data coming from ros2 topic echo /yolo/detections.

If you think your code can be run on windows 10, then I would like to ask some questions about the error I've encountered.

Thank you so much for your code!

godcapable, 2024-06-04:
I will upload the log file tomorrow.

mgonzs13, 2024-06-04:
@godcapable, I have never tried to run it on Windows but I suppose it can be used on Windows 10 since I am using python packages that should run on it and normal rclpy stuff. Have you modified the QoS of the topics?

godcapable, 2024-06-05:
@mgonzs13 , Thank you for your kind reply. I will fix my error to run code in Windows 10. 
    ※ Have you modified the QoS of the topics? : No I did not modified QoS of the topics. 

■ The image data is generated from NVIDIA Isaac sim camera, 

C:\Users\mm\Desktop>ros2 topic list
/camera/rgb/image_raw
/camera_camera_info
/camera_depth
/clock
/parameter_events
/rosout
/tf

The topic named "/camera/rgb/image_raw" is publishing the image data successfully, which can be visualized in RVIZ2. 

■ After ros2 launch yolov8_bringup yolov8.launch.py

C:\Users\mm\Desktop>ros2 topic list
/camera/rgb/image_raw
/camera_camera_info
/camera_depth
/clock
/parameter_events
/rosout
/tf
/yolo/dbg_image
/yolo/debug_node/transition_event
/yolo/detections
/yolo/dgb_bb_markers
/yolo/dgb_kp_markers
/yolo/tracking
/yolo/yolov8_node/transition_event

■ launch.log :

C:\Users\mm\Desktop>ros2 launch yolov8_bringup yolov8.launch.py
[INFO] [launch]: All log files can be found below C:\Users\mm\.ros\log\2024-06-05-10-55-06-993300-KORCO087849-1968
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [yolov8_node.EXE-1]: process started with pid [9852]
[INFO] [tracking_node.EXE-2]: process started with pid [12972]
[INFO] [debug_node.EXE-3]: process started with pid [6036]
[ERROR] [yolov8_node.EXE-1]: process has died [pid 9852, exit code 1, cmd 'c:\ros2_workspace\install\yolov8_ros\lib\yolov8_ros\yolov8_node.EXE --ros-args -r __node:=yolov8_node -r __ns:=/yolo --params-file C:\Users\mm\AppData\Local\Temp\launch_params_lww19mzd -r image_raw:=/camera/rgb/image_raw'].
[yolov8_node.EXE-1] [INFO] [1717552525.017452100] [yolo.yolov8_node]: Yolov8 Node created
**(After the log the terminal is stopped forever, so I pressed ctrl-c, This behavior is repeating for every single trial)**

■ ros2 topic echo /yolo/detections
    I see no data transferred.

■ rqt graph looks like
[image]

To see QoS setting of the image publish topic,
■ ros2 topic info /camera/rgb/image_raw -v

C:\Users\mm\Desktop>ros2 topic info /camera/rgb/image_raw -v
Type: sensor_msgs/msg/Image

Publisher count: 1

Node name: _Render_PostProcess_SDGPipeline_RenderProduct_Replicator_NodeWriterWriter_01
Node namespace: /
Topic type: sensor_msgs/msg/Image
Endpoint type: PUBLISHER
GID: 01.0f.c5.df.80.65.76.d6.01.00.00.00.00.00.08.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 2

Node name: debug_node
Node namespace: /yolo
Topic type: sensor_msgs/msg/Image
Endpoint type: SUBSCRIPTION
GID: 01.0f.c5.df.38.0f.99.ba.01.00.00.00.00.00.1f.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: tracking_node
Node namespace: /yolo
Topic type: sensor_msgs/msg/Image
Endpoint type: SUBSCRIPTION
GID: 01.0f.c5.df.78.4c.15.d7.01.00.00.00.00.00.1d.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

■ ros2 topic info /yolo/detections
Type: yolov8_msgs/msg/DetectionArray
Publisher count: 0
Subscription count: 2

Could you give me any suggestion to modify QoS setting? 

Thank you!

mgonzs13, 2024-06-05:
The rqt_graph seems fine. You can use launch args to change the QoS of yolo ```ros2 launch yolov8_bringup yolov8.launch.py image_reliability:=1```. Could you check the README with the other arguments?

I need more information about the error of yolov8_node. It seems to break at the very start. You can add more logs to check where the error is.

godcapable, 2024-06-05:
I tried QoS value 0,1,2 and could not resolve the problem.
Here is the log file.

1717589010.3518336 [INFO] [launch]: All log files can be found below C:\Users\oh\.ros\log\2024-06-05-21-03-29-794825-KORCO087849-17168
1717589010.3528259 [INFO] [launch]: Default logging verbosity is set to INFO
1717589011.3120906 [INFO] [yolov8_node.EXE-1]: process started with pid [28268]
1717589011.3125870 [INFO] [tracking_node.EXE-2]: process started with pid [15356]
1717589011.3130825 [INFO] [debug_node.EXE-3]: process started with pid [25888]
1717589026.8383901 [yolov8_node.EXE-1] YOLOv8m summary (fused): 218 layers, 25886080 parameters, 0 gradients, 78.9 GFLOPs
1717589028.7608855 [ERROR] [yolov8_node.EXE-1]: process has died [pid 28268, exit code 1, cmd 'c:\ros2_workspace\install\yolov8_ros\lib\yolov8_ros\yolov8_node.EXE --ros-args -r __node:=yolov8_node -r __ns:=/yolo --params-file C:\Users\oh\AppData\Local\Temp\launch_params_4wwcwo67 -r image_raw:=/camera/rgb/image_raw'].
1717589028.7618780 [yolov8_node.EXE-1] [INFO] [1717589025.905938700] [yolo.yolov8_node]: Yolov8 Node created

1717589037.6954601 [WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
1717589039.5301654 [ERROR] [debug_node.EXE-3]: process has died [pid 25888, exit code 3221225786, cmd 'c:\ros2_workspace\install\yolov8_ros\lib\yolov8_ros\debug_node.EXE --ros-args -r __node:=debug_node -r __ns:=/yolo --params-file C:\Users\oh\AppData\Local\Temp\launch_params_vq4nurin -r image_raw:=/camera/rgb/image_raw -r detections:=tracking'].
1717589039.5311575 [debug_node.EXE-3] [INFO] [1717589026.026702800] [yolo.debug_node]: Debug node created

1717589039.5549669 [ERROR] [tracking_node.EXE-2]: process has died [pid 15356, exit code 3221225786, cmd 'c:\ros2_workspace\install\yolov8_ros\lib\yolov8_ros\tracking_node.EXE --ros-args -r __node:=tracking_node -r __ns:=/yolo --params-file C:\Users\oh\AppData\Local\Temp\launch_params_aeyu3ec3 -r image_raw:=/camera/rgb/image_raw'].
1717589039.5559587 [tracking_node.EXE-2] [INFO] [1717589026.468462900] [yolo.tracking_node]: Configuring tracking_node

It seems c:\ros2_workspace\install\yolov8_ros\lib\yolov8_ros\yolov8_node.EXE, shows some error :

[image]

mgonzs13, 2024-06-05:
You should use image_reliability:=1 because of the QoS of the topics of your camera. I don't understand why you are getting the yolov8_node error before configuring and activating it. Can you add some logs in the __init__ of the node?

godcapable, 2024-06-05:
Could you help me how to add logs in the init of the node?

godcapable, 2024-06-05:
`C:\Users\oh\Desktop>c:\ros2_workspace\install\yolov8_ros\lib\yolov8_ros\yolov8_node.EXE --ros-args -r __node:=yolov8_node -r __ns:=/yolo --params-file C:\Users\skyun.oh\AppData\Local\Temp\launch_params_vbnoygpe -r image_raw:=/camera/rgb/image_raw
`

[INFO] [1717594310.802842700] [yolo.yolov8_node]: Yolov8 Node created
[INFO] [1717594310.811385300] [yolo.yolov8_node]: Configuring yolov8_node
[INFO] [1717594311.217662700] [yolo.yolov8_node]: Activating yolov8_node
YOLOv8m summary (fused): 218 layers, 25886080 parameters, 0 gradients, 78.9 GFLOPs
Traceback (most recent call last):
  File "c:\ros2_workspace\install\yolov8_ros\lib\yolov8_ros\yolov8_node-script.py", line 33, in <module>
    sys.exit(load_entry_point('yolov8-ros==0.0.0', 'console_scripts', 'yolov8_node')())
  File "c:\ros2_workspace\install\yolov8_ros\Lib\site-packages\yolov8_ros\yolov8_node.py", line 293, in main
    rclpy.spin(node)
  File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\__init__.py", line 222, in spin
    executor.spin_once()
  File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\executors.py", line 712, in spin_once
    raise handler.exception()
  File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\task.py", line 239, in __call__
    self._handler.send(None)
  File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\executors.py", line 418, in handler
    await call_coroutine(entity, arg)
  File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\executors.py", line 343, in _execute_subscription
    await await_or_execute(sub.callback, msg)
  File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\executors.py", line 107, in await_or_execute
    return callback(*args)
  File "c:\ros2_workspace\install\yolov8_ros\Lib\site-packages\yolov8_ros\yolov8_node.py", line 238, in image_cb
    cv_image = self.cv_bridge.imgmsg_to_cv2(msg)
  File "c:\opt\ros\humble\x64\Lib\site-packages\cv_bridge\core.py", line 169, in imgmsg_to_cv2
    dtype, n_channels = self.encoding_to_dtype_with_channels(img_msg.encoding)
  File "c:\opt\ros\humble\x64\Lib\site-packages\cv_bridge\core.py", line 104, in encoding_to_dtype_with_channels
    return self.cvtype2_to_dtype_with_channels(self.encoding_to_cvtype2(encoding))
  File "c:\opt\ros\humble\x64\Lib\site-packages\cv_bridge\core.py", line 96, in encoding_to_cvtype2
    from cv_bridge.boost.cv_bridge_boost import getCvType
ImportError: DLL load failed while importing cv_bridge_boost: The specified procedure cannot be found.

godcapable, 2024-06-05:
> I have pushed a new version. Can you launch it and check if the error persists?

Okay I will try!

godcapable, 2024-06-05:
> I have pushed a new version. Can you launch it and check if the error persists?

Unfortunately, new version still have the same issue.

mgonzs13, 2024-06-05:
> `C:\Users\oh\Desktop>c:\ros2_workspace\install\yolov8_ros\lib\yolov8_ros\yolov8_node.EXE --ros-args -r __node:=yolov8_node -r __ns:=/yolo --params-file C:\Users\oh\AppData\Local\Temp\launch_params_vbnoygpe -r image_raw:=/camera/rgb/image_raw `
> 
> [INFO] [1717594310.802842700] [yolo.yolov8_node]: Yolov8 Node created [INFO] [1717594310.811385300] [yolo.yolov8_node]: Configuring yolov8_node [INFO] [1717594311.217662700] [yolo.yolov8_node]: Activating yolov8_node YOLOv8m summary (fused): 218 layers, 25886080 parameters, 0 gradients, 78.9 GFLOPs Traceback (most recent call last): File "c:\ros2_workspace\install\yolov8_ros\lib\yolov8_ros\yolov8_node-script.py", line 33, in sys.exit(load_entry_point('yolov8-ros==0.0.0', 'console_scripts', 'yolov8_node')()) File "c:\ros2_workspace\install\yolov8_ros\Lib\site-packages\yolov8_ros\yolov8_node.py", line 293, in main rclpy.spin(node) File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy__init__.py", line 222, in spin executor.spin_once() File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\executors.py", line 712, in spin_once raise handler.exception() File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\task.py", line 239, in **call** self._handler.send(None) File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\executors.py", line 418, in handler await call_coroutine(entity, arg) File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\executors.py", line 343, in _execute_subscription await await_or_execute(sub.callback, msg) File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\executors.py", line 107, in await_or_execute return callback(*args) File "c:\ros2_workspace\install\yolov8_ros\Lib\site-packages\yolov8_ros\yolov8_node.py", line 238, in image_cb cv_image = self.cv_bridge.imgmsg_to_cv2(msg) File "c:\opt\ros\humble\x64\Lib\site-packages\cv_bridge\core.py", line 169, in imgmsg_to_cv2 dtype, n_channels = self.encoding_to_dtype_with_channels(img_msg.encoding) File "c:\opt\ros\humble\x64\Lib\site-packages\cv_bridge\core.py", line 104, in encoding_to_dtype_with_channels return self.cvtype2_to_dtype_with_channels(self.encoding_to_cvtype2(encoding)) File "c:\opt\ros\humble\x64\Lib\site-packages\cv_bridge\core.py", line 96, in encoding_to_cvtype2 from cv_bridge.boost.cv_bridge_boost import getCvType ImportError: DLL load failed while importing cv_bridge_boost: The specified procedure cannot be found.

This seems a different error. Have you installed cv_bridge?

godcapable, 2024-06-07:
Yes, I already have cv_bridge installed in C:\opt\ros\humble\x64\Lib\site-packages

I also tried installing additional cv_bridge from [https://github.com/ros-perception/vision_opencv](https://github.com/ros-perception/vision_opencv), but the error persists.

godcapable, 2024-06-07:
Fresh terminal, fail to use function in cv_bridge in python 

`from cv_bridge boost cv_bridge_boost import getCvType`
`ImportError: DLL load failed while importing cv_bridge_boost: The specified procedure cannot be found.`

Similar error reported : [https://github.com/ros-perception/vision_opencv/issues/207](https://github.com/ros-perception/vision_opencv/issues/207)

[image]

Is using numpy instead of cv_bridge possible solution?

godcapable, 2024-06-07:
I modified yolov8_node.py (and tracking, debug nodes) code to use numpy instead of cv_bridge. This solved the problem.

```

# Copyright (C)```
 2023  Miguel Ángel González Santamarta

# from cv_bridge import CvBridge
import numpy as np

...

class Yolov8Node(LifecycleNode):
...
        # self.cv_bridge = CvBridge()
...
    def image_cb(self, msg: Image) -> None:

            # convert image + predict
            # cv_image = self.cv_bridge.imgmsg_to_cv2(msg)
            im = np.frombuffer(msg.data, dtype=np.uint8).reshape(msg.height, msg.width, -1)

```
```

closed 2024-06-07 by godcapable