mgonzs13/yolo_ros #45: IndexError with Pose Model

kylenietfeld, 2024-08-02:
Hello. The readme mentions that this repo is compatible with pose detection. I trained a pose model and named it Platform.pt. The node runs and I can see unsucessful detections in /yolo/detections. However, as soon as it detects a class, I get this error:
```
[ros2-3] [yolov8_node-1] Traceback (most recent call last):
[ros2-3] [yolov8_node-1]   File "/root/ros2_ws/install/yolov8_ros/lib/yolov8_ros/yolov8_node", line 33, in <module>
[ros2-3] [yolov8_node-1]     sys.exit(load_entry_point('yolov8-ros==0.0.0', 'console_scripts', 'yolov8_node')())
[ros2-3] [yolov8_node-1]   File "/root/ros2_ws/install/yolov8_ros/lib/python3.10/site-packages/yolov8_ros/yolov8_node.py", line 320, in main
[ros2-3] [yolov8_node-1]     rclpy.spin(node)
[ros2-3] [yolov8_node-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 222, in spin
[ros2-3] [yolov8_node-1]     executor.spin_once()
[ros2-3] [yolov8_node-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 739, in spin_once
[ros2-3] [yolov8_node-1]     self._spin_once_impl(timeout_sec)
[ros2-3] [yolov8_node-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 736, in _spin_once_impl
[ros2-3] [yolov8_node-1]     raise handler.exception()
[ros2-3] [yolov8_node-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
[ros2-3] [yolov8_node-1]     self._handler.send(None)
[ros2-3] [yolov8_node-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 437, in handler
[ros2-3] [yolov8_node-1]     await call_coroutine(entity, arg)
[ros2-3] [yolov8_node-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 362, in _execute_subscription
[ros2-3] [yolov8_node-1]     await await_or_execute(sub.callback, msg)
[ros2-3] [yolov8_node-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
[ros2-3] [yolov8_node-1]     return callback(*args)
[ros2-3] [yolov8_node-1]   File "/root/ros2_ws/install/yolov8_ros/lib/python3.10/site-packages/yolov8_ros/yolov8_node.py", line 303, in image_cb
[ros2-3] [yolov8_node-1]     aux_msg.keypoints = keypoints[i]
[ros2-3] [yolov8_node-1] IndexError: list index out of range
```

My pose model has 4 key points. Any idea what is causing this error?

mgonzs13, 2024-08-02:
Hey @kylenietfeld, I have only tried the default pose YOLOv8 model. Could you share with me your model to try it?

kylenietfeld, 2024-08-30:
Hi @mgonzs13, yes I can! I had to put it in a zip file. Let me know if you cannot open it. 
[Model.zip](https://github.com/user-attachments/files/16821318/Model.zip)

mgonzs13, 2024-09-03:
Hey @kylenietfeld, I have tested your model and I have fixed the error. However, I have noticed that your model is no retuning conf values for the detections.

closed 2024-12-29 by github-actions[bot]