mgonzs13/yolo_ros #48: [ERROR] [debug_node-3]: process has died [pid 14347, exit code 1

Ragnar7982, 2024-08-05:
Thanks for giving us a great code, but when I run ros2 launch yolov8_bringup yolov8.launch.py it shows:
[INFO] [launch]: All log files can be found below /home/ragnar/.ros/log/2024-08-05-11-13-50-085883-ragnar-14342
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [yolov8_node-1]: process started with pid [14343]
[INFO] [tracking_node-2]: process started with pid [14345]
[INFO] [debug_node-3]: process started with pid [14347]
[tracking_node-2] Traceback (most recent call last):
[tracking_node-2]   File "/home/ragnar/yolo_ws/install/yolov8_ros/lib/yolov8_ros/tracking_node", line 33, in <module>
[tracking_node-2]     sys.exit(load_entry_point('yolov8-ros==0.0.0', 'console_scripts', 'tracking_node')())
[tracking_node-2]   File "/home/ragnar/yolo_ws/install/yolov8_ros/lib/yolov8_ros/tracking_node", line 25, in importlib_load_entry_point
[tracking_node-2]     return next(matches).load()
[tracking_node-2]   File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
[tracking_node-2]     module = import_module(match.group('module'))
[tracking_node-2]   File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
[tracking_node-2]     return _bootstrap._gcd_import(name[level:], package, level)
[tracking_node-2]   File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
[tracking_node-2]   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
[tracking_node-2]   File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
[tracking_node-2]   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
[tracking_node-2]   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
[tracking_node-2]   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
[tracking_node-2]   File "/home/ragnar/yolo_ws/install/yolov8_ros/lib/python3.10/site-packages/yolov8_ros/tracking_node.py", line 30, in <module>
[tracking_node-2]     from ultralytics.engine.results import Boxes
[tracking_node-2]   File "/home/ragnar/.local/lib/python3.10/site-packages/ultralytics/__init__.py", line 10, in <module>
[tracking_node-2]     from ultralytics.data.explorer.explorer import Explorer
[tracking_node-2]   File "/home/ragnar/.local/lib/python3.10/site-packages/ultralytics/data/__init__.py", line 3, in <module>
[tracking_node-2]     from .base import BaseDataset
[tracking_node-2]   File "/home/ragnar/.local/lib/python3.10/site-packages/ultralytics/data/base.py", line 15, in <module>
[tracking_node-2]     from torch.utils.data import Dataset
[tracking_node-2] ModuleNotFoundError: No module named 'torch'
[yolov8_node-1] Traceback (most recent call last):
[yolov8_node-1]   File "/home/ragnar/yolo_ws/install/yolov8_ros/lib/yolov8_ros/yolov8_node", line 33, in <module>
[yolov8_node-1]     sys.exit(load_entry_point('yolov8-ros==0.0.0', 'console_scripts', 'yolov8_node')())
[yolov8_node-1]   File "/home/ragnar/yolo_ws/install/yolov8_ros/lib/yolov8_ros/yolov8_node", line 25, in importlib_load_entry_point
[yolov8_node-1]     return next(matches).load()
[yolov8_node-1]   File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
[yolov8_node-1]     module = import_module(match.group('module'))
[yolov8_node-1]   File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
[yolov8_node-1]     return _bootstrap._gcd_import(name[level:], package, level)
[yolov8_node-1]   File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
[yolov8_node-1]   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
[yolov8_node-1]   File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
[yolov8_node-1]   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
[yolov8_node-1]   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
[yolov8_node-1]   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
[yolov8_node-1]   File "/home/ragnar/yolo_ws/install/yolov8_ros/lib/python3.10/site-packages/yolov8_ros/yolov8_node.py", line 30, in <module>
[yolov8_node-1]     import torch
[yolov8_node-1] ModuleNotFoundError: No module named 'torch'
[ERROR] [tracking_node-2]: process has died [pid 14345, exit code 1, cmd '/home/ragnar/yolo_ws/install/yolov8_ros/lib/yolov8_ros/tracking_node --ros-args -r __node:=tracking_node -r __ns:=/yolo --params-file /tmp/launch_params_qzl4nm50 -r /oakd/rgb/preview/image_raw:=/oakd/rgb/preview/image_raw'].
[ERROR] [yolov8_node-1]: process has died [pid 14343, exit code 1, cmd '/home/ragnar/yolo_ws/install/yolov8_ros/lib/yolov8_ros/yolov8_node --ros-args -r __node:=yolov8_node -r __ns:=/yolo --params-file /tmp/launch_params_q1fsv89o -r /oakd/rgb/preview/image_raw:=/oakd/rgb/preview/image_raw'].
[debug_node-3] Traceback (most recent call last):
[debug_node-3]   File "/home/ragnar/yolo_ws/install/yolov8_ros/lib/yolov8_ros/debug_node", line 33, in <module>
[debug_node-3]     sys.exit(load_entry_point('yolov8-ros==0.0.0', 'console_scripts', 'debug_node')())
[debug_node-3]   File "/home/ragnar/yolo_ws/install/yolov8_ros/lib/yolov8_ros/debug_node", line 25, in importlib_load_entry_point
[debug_node-3]     return next(matches).load()
[debug_node-3]   File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
[debug_node-3]     module = import_module(match.group('module'))
[debug_node-3]   File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
[debug_node-3]     return _bootstrap._gcd_import(name[level:], package, level)
[debug_node-3]   File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
[debug_node-3]   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
[debug_node-3]   File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
[debug_node-3]   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
[debug_node-3]   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
[debug_node-3]   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
[debug_node-3]   File "/home/ragnar/yolo_ws/install/yolov8_ros/lib/python3.10/site-packages/yolov8_ros/debug_node.py", line 34, in <module>
[debug_node-3]     from ultralytics.utils.plotting import Annotator, colors
[debug_node-3]   File "/home/ragnar/.local/lib/python3.10/site-packages/ultralytics/__init__.py", line 10, in <module>
[debug_node-3]     from ultralytics.data.explorer.explorer import Explorer
[debug_node-3]   File "/home/ragnar/.local/lib/python3.10/site-packages/ultralytics/data/__init__.py", line 3, in <module>
[debug_node-3]     from .base import BaseDataset
[debug_node-3]   File "/home/ragnar/.local/lib/python3.10/site-packages/ultralytics/data/base.py", line 15, in <module>
[debug_node-3]     from torch.utils.data import Dataset
[debug_node-3] ModuleNotFoundError: No module named 'torch'
[ERROR] [debug_node-3]: process has died [pid 14347, exit code 1, cmd '/home/ragnar/yolo_ws/install/yolov8_ros/lib/yolov8_ros/debug_node --ros-args -r __node:=debug_node -r __ns:=/yolo --params-file /tmp/launch_params_lat3ryi1 -r /oakd/rgb/preview/image_raw:=/oakd/rgb/preview/image_raw -r detections:=tracking'].
The main problem might be ModuleNotFoundError: No module named 'torch',But I have torch in conda env, you know where might be the problems? Thank you.
(pytorch) ragnar@ragnar:~$ pip list |grep torch
torch                                2.0.1
torchaudio                           2.0.2
torchmetrics                         0.8.0
torchvision                          0.15.2

mgonzs13, 2024-08-05:
Hi @Ragnar7982, I don't think mixing conda envs and ROS is a good idea. Have you tried to use this repo without conda?

Ragnar7982, 2024-08-05:
I tried once before, but I run it today it shows ERROR:
[debug_node-3] ModuleNotFoundError: No module named 'torch'
[ERROR] [debug_node-3]: process has died [pid 14347, exit code 1, cmd '/home/ragnar/yolo_ws/install/yolov8_ros/lib/yolov8_ros/debug_node --ros-args -r __node:=debug_node -r __ns:=/yolo --params-file /tmp/launch_params_lat3ryi1 -r /oakd/rgb/preview/image_raw:=/oakd/rgb/preview/image_raw -r detections:=tracking'].

That 's way I create conda env? If I run yolo without conda env, it shows the same error, so I don't know where is the problem.
Please let me know if you know how to fix it, thank you.

mgonzs13, 2024-08-05:
Are you installing torch with pip when not using conda? Dopython-is-python3? Have tried installing torch with pip3 and not using a conda env?

AnatoliyZabrovskiy, 2024-10-15:
I have the same problem! Does anyone have a solution? Thanks!

closed 2024-12-29 by github-actions[bot]