mgonzs13/yolo_ros #49: Disabling debug node

godcapable, 2024-08-30:
Hi, I have problem where detection and tracking node is sending message, but debug node is making an error.

I guess it is error comping from cv2_bridge extension.

Would you let me know in which pare of the code might be modified to disable only the debug node?

mgonzs13, 2024-08-30:
Hey @godcapable, you can modify the launch files and comment the debug node. Do you have more logs about the error (this line, the error/exception,...)?

godcapable, 2024-09-02:
Hi @mgonzs13 , actually your code was operating well last week, but it just started to crash.

I removed the installation in ros2 workspace, and try to re-install, but had following error :

It seems that there is import error : 
**ImportError: cannot import name 'Union' from 'types' (c:\opt\ros\humble\x64\lib\types.py)**

Other packages such as opencv are can be installed without any problem. Only yolov8 packages is generating this error.

```

c:\ros2_workspace>colcon build --packages-select yolov8_ros
Starting >>> yolov8_ros
[5.945s] colcon.colcon_ros.prefix_path.catkin WARNING The path 'c:\opt\ros\humble\x64\tools\vcpkg\installed\x64-windows' in the environment variable CMAKE_PREFIX_PATH doesn't exist
--- stderr: yolov8_ros
Traceback (most recent call last):
  File "c:\opt\ros\humble\x64\Lib\site-packages\colcon_core\executor\__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "c:\opt\ros\humble\x64\Lib\site-packages\colcon_core\task\__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "c:\opt\ros\humble\x64\Lib\site-packages\colcon_ros\task\ament_python\build.py", line 105, in build
    return await extension.build(additional_hooks=additional_hooks)
  File "c:\opt\ros\humble\x64\Lib\site-packages\colcon_core\task\python\build.py", line 81, in build
    available_commands = await self._get_available_commands(
  File "c:\opt\ros\humble\x64\Lib\site-packages\colcon_core\task\python\build.py", line 171, in _get_available_commands
    output = await check_output(
  File "c:\opt\ros\humble\x64\Lib\site-packages\colcon_core\subprocess.py", line 128, in check_output
    assert not rc, f'Expected {args} to pass: {stderr_data}'
AssertionError: Expected ['c:\\opt\\ros\\humble\\x64\\python.exe', 'setup.py', '--help-commands'] to pass: Traceback (most recent call last):
  File "c:\ros2_workspace\src\yolov8_ros\yolov8_ros\setup.py", line 5, in <module>
    setup(
  File "c:\opt\ros\humble\x64\Lib\site-packages\setuptools\__init__.py", line 87, in setup
    return distutils.core.setup(**attrs)
  File "c:\opt\ros\humble\x64\Lib\site-packages\setuptools\_distutils\core.py", line 172, in setup
    ok = dist.parse_command_line()
  File "c:\opt\ros\humble\x64\Lib\site-packages\setuptools\_distutils\dist.py", line 472, in parse_command_line
    if self.handle_display_options(option_order):
  File "c:\opt\ros\humble\x64\Lib\site-packages\setuptools\dist.py", line 1184, in handle_display_options
    return _Distribution.handle_display_options(self, option_order)
  File "c:\opt\ros\humble\x64\Lib\site-packages\setuptools\_distutils\dist.py", line 686, in handle_display_options
    self.print_commands()
  File "c:\opt\ros\humble\x64\Lib\site-packages\setuptools\dist.py", line 961, in print_commands
    cmdclass = ep.load()
  File "c:\opt\ros\humble\x64\lib\importlib\metadata\__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "c:\opt\ros\humble\x64\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "c:\opt\ros\humble\x64\Lib\site-packages\sphinx\setup_command.py", line 21, in <module>
    from sphinx.application import Sphinx
  File "c:\opt\ros\humble\x64\Lib\site-packages\sphinx\application.py", line 31, in <module>
    from sphinx.config import Config
  File "c:\opt\ros\humble\x64\Lib\site-packages\sphinx\config.py", line 21, in <module>
    from sphinx.util import logging
  File "c:\opt\ros\humble\x64\Lib\site-packages\sphinx\util\__init__.py", line 41, in <module>
    from sphinx.util.typing import PathMatcher
  File "c:\opt\ros\humble\x64\Lib\site-packages\sphinx\util\typing.py", line 37, in <module>
    from types import Union as types_Union
ImportError: cannot import name 'Union' from 'types' (c:\opt\ros\humble\x64\lib\types.py)

---
Failed   <<< yolov8_ros [18.2s, exited with code 1]

Summary: 0 packages finished [19.0s]
  1 package failed: yolov8_ros
[... 14 lines cut ...]
  File "c:\opt\ros\humble\x64\Lib\site-packages\setuptools\dist.py", line 961, in print_commands
    cmdclass = ep.load()
  File "c:\opt\ros\humble\x64\lib\importlib\metadata\__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "c:\opt\ros\humble\x64\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "c:\opt\ros\humble\x64\Lib\site-packages\sphinx\setup_command.py", line 21, in <module>
    from sphinx.application import Sphinx
[... cut ...]
    from sphinx.config import Config
  File "c:\opt\ros\humble\x64\Lib\site-packages\sphinx\config.py", line 21, in <module>
    from sphinx.util import logging
  File "c:\opt\ros\humble\x64\Lib\site-packages\sphinx\util\__init__.py", line 41, in <module>
    from sphinx.util.typing import PathMatcher
  File "c:\opt\ros\humble\x64\Lib\site-packages\sphinx\util\typing.py", line 37, in <module>
    from types import Union as types_Union
ImportError: cannot import name 'Union' from 'types' (c:\opt\ros\humble\x64\lib\types.py)

```

godcapable, 2024-09-02:
Okay, here is the detailed log during debug_node. 

```
C:\Users\skyun.oh\Desktop>c:\ros2_workspace\install\yolov8_ros\lib\yolov8_ros\debug_node.EXE --ros-args -r __node:=debug_node -r __ns:=/yolo --params-file C:\Users\skyun.oh\AppData\Local\Temp\launch_params_h616yxn1 -r image_raw:=/camera/rgb/image_raw -r detections:=tracking
OMP: Info #268: KMP_ALL_THREADS variable deprecated, please use KMP_DEVICE_THREAD_LIMIT instead.
[INFO] [1725281046.320399100] [yolo.debug_node]: Debug node created
[INFO] [1725281046.326683800] [yolo.debug_node]: Configuring debug_node
[INFO] [1725281046.682858800] [yolo.debug_node]: Activating debug_node
Traceback (most recent call last):
  File "c:\ros2_workspace\install\yolov8_ros\lib\yolov8_ros\debug_node-script.py", line 33, in <module>
    sys.exit(load_entry_point('yolov8-ros==0.0.0', 'console_scripts', 'debug_node')())
  File "c:\ros2_workspace\install\yolov8_ros\Lib\site-packages\yolov8_ros\debug_node.py", line 306, in main
    rclpy.spin(node)
  File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\__init__.py", line 222, in spin
    executor.spin_once()
  File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\executors.py", line 712, in spin_once
    raise handler.exception()
  File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\task.py", line 239, in __call__
    self._handler.send(None)
  File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\executors.py", line 418, in handler
    await call_coroutine(entity, arg)
  File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\executors.py", line 343, in _execute_subscription
    await await_or_execute(sub.callback, msg)
  File "c:\opt\ros\humble\x64\Lib\site-packages\rclpy\executors.py", line 107, in await_or_execute
    return callback(*args)
  File "c:\opt\ros\humble\x64\Lib\site-packages\message_filters\__init__.py", line 83, in callback
    self.signalMessage(msg)
  File "c:\opt\ros\humble\x64\Lib\site-packages\message_filters\__init__.py", line 64, in signalMessage
    cb(*(msg + args))
  File "c:\opt\ros\humble\x64\Lib\site-packages\message_filters\__init__.py", line 313, in add
    self.signalMessage(*msgs)
  File "c:\opt\ros\humble\x64\Lib\site-packages\message_filters\__init__.py", line 64, in signalMessage
    cb(*(msg + args))
  File "c:\ros2_workspace\install\yolov8_ros\Lib\site-packages\yolov8_ros\debug_node.py", line 295, in detections_cb
    self._dbg_pub.publish(self.cv_bridge.cv2_to_imgmsg(cv_image,encoding=img_msg.encoding))
  File "c:\ros2_workspace\install\cv_bridge\Lib\site-packages\cv_bridge\core.py", line 276, in cv2_to_imgmsg
    if self.cvtype_to_name[self.encoding_to_cvtype2(encoding)] != cv_type:
  File "c:\ros2_workspace\install\cv_bridge\Lib\site-packages\cv_bridge\core.py", line 96, in encoding_to_cvtype2
    from cv_bridge.boost.cv_bridge_boost import getCvType
**ImportError: DLL load failed while importing cv_bridge_boost: no such procedure**

```

The error is due to usage of the usage of cv_bridge: My computer doens't allow using cv_bridge.
As such, I should use numpy instead of cv2.

I modified yolov8_node.py and tracking_node.py

```
    def image_cb(self, msg: Image) -> None:

        if self.enable:

            # convert image + predict
            # cv_image = self.cv_bridge.imgmsg_to_cv2(msg)
            im = np.frombuffer(msg.data, dtype=np.uint8).reshape(msg.height, msg.width, -1)
```

This is how I changed the cv2 format to numpy format.
This way, I could safely get message from detection node and tracking node.
But [debug_node.EXE-3] generate error, because I don't know how to re-convert numpy to cv2.

in debug_node.py

```
        # publish dbg image
        self._dbg_pub.publish(self.cv_bridge.cv2_to_imgmsg(cv_image,encoding=img_msg.encoding))
                    
```

**Could you help me change "cv2 to imgmsg conversion" to "numpy to imgmsg conversion"?**

mgonzs13, 2024-09-03:
Hey @godcapable, you can check the [code](https://github.com/ros-perception/vision_opencv/blob/066793a23e5d06d76c78ca3d69824a501c3554fd/cv_bridge/python/cv_bridge/core.py#L239) of CvBridge to get how to do it. So you can do the following:

```python
new_img_msg = sensor_msgs.msg.Image()
new_img_msg.height = cv_image.shape[0]
new_img_msg.width = cv_image.shape[1]
new_img_msg.encoding = img_msg.encoding
new_img_msg.data.frombytes(cv_image.tobytes())
new_img_msg.step = len(new_img_msg.data) 
```

closed 2024-12-29 by github-actions[bot]