mgonzs13/yolo_ros #51: TypeError: canonicalize_version() got an unexpected keyword argument 'strip_trailing_zero'

Savior2016, 2024-09-19:
I got this log after run colcon：TypeError: canonicalize_version() got an unexpected keyword argument 'strip_trailing_zero'

I tried to upgrade my setuptools and colcon， it not helped。

 Below is all my log：

```
root@apollo-nx:/workspace/ros2_yolov# colcon build
Starting >>> yolov8_msgs
--- stderr: yolov8_msgs                              
Traceback (most recent call last):
  File "/workspace/ros2_yolov/build/yolov8_msgs/ament_cmake_python/yolov8_msgs/setup.py", line 4, in <module>
    setup(
  File "/usr/local/lib/python3.10/dist-packages/setuptools/__init__.py", line 117, in setup
    return distutils.core.setup(**attrs)
  File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/core.py", line 183, in setup
    return run_commands(dist)
  File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/core.py", line 199, in run_commands
    dist.run_commands()
  File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py", line 954, in run_commands
    self.run_command(cmd)
  File "/usr/local/lib/python3.10/dist-packages/setuptools/dist.py", line 950, in run_command
    super().run_command(command)
  File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py", line 973, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.10/dist-packages/setuptools/command/egg_info.py", line 311, in run
    self.find_sources()
  File "/usr/local/lib/python3.10/dist-packages/setuptools/command/egg_info.py", line 319, in find_sources
    mm.run()
  File "/usr/local/lib/python3.10/dist-packages/setuptools/command/egg_info.py", line 545, in run
    self.prune_file_list()
  File "/usr/local/lib/python3.10/dist-packages/setuptools/command/sdist.py", line 161, in prune_file_list
    super().prune_file_list()
  File "/usr/local/lib/python3.10/dist-packages/setuptools/_distutils/command/sdist.py", line 380, in prune_file_list
    base_dir = self.distribution.get_fullname()
  File "/usr/local/lib/python3.10/dist-packages/setuptools/_core_metadata.py", line 267, in get_fullname
    return _distribution_fullname(self.get_name(), self.get_version())
  File "/usr/local/lib/python3.10/dist-packages/setuptools/_core_metadata.py", line 285, in _distribution_fullname
    canonicalize_version(version, strip_trailing_zero=False),
TypeError: canonicalize_version() got an unexpected keyword argument 'strip_trailing_zero'
gmake[2]: *** [CMakeFiles/ament_cmake_python_build_yolov8_msgs_egg.dir/build.make:70: CMakeFiles/ament_cmake_python_build_yolov8_msgs_egg] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:447: CMakeFiles/ament_cmake_python_build_yolov8_msgs_egg.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< yolov8_msgs [1.01s, exited with code 2]

Summary: 0 packages finished [1.18s]
  1 package failed: yolov8_msgs
  1 package had stderr output: yolov8_msgs
  2 packages not processed
```

mrepicfails, 2024-10-01:
Same issue for me
```
drone@drone:~/ros2_ws$ colcon build
Starting >>> yolov8_msgs
--- stderr: yolov8_msgs                             
Traceback (most recent call last):
  File "/home/drone/ros2_ws/build/yolov8_msgs/ament_cmake_python/yolov8_msgs/setup.py", line 4, in <module>
    setup(
  File "/home/drone/.local/lib/python3.10/site-packages/setuptools/__init__.py", line 117, in setup
    return distutils.core.setup(**attrs)
  File "/home/drone/.local/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 183, in setup
    return run_commands(dist)
  File "/home/drone/.local/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 199, in run_commands
    dist.run_commands()
  File "/home/drone/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 954, in run_commands
    self.run_command(cmd)
  File "/home/drone/.local/lib/python3.10/site-packages/setuptools/dist.py", line 950, in run_command
    super().run_command(command)
  File "/home/drone/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 973, in run_command
    cmd_obj.run()
  File "/home/drone/.local/lib/python3.10/site-packages/setuptools/command/egg_info.py", line 311, in run
    self.find_sources()
  File "/home/drone/.local/lib/python3.10/site-packages/setuptools/command/egg_info.py", line 319, in find_sources
    mm.run()
  File "/home/drone/.local/lib/python3.10/site-packages/setuptools/command/egg_info.py", line 545, in run
    self.prune_file_list()
  File "/home/drone/.local/lib/python3.10/site-packages/setuptools/command/sdist.py", line 161, in prune_file_list
    super().prune_file_list()
  File "/home/drone/.local/lib/python3.10/site-packages/setuptools/_distutils/command/sdist.py", line 380, in prune_file_list
    base_dir = self.distribution.get_fullname()
  File "/home/drone/.local/lib/python3.10/site-packages/setuptools/_core_metadata.py", line 267, in get_fullname
    return _distribution_fullname(self.get_name(), self.get_version())
  File "/home/drone/.local/lib/python3.10/site-packages/setuptools/_core_metadata.py", line 285, in _distribution_fullname
    canonicalize_version(version, strip_trailing_zero=False),
TypeError: canonicalize_version() got an unexpected keyword argument 'strip_trailing_zero'
gmake[2]: *** [CMakeFiles/ament_cmake_python_build_yolov8_msgs_egg.dir/build.make:70: CMakeFiles/ament_cmake_python_build_yolov8_msgs_egg] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:447: CMakeFiles/ament_cmake_python_build_yolov8_msgs_egg.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< yolov8_msgs [2.92s, exited with code 2]

Summary: 0 packages finished [3.02s]
  1 package failed: yolov8_msgs
  1 package had stderr output: yolov8_msgs
  2 packages not processed

```

mgonzs13, 2024-10-01:
Hey @Savior2016, sorry for the delay, that issue comes from the python setuptools packages, upgrade it to try to fix the issue.

Alejandro-M-Perez, 2024-10-17:
I ran `sudo pip install setuptools --upgrade` and then re-ran `colcon build` but got the same error, is there a specific version of the package I should use?

mgonzs13, 2024-10-17:
@Alejandro-M-Perez, have you tried ```pip3 install setuptools --upgrade```? Which setuptools --upgrade are you using (```pip3 show setuptools --upgrade```)?

CarlosMelloeSouza, 2024-10-17:
i have the same issue, and my setuptools are on the latest version (75.2.0) have something that i can do? like turn it to an older version?

mgonzs13, 2024-10-17:
Try with the following versions: packaging==23.2 setuptools==72.1.0

Alejandro-M-Perez, 2024-10-18:
Sorry for taking so long to get back to you.
I tried `pip3 install setuptools --upgrade` and `sudo pip3 install setuptools --upgrade`. My setuptools version is 75.2.0. As for the setuptools --upgrade I'm using, runing the command you gave returns nothing, but runing `pip3 show setuptools` returns the version of setuptools I have (75.2.0).

Alejandro-M-Perez, 2024-10-18:
Sorry for taking so long to get back to you.

I tried `pip3 install setuptools --upgrade` and `sudo pip3 install setuptools --upgrade`. My setuptools version is 75.2.0. As for the `setuptools --upgrade` I'm using, running the command you gave returns nothing, but running `pip3 show setuptools` returns the version of setuptools I have (75.2.0).

Then I installed packaging==23.2 and setuptools==72.1.0 and that seems to have helped, but when running colcon build i now get:
```
loopking@DESKTOP-2HI7PO0:~/ros2_ws$ colcon build
Starting >>> yolov8_msgs
Finished <<< yolov8_msgs [1.66s]
Starting >>> yolov8_ros
/home/loopking/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:268: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
--- stderr: yolov8_ros
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_ros/task/ament_python/build.py", line 102, in build
    return await extension.build(additional_hooks=additional_hooks)
  File "/usr/lib/python3/dist-packages/colcon_core/task/python/build.py", line 94, in build
    available_commands = await self._get_available_commands(
  File "/usr/lib/python3/dist-packages/colcon_core/task/python/build.py", line 192, in _get_available_commands
    output = await check_output(
  File "/usr/lib/python3/dist-packages/colcon_core/subprocess.py", line 129, in check_output
    assert not rc, f'Expected {args} to pass: {stderr_data}'
AssertionError: Expected ['/usr/bin/python3', '-W', 'ignore:setup.py install is deprecated', 'setup.py', '--help-commands'] to pass: /home/loopking/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py:268: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
Traceback (most recent call last):
  File "/home/loopking/ros2_ws/src/yolov8_ros/yolov8_ros/setup.py", line 5, in <module>
    setup(
  File "/home/loopking/.local/lib/python3.10/site-packages/setuptools/__init__.py", line 108, in setup
    return distutils.core.setup(**attrs)
  File "/home/loopking/.local/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 171, in setup
    ok = dist.parse_command_line()
  File "/home/loopking/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 475, in parse_command_line
    if self.handle_display_options(option_order):
  File "/home/loopking/.local/lib/python3.10/site-packages/setuptools/dist.py", line 921, in handle_display_options
    return _Distribution.handle_display_options(self, option_order)
  File "/home/loopking/.local/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 689, in handle_display_options
    self.print_commands()
  File "/home/loopking/.local/lib/python3.10/site-packages/setuptools/dist.py", line 700, in print_commands
    cmdclass = ep.load()
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/loopking/.local/lib/python3.10/site-packages/sphinx/setup_command.py", line 21, in <module>
    from sphinx.application import Sphinx
  File "/home/loopking/.local/lib/python3.10/site-packages/sphinx/application.py", line 31, in <module>
    from sphinx.config import Config
  File "/home/loopking/.local/lib/python3.10/site-packages/sphinx/config.py", line 21, in <module>
    from sphinx.util import logging
  File "/home/loopking/.local/lib/python3.10/site-packages/sphinx/util/__init__.py", line 41, in <module>
    from sphinx.util.typing import PathMatcher
  File "/home/loopking/.local/lib/python3.10/site-packages/sphinx/util/typing.py", line 37, in <module>
    from types import Union as types_Union
ImportError: cannot import name 'Union' from 'types' (/usr/lib/python3.10/types.py)

---
Failed   <<< yolov8_ros [0.87s, exited with code 1]

Summary: 1 package finished [2.82s]
  1 package failed: yolov8_ros
[... 16 lines cut ...]
    self.print_commands()
  File "/home/loopking/.local/lib/python3.10/site-packages/setuptools/dist.py", line 700, in print_commands
    cmdclass = ep.load()
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
[... cut ...]
  File "/home/loopking/.local/lib/python3.10/site-packages/sphinx/config.py", line 21, in <module>
    from sphinx.util import logging
  File "/home/loopking/.local/lib/python3.10/site-packages/sphinx/util/__init__.py", line 41, in <module>
    from sphinx.util.typing import PathMatcher
  File "/home/loopking/.local/lib/python3.10/site-packages/sphinx/util/typing.py", line 37, in <module>
    from types import Union as types_Union
ImportError: cannot import name 'Union' from 'types' (/usr/lib/python3.10/types.py)
```

mgonzs13, 2024-10-20:
@Alejandro-M-Perez, I've been checking the Python packages and the problem is the sphinx package. Since super-gradients, used in YOLO-NAS, requires sphinx sphinx~=4.0.2, you have to install a more recent version. I've been testing with sphinx==8.0.0 sphinx-rtd-theme==3.0.0. Btw, I've created a Dockerfile for the new incoming version, [here](https://github.com/mgonzs13/yolov8_ros/blob/yolo_ros/Dockerfile) you have.

HesamTaherzadeh, 2024-11-29:
My issue got Resolved after sudo pip install sphinx --upgrade, as mentioned by the author

closed 2025-01-13 by github-actions[bot]