mgonzs13/yolo_ros #53: ImportError: cannot import name 'Union' from 'types'

squarcia, 2024-10-06:
Hi, I'm trying to build a repository, but I keep encountering errors. Do you know if this could be related to the version of setuptools? I'm currently using Python 3.10.12 and setuptools version 58.2.0. I previously had version 70.0, but faced the same issues, so I downgraded to 58.2.0.

I'm not using Conda environments, but I still can't resolve the error. Do you have any suggestions on how to proceed?

colcon build
Starting >>> yolov8_msgs
Finished <<< yolov8_msgs [7.51s]                     
Starting >>> yolov8_ros
--- stderr: yolov8_ros                   
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_ros/task/ament_python/build.py", line 102, in build
    return await extension.build(additional_hooks=additional_hooks)
  File "/usr/lib/python3/dist-packages/colcon_core/task/python/build.py", line 94, in build
    available_commands = await self._get_available_commands(
  File "/usr/lib/python3/dist-packages/colcon_core/task/python/build.py", line 192, in _get_available_commands
    output = await check_output(
  File "/usr/lib/python3/dist-packages/colcon_core/subprocess.py", line 129, in check_output
    assert not rc, f'Expected {args} to pass: {stderr_data}'
AssertionError: Expected ['/usr/bin/python3', '-W', 'ignore:setup.py install is deprecated', 'setup.py', '--help-commands'] to pass: Traceback (most recent call last):
  File "/home/ab2924/Desktop/ros2_ws/src/yolov8_ros/yolov8_ros/setup.py", line 5, in <module>
    setup(
  File "/home/ab2924/.local/lib/python3.10/site-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.10/distutils/core.py", line 134, in setup
    ok = dist.parse_command_line()
  File "/usr/lib/python3.10/distutils/dist.py", line 480, in parse_command_line
    if self.handle_display_options(option_order):
  File "/home/ab2924/.local/lib/python3.10/site-packages/setuptools/dist.py", line 1120, in handle_display_options
    return _Distribution.handle_display_options(self, option_order)
  File "/usr/lib/python3.10/distutils/dist.py", line 683, in handle_display_options
    self.print_commands()
  File "/home/ab2924/.local/lib/python3.10/site-packages/setuptools/dist.py", line 896, in print_commands
    cmdclass = ep.resolve()
  File "/home/ab2924/.local/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2456, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/ab2924/.local/lib/python3.10/site-packages/sphinx/setup_command.py", line 21, in <module>
    from sphinx.application import Sphinx
  File "/home/ab2924/.local/lib/python3.10/site-packages/sphinx/application.py", line 31, in <module>
    from sphinx.config import Config
  File "/home/ab2924/.local/lib/python3.10/site-packages/sphinx/config.py", line 21, in <module>
    from sphinx.util import logging
  File "/home/ab2924/.local/lib/python3.10/site-packages/sphinx/util/__init__.py", line 41, in <module>
    from sphinx.util.typing import PathMatcher
  File "/home/ab2924/.local/lib/python3.10/site-packages/sphinx/util/typing.py", line 37, in <module>
    from types import Union as types_Union
ImportError: cannot import name 'Union' from 'types' (/usr/lib/python3.10/types.py)

---
Failed   <<< yolov8_ros [0.55s, exited with code 1]

Summary: 1 package finished [8.33s]
  1 package failed: yolov8_ros
  1 package had stderr output: yolov8_ros
  1 package not processed
Expected ['/usr/bin/python3', '-W', 'ignore:setup.py install is deprecated', 'setup.py', '--help-commands'] to pass: Traceback (most recent call last):
  File "/home/ab2924/Desktop/ros2_ws/src/yolov8_ros/yolov8_ros/setup.py", line 5, in <module>
    setup(
  File "/home/ab2924/.local/lib/python3.10/site-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.10/distutils/core.py", line 134, in setup
    ok = dist.parse_command_line()
  File "/usr/lib/python3.10/distutils/dist.py", line 480, in parse_command_line
    if self.handle_display_options(option_order):
  File "/home/ab2924/.local/lib/python3.10/site-packages/setuptools/dist.py", line 1120, in handle_display_options
    return _Distribution.handle_display_options(self, option_order)
  File "/usr/lib/python3.10/distutils/dist.py", line 683, in handle_display_options
    self.print_commands()
  File "/home/ab2924/.local/lib/python3.10/site-packages/setuptools/dist.py", line 896, in print_commands
    cmdclass = ep.resolve()
  File "/home/ab2924/.local/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2456, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/ab2924/.local/lib/python3.10/site-packages/sphinx/setup_command.py", line 21, in <module>
    from sphinx.application import Sphinx
  File "/home/ab2924/.local/lib/python3.10/site-packages/sphinx/application.py", line 31, in <module>
    from sphinx.config import Config
  File "/home/ab2924/.local/lib/python3.10/site-packages/sphinx/config.py", line 21, in <module>
    from sphinx.util import logging
  File "/home/ab2924/.local/lib/python3.10/site-packages/sphinx/util/__init__.py", line 41, in <module>
    from sphinx.util.typing import PathMatcher
  File "/home/ab2924/.local/lib/python3.10/site-packages/sphinx/util/typing.py", line 37, in <module>
    from types import Union as types_Union
ImportError: cannot import name 'Union' from 'types' (/usr/lib/python3.10/types.py)

mgonzs13, 2024-10-07:
Hi @squarcia, that seems to be a problem with sphinx, try to upgrade it.

soybeanboy15, 2024-10-12:
I am also getting this error `ImportError: cannot import name 'Union' from 'types' (/usr/lib/python3.10/types.py)` however i am running this in docker here is what my docker looks like, is there anything wrong with my setup :/ ?
```
ARG ROS_DISTRO=humble

# ----- Base Stage -----
FROM ros:${ROS_DISTRO}-ros-base AS base
ENV ROS_DISTRO=${ROS_DISTRO}

# Use bash for the following RUN commands
SHELL ["/bin/bash", "-c"]

# Create directory for the workspace
RUN mkdir -p /walter-hq
WORKDIR /walter-hq

# Install dependencies
RUN apt-get update \
    && apt-get install -y --no-install-recommends \
    python3-pip \
    ros-${ROS_DISTRO}-cv-bridge \
    ros-${ROS_DISTRO}-image-transport

COPY /src/yolov8_ros/ /walter-hq/src/yolov8_ros

RUN pip3 install -r src/yolov8_ros/requirements.txt

# RUN pip install opencv-python==4.8.1.78 \
#                 ultralytics==8.3.3

RUN source /opt/ros/${ROS_DISTRO}/setup.bash

# ------ Deploy Stage -----
FROM base AS deploy
WORKDIR /walter-hq

# Use bash for the following RUN commands
SHELL ["/bin/bash", "-c"]

COPY /src/hq_vision/ /walter-hq/src/hq_vision/
COPY /src/walter_interfaces/ /walter-hq/src/walter_interfaces/

# Source Underlay, Build Workspace
RUN source /opt/ros/${ROS_DISTRO}/setup.bash \
    && rosdep install --from-paths src --ignore-src --rosdistro ${ROS_DISTRO} -y \
    && colcon build --executor sequential

# Set up the entrypoint
COPY ./entrypoint.sh .
RUN chmod +x ./entrypoint.sh

# For dev
WORKDIR /walter-j1

ENTRYPOINT ["./entrypoint.sh"]
```

mgonzs13, 2024-10-13:
Hey @soybeanboy15, I have tested your Dockerfile and, as I have mentioned, upgrading sphinx will fix your error:
```
RUN pip3 install sphinx --upgrade sphinx-rtd-theme
```

mlim97, 2024-10-14:
I updated my dockerfile with the pip update and it now works thanks!

closed 2024-10-14 by squarcia