mgonzs13/yolo_ros #54: Visualizing in Rviz

BingPeng0, 2024-10-29:
Hi. I have managed to run the code and can view it by subscribing to the yolo/dbg_image. But I  wanted to ask how can I visualize the 3D image in the Rviz world like what is shown in the demo.

mgonzs13, 2024-10-29:
@BingPeng0 You can use the marker visualization to show the 3D boxes.

BingPeng0, 2024-10-30:
Sorry I still dont understand as I am quite new in ROS. Do I have to make another node to run the marker visualization or there is already topic being published provided in yolov8 launch file.

mgonzs13, 2024-10-30:
In rviz, using the `Add` button and then the `By topic`, you should see a marker topic. Adding it will visualize the 3D boxes.

YumTaha, 2024-12-03:
I had similar question and I did the marker topic (there is 2, I don't know which one). This is what I see and it keeps blinking:

[image]

YumTaha, 2024-12-04:
How can I check both image and detections rate ?

mgonzs13, 2024-12-05:
> How can I check both image and detections rate ?

Use `ros2 topic hz <your-topic>` to check the rate.

> Regarding visualizing the point cloud
> Both have no topic to subscribe to

You can use the PointCloud2 to visualize the point cloud. If you are not getting topics for PointCloud2, check your realsense, there should be a param to activate the point cloud.

YumTaha, 2024-12-05:
> Use ros2 topic hz <your-topic> to check the rate.

```
iras@iras:~/yolo_ros$ ros2 topic hz /yolo/dgb_bb_markers
WARNING: topic [/yolo/dgb_bb_markers] does not appear to be published yet
average rate: 13.851
	min: 0.003s max: 0.307s std dev: 0.12209s window: 14
average rate: 13.868
	min: 0.002s max: 0.307s std dev: 0.12180s window: 28
average rate: 13.908
	min: 0.002s max: 0.307s std dev: 0.12180s window: 42
average rate: 13.670
	min: 0.002s max: 0.308s std dev: 0.12282s window: 55
^Ciras@iras:~/yolo_ros$ ros2 topic hz /yolo/dgb_kp_markers
WARNING: topic [/yolo/dgb_kp_markers] does not appear to be published yet
average rate: 6.957
	min: 0.004s max: 0.329s std dev: 0.15986s window: 7
average rate: 6.083
	min: 0.004s max: 0.329s std dev: 0.15948s window: 14
average rate: 6.661
	min: 0.004s max: 0.329s std dev: 0.15891s window: 22
average rate: 7.901
	min: 0.004s max: 0.329s std dev: 0.15442s window: 34
^Ciras@iras:~/yolo_ros$ ros2 topic hz /yolo/dbg_image
WARNING: topic [/yolo/dbg_image] does not appear to be published yet
average rate: 5.857
	min: 0.010s max: 0.657s std dev: 0.24255s window: 6
average rate: 4.296
	min: 0.010s max: 0.977s std dev: 0.32154s window: 10
average rate: 6.308
	min: 0.007s max: 0.977s std dev: 0.25222s window: 21
^Ciras@iras:~/yolo_ros$ ros2 topic hz /camera/camera/color/image_raw
WARNING: topic [/camera/camera/color/image_raw] does not appear to be published yet
average rate: 6.776
	min: 0.033s max: 0.433s std dev: 0.16122s window: 7
average rate: 10.326
	min: 0.033s max: 0.433s std dev: 0.11502s window: 21
average rate: 9.814
	min: 0.033s max: 0.834s std dev: 0.15560s window: 36
average rate: 10.127
	min: 0.032s max: 0.834s std dev: 0.14562s window: 51
^Ciras@iras:~/yolo_ros$ 
```

> You can use the PointCloud2 to visualize the point cloud. If you are not getting topics for PointCloud2, check your realsense, there should be a param to activate the point cloud.

Here’s my new command: `ros2 launch realsense2_camera rs_launch.py initial_reset:=true pointcloud.enable:=true` I had to reset the camera because I was encountering an error when the reset wasn’t performed.
[image]

As shown in the image below, the 3D boxes are misplaced and don't align with the colored blocks.
[image]

YumTaha, 2024-12-05:
Thanks, seems to work now

BingPeng0, 2025-01-10:
[image]
There is nothing appear in my Rviz

mgonzs13, 2025-01-13:
@BingPeng0 are you using the 3D version?

BingPeng0, 2025-01-13:
[20250113 - Yolo Marker Array Not Working.webm](https://github.com/user-attachments/assets/58deab6e-2103-416f-82ab-af8d8bf297f6)
Yes

mgonzs13, 2025-01-13:
@BingPeng0 check the `dbg_bb_markers` messages and QoS. What is the world frame, does it exist in your tfs? You can also pull the new changes from this repo.

Razvan2905, 2025-02-08:
Hi.I have a problem with viewing the results.I dont see anything in Rviz. I can see my camera in "camera/image_raw" topic and I changed it in the code instead of "/camera/rgb/image_raw".  also the topic /yolo/dbg_image doesn't publish anything

[image]

mgonzs13, 2025-02-10:
@Razvan2905 how are you launching yolo? Have you checked the QoS of the topics?

Razvan2905, 2025-02-10:
[image]

mgonzs13, 2025-02-11:
@Razvan2905 have you checked the detection topic? Are you sure your camera topic is getting messages?

closed 2025-03-29 by github-actions[bot]