mgonzs13/yolo_ros #56: Custom Yolo model error (Activating yolo_node)

SeokHyunAhn, 2024-11-07:
Hi Hello, I want to use my custom YOLOv8 model with ROS2. I tried using the following code:

`ros2 launch yolo_bringup yolov8.launch.py input_image_topic:="/oak/rgb/image_raw" model:="home/agmo/YOLO/runs/detect/train2/weights/best.pt"`

However, it doesn’t proceed past "Activating." When I used the default model, I was able to visualize it correctly in Rviz. Could you help me identify what might be causing the issue?

[image]
[image]

mgonzs13, 2024-11-07:
Hi @SeokHyunAhn, it seems that Ultralytics is stuck at loading the model. Which Ultralytics version have you used to train the model? Can you share the model to try it?

SeokHyunAhn, 2024-11-07:
Hi @mgonzs13 
Thank you for your reply.
My ultrayltics version is 8.3.10. This is the Yolov8 model I used(best.pt).
[weights.zip](https://github.com/user-attachments/files/17663261/weights.zip)

mgonzs13, 2024-11-07:
I have tested your model and the yolo_node is working. There is a missing `/` at the beginning of the model path, which may be causing this problem. I'll add a check for the model file to the yolo_node.

> ros2 launch yolo_bringup yolov8.launch.py input_image_topic:="/oak/rgb/image_raw" model:="home/agmo/YOLO/runs/detect/train2/weights/best.pt"

closed 2024-11-11 by SeokHyunAhn