mgonzs13/yolo_ros #57: Non-pt yolo model format stucks activation of yolo_node (onnx)

dixiatielu, 2024-11-13:
Everything works well when I executed the demo with `.pt` format models.
```shell
dxtl@dxtl-MATEBOOK:~/ros_ws$ ros2 launch yolo_bringup yolo.launch.py input_image_topic:=/image_raw device:=cpu use_tracking:=False model:='/home/dxtl/ros_ws/yolo11n.pt'
[INFO] [launch]: All log files can be found below /home/dxtl/.ros/log/2024-11-14-00-50-57-639298-dxtl-MATEBOOK-96382
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [yolo_node-1]: process started with pid [96386]
[INFO] [debug_node-2]: process started with pid [96388]
[yolo_node-1] [INFO] [1731516662.434530224] [yolo.yolo_node]: [yolo_node] Configuring...
[debug_node-2] [INFO] [1731516662.450361058] [yolo.debug_node]: [debug_node] Configuring...
[yolo_node-1] [INFO] [1731516662.575930683] [yolo.yolo_node]: [yolo_node] Configured
[yolo_node-1] [INFO] [1731516662.577090035] [yolo.yolo_node]: [yolo_node] Activating...
[debug_node-2] [INFO] [1731516662.595726388] [yolo.debug_node]: [debug_node] Configured
[debug_node-2] [INFO] [1731516662.596516636] [yolo.debug_node]: [debug_node] Activating...
[debug_node-2] [INFO] [1731516662.606315769] [yolo.debug_node]: [debug_node] Activated
[yolo_node-1] [INFO] [1731516663.124036556] [yolo.yolo_node]: [yolo_node] Activated
```
However, when I tried to change the model into `onnx` format, the `yolo_node` no longer run into `activate` state... thus the inference cannot be performed anymore. I am pretty sure both the `onnx` and `pt` model exists in my file system. Plus the model works well with ultralytics's official cli program `yolo predict model=yolo11n.onnx source=0`
```shell
dxtl@dxtl-MATEBOOK:~/ros_ws$ ros2 launch yolo_bringup yolo.launch.py input_image_topic:=/image_raw device:=cpu use_tracking:=False model:='/home/dxtl/ros_ws/yolo11n.onnx'
[INFO] [launch]: All log files can be found below /home/dxtl/.ros/log/2024-11-14-00-59-35-687322-dxtl-MATEBOOK-101408
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [yolo_node-1]: process started with pid [101409]
[INFO] [debug_node-2]: process started with pid [101411]
[debug_node-2] [INFO] [1731517180.711284177] [yolo.debug_node]: [debug_node] Configuring...
[yolo_node-1] [INFO] [1731517180.773805836] [yolo.yolo_node]: [yolo_node] Configuring...
[debug_node-2] [INFO] [1731517180.860582863] [yolo.debug_node]: [debug_node] Configured
[debug_node-2] [INFO] [1731517180.861313325] [yolo.debug_node]: [debug_node] Activating...
[debug_node-2] [INFO] [1731517180.871304711] [yolo.debug_node]: [debug_node] Activated
[yolo_node-1] [INFO] [1731517180.912917257] [yolo.yolo_node]: [yolo_node] Configured
[yolo_node-1] [INFO] [1731517180.914163414] [yolo.yolo_node]: [yolo_node] Activating...
```

Is there any parameter I forgot to add?

dixiatielu, 2024-11-13:
The problem is... `model.fuse()` can only affect `.pt` models, not available to other formats.

https://github.com/mgonzs13/yolo_ros/blob/b7a5d6ff5a356b5b59386f7f5ccf8fc94b4ecef5/yolo_ros/yolo_ros/yolo_node.py#L131

Maybe consider adding a validation before issuing `model.fuse()` ?

mgonzs13, 2024-11-13:
@dixiatielu Hi, I have just added a try/except for the `fuse` call. You can find the changes in d587d433bb51bb284ce3ee02e3feab1495b7b6fb.

dixiatielu, 2024-11-13:
> @dixiatielu Hi, I have just added a try/except for the `fuse` call. You can find the changes in [d587d43](https://github.com/mgonzs13/yolo_ros/commit/d587d433bb51bb284ce3ee02e3feab1495b7b6fb).

Thank you for the quick reply! I've updated to the latest version and the node runs well!

closed 2024-11-13 by dixiatielu