mgonzs13/yolo_ros #61: YOLO 3D Detection Causes Object Recognition Loss and Camera Lag

YumTaha, 2024-12-03:
### What happened?
When I run `ros2 launch yolo_bringup yolov11.launch.py`, everything works fine, and the system can identify objects within the camera's view. However, when I add `use_3d:=True` to the command, YOLO can only detect objects within a specific region. For instance, if I move an object to the bottom of the camera view, YOLO no longer recognizes it, even though it did so without 3D detection enabled.
### Image displayed in RViz2 after running: `ros2 launch yolo_bringup yolov11.launch.py`

[image]
### Image displayed in RViz2 after running: `ros2 launch yolo_bringup yolov11.launch.py use_3d:=True`

[image]

Additionally, changing the position of the blocks sometimes causes the process to crash in 3D mode, with an "out of bounds" error.

Another issue is camera lag. Using the Realsense camera without YOLO works smoothly, but when YOLO is added, the camera becomes laggy, despite **CPU usage remaining below 20%**. Running the Realsense viewer does not cause any lag.

[image]

### ROS2 Topics
Running **without** `use_3d`
```
iras@iras:~$ ros2 topic list
/camera/camera/color/camera_info
/camera/camera/color/image_raw
/camera/camera/color/metadata
/camera/camera/depth/camera_info
/camera/camera/depth/image_rect_raw
/camera/camera/depth/metadata
/camera/camera/extrinsics/depth_to_color
/clicked_point
/goal_pose
/initialpose
/parameter_events
/rosout
/tf
/tf_static
/yolo/dbg_image
/yolo/debug_node/transition_event
/yolo/detections
/yolo/dgb_bb_markers
/yolo/dgb_kp_markers
/yolo/tracking
/yolo/tracking_node/transition_event
/yolo/yolo_node/transition_event
```
Running **with** `use_3d`
```
iras@iras:~$ ros2 topic list
/camera/camera/color/camera_info
/camera/camera/color/image_raw
/camera/camera/color/metadata
/camera/camera/depth/camera_info
/camera/camera/depth/image_rect_raw
/camera/camera/depth/metadata
/camera/camera/extrinsics/depth_to_color
/clicked_point
/goal_pose
/initialpose
/parameter_events
/rosout
/tf
/tf_static
/yolo/dbg_image
/yolo/debug_node/transition_event
/yolo/detect_3d_node/transition_event
/yolo/detections
/yolo/detections_3d
/yolo/dgb_bb_markers
/yolo/dgb_kp_markers
/yolo/tracking
/yolo/tracking_node/transition_event
/yolo/yolo_node/transition_event
```
[... 71 lines cut ...]
[detect_3d_node-3] IndexError: index 484 is out of bounds for axis 0 with size 480
[ERROR] [detect_3d_node-3]: process has died [pid 237124, exit code 1, cmd '/home/iras/yolo_ros/install/yolo_ros/lib/yolo_ros/detect_3d_node --ros-args -r __node:=detect_3d_node -r __ns:=/yolo --params-file /tmp/launch_params_t07wx2jk -r depth_image:=/camera/camera/depth/image_rect_raw -r depth_info:=/camera/camera/depth/camera_info -r detections:=tracking'].
```

```
iras@iras:~$ ros2 launch yolo_bringup yolov11.launch.py
[INFO] [launch]: All log files can be found below /home/iras/.ros/log/2024-12-03-15-50-22-000831-iras-263092
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [yolo_node-1]: process started with pid [263094]
[INFO] [tracking_node-2]: process started with pid [263096]
[INFO] [debug_node-3]: process started with pid [263098]
[yolo_node-1] [INFO] [1733259023.199205878] [yolo.yolo_node]: [yolo_node] Configuring...
[tracking_node-2] [INFO] [1733259023.226350960] [yolo.tracking_node]: [tracking_node] Configuring...
[yolo_node-1] [INFO] [1733259023.228191795] [yolo.yolo_node]: [yolo_node] Configured
[yolo_node-1] [INFO] [1733259023.228345873] [yolo.yolo_node]: [yolo_node] Activating...
[debug_node-3] [INFO] [1733259023.247136625] [yolo.debug_node]: [debug_node] Configuring...
[yolo_node-1] [INFO] [1733259023.254280198] [yolo.yolo_node]: Trying to fuse model...
[tracking_node-2] [INFO] [1733259023.259252221] [yolo.tracking_node]: [tracking_node] Configured
[tracking_node-2] [INFO] [1733259023.259426333] [yolo.tracking_node]: [tracking_node] Activating...
[tracking_node-2] [INFO] [1733259023.260901327] [yolo.tracking_node]: [tracking_node] Activated
[debug_node-3] [INFO] [1733259023.281731796] [yolo.debug_node]: [debug_node] Configured
[debug_node-3] [INFO] [1733259023.281905077] [yolo.debug_node]: [debug_node] Activating...
[debug_node-3] [INFO] [1733259023.283609896] [yolo.debug_node]: [debug_node] Activated
[yolo_node-1] [INFO] [1733259023.374951668] [yolo.yolo_node]: [yolo_node] Activated
```

mgonzs13, 2024-12-04:
@YumTaha I'll check the error with the out of bounds. Do you have a bag to test with your same case?

> Another issue is camera lag. Using the Realsense camera without YOLO works smoothly, but when YOLO is added, the camera becomes laggy, despite CPU usage remaining below 20%. Running the Realsense viewer does not cause any lag.

This may be an issue related to the QoS of the topics or the DDS. Check if the image topic publishers and subscribers have the same QoS and mainly the same reliability.

> You may notice some topics that aren't part of the default setup for this repository, such as /camera/camera/.... Originally, these topics were /camera/..., but Realsense updated them to the new [...]

You can change the topics without modifying the launch files by using the launch args:

```bash
ros2 launch yolo_bringup yolov11.launch.py input_image_topic:="your topic" image_reliability:="you reliability"
```

mgonzs13, 2024-12-04:
@YumTaha for the out of bounds error, are both the color and depth images of the same size?

> For instance, if I move an object to the bottom of the camera view, YOLO no longer recognizes it, even though it did so without 3D detection enabled. 

In the 3D mode, objects that are not detected in the depth images are not shown in both the 2D image and the 3D boxes.

YumTaha, 2024-12-04:
Hey @mgonzs13, thanks for the quick reply. I wanted to let you know I just started learning YOLO and ROS2, I don't have deep knowledge about them, I might ask some `stupid` questions sometimes lol 

> Do you have a bag to test with your same case?

what do you mean by that?

> This may be an issue related to the QoS of the topics or the DDS. Check if the image topic publishers and subscribers have the same QoS and mainly the same reliability. Which DDS ar you using?

I don't know what is DDS but in the launch file both `depth_info_reliability` and `image_reliability` are both on 2 `Best Effort`
 

> You can change the topics without modifying the launch files by using the launch args:

Is it better to change it in hard-coded or better in launch arguments

> are both the color and depth images of the same size?

I know from launch file that the image is an 640x480, I don't know where to check the color image and depth image seperatly

mgonzs13, 2024-12-05:
> what do you mean by that?

A bag is a data-structure that stores recorded data from topics. Check the [ROS 2 doc](https://docs.ros.org/en/foxy/Tutorials/Beginner-CLI-Tools/Recording-And-Playing-Back-Data/Recording-And-Playing-Back-Data.html).

> I don't know what is DDS but in the launch file both depth_info_reliability and image_reliability are both on 2 Best Effort

Yes, by default is set to 2 that is Best Effort. You can check the QoS of your camera topics by using `ros2 topic info -v <your-camera-topic>`. Both, the publisher and subscriber should have the same reliability.

> How to check DDS?

[DDS](https://design.ros2.org/articles/ros_on_dds.html) By default, Humble uses [Fast DDS](https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/ros2.html). You cna try the [Cyclone DDS](https://docs.ros.org/en/humble/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html).

> Is it better to change it in hard-coded or better in launch arguments

Using launch args improves the flexibility and scalability.

> I know from launch file that the image is an 640x480, I don't know where to check the color image and depth image seperatly

That size is the expected image size not the real image size. You can check the size by using `ros2 topic echo <your-image-topic>`.

YumTaha, 2024-12-05:
### UPDATE
With the assistance of a coworker experienced in ROS2, we discovered that the camera and depth sensor had different dimensions.

```
iras@iras:~$ ros2 topic echo /camera/camera/color/camera_info
header:
  stamp:
    sec: 1733418584
    nanosec: 565581787
  frame_id: camera_color_optical_frame
height: 720
width: 1280
...
iras@iras:~$ ros2 topic echo /camera/camera/depth/image_rect_raw
header:
  stamp:
    sec: 1733419247
    nanosec: 180940186
  frame_id: camera_depth_optical_frame
height: 480
width: 848
encoding: 16UC1
is_bigendian: 0
step: 1696
```
We attempted to adjust the settings, but they didn't seem to take effect (possibly being overwritten elsewhere)
```
iras@iras:~/yolo_ros$ ros2 launch realsense2_camera rs_launch.py initial_reset:=true pointcloud.enable:=true depth_module.infra_profile:=640x480x30 rgb_camera.color_profile:=1280x720x30
[INFO] [launch]: All log files can be found below /home/iras/.ros/log/2024-12-05-12-30-58-047578-iras-514035
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [514036]
[realsense2_camera_node-1] [INFO] [1733419858.188824660] [camera.camera]: RealSense ROS v4.55.1
[realsense2_camera_node-1] [INFO] [1733419858.188872538] [camera.camera]: Built with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1733419858.188889709] [camera.camera]: Running with LibRealSense v2.55.1
[realsense2_camera_node-1] [INFO] [1733419858.214015686] [camera.camera]: Device with serial number 215222075579 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1733419858.214036258] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1733419858.214040699] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1733419858.214122482] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1733419858.214126536] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1733419858.214129395] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1733419864.238039080] [camera.camera]: Device with serial number 215222075579 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1733419864.238077484] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1733419864.238080821] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1733419864.238166334] [camera.camera]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO] [1733419864.238175364] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1733419864.238210074] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1733419864.238432362] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1733419864.238446442] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1733419864.238449845] [camera.camera]: Device Serial No: 215222075579
[realsense2_camera_node-1] [INFO] [1733419864.238452516] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1733419864.238455497] [camera.camera]: Device FW version: 5.16.0.1
[realsense2_camera_node-1] [INFO] [1733419864.238457849] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1733419864.238460297] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1733419864.323578525] [camera.camera]: Set ROS param depth_module.depth_profile to default: 848x480x30
[realsense2_camera_node-1] [WARN] [1733419864.323855664] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1733419864.327310934] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1733419864.327328041] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1733419864.329052622] [camera.camera]: Stopping Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1733419864.329083048] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1733419864.330536429] [camera.camera]: Starting Sensor: Depth Module
[realsense2_camera_node-1] [INFO] [1733419864.332234894] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
[realsense2_camera_node-1] [INFO] [1733419864.332633112] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1733419864.335067563] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1733419864.335662943] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1733419864.584429957] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
```
Then, we tried aligning them
`ros2 launch realsense2_camera rs_launch.py initial_reset:=true pointcloud.enable:=true align_depth.enable:=true`
[... 41 lines cut ...]
[realsense2_camera_node-1] [WARN] [1733420041.419279968] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
```
New added topics
```
[... cut ...]
[INFO] [1733412686.281167133] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1733412689.994556502] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1733419203.984723365] [rviz]: Message Filter dropping message: frame 'camera_depth_optical_frame' at time 1733419203.449 for reason 'discarding message because the queue is full'
[INFO] [1733419204.016809804] [rviz]: Message Filter dropping message: frame 'camera_depth_optical_frame' at time 1733419203.482 for reason 'discarding message because the queue is full'
(... goes on spam)
```

YumTaha, 2024-12-05:
### Here is a video preview
It's an improvement over the previous version, but it's still not ideal for practical use, especially since I'll be trying to locate these boxes and pick them up with a robotic arm.
[Screencast from 12-05-2024 12:53:05 PM.webm](https://github.com/user-attachments/assets/8cbfa0d3-fdef-4a9b-8aba-f309fe2c6643)

YumTaha, 2024-12-05:
I encountered a new issue when using the 3D node `use_3d:=True`. After some time of running YOLO _(2 min or more)_, one of the detected boxes starts blinking _(only one and usually different, not always same block every run)_, which is odd because this never happened with the 2D detection only `use_3d:=False`. However, when I enable 3D detection, the issue occurs on both 2D and 3D. You might assume that the problem lies with my model, but this was tested using both 2D and 3D detection, and the blinking issue only occurs with 3D detection. I never experienced blinking (only one box at a time) with 2D detection, so the issue is definitely related to the 3D detection.

### Here is the recordings and the ros bags:

- 3D node is on

command:
`ros2 launch yolo_bringup yolov11.launch.py use_3d:=True`
[Screencast from 12-05-2024 05:55:36 PM.webm](https://github.com/user-attachments/assets/76afd466-46d0-49e3-98ea-2784f9a83f1c)
Recorded bag: [recording_2](https://file.io/LAsR2wH547Dx)

- 2D node is on

command:
`ros2 launch yolo_bringup yolov11.launch.py`
[Screencast from 12-05-2024 05:57:04 PM.webm](https://github.com/user-attachments/assets/f4f48861-d69e-4567-92e0-6f51176663fe)
Recorded bag: [recording_3](https://file.io/RruMqqwS7KWq)

mgonzs13, 2024-12-06:
Since the 3D boxes are calculated using the 2D box and the depth image, there may be errors with the depth when the 2D boxes do not match the objects properly. You can use segmentation mask models to try to fix it. In this case, the mask is used to create the 3D boxes, which should improve the process.

About the bags, can you include the `/tf` and `/tf_static` topics? They are used in the 3D node.

YumTaha, 2024-12-06:
I will try to use segmentation masks Monday, I don't know much about it but I will do some research in the meantime.
Here are the new bags and their recordings:
(they are same as the ones I did previously, just added the new topic you wanted.

https://github.com/user-attachments/assets/d530202f-aab5-473c-99f9-3c8c68f90ec3

[recording_1](https://file.io/HM7n1AcG3Igu)

https://github.com/user-attachments/assets/9abc5050-5466-46a8-97a5-39ae84f2f29a

[recording_2](https://file.io/U3bbfH67I0VO)

https://github.com/user-attachments/assets/9bb2450a-6c61-4b7c-883c-f90e7b33e401

[recording_3](https://file.io/ZsGIGQITfjz8)

YumTaha, 2024-12-08:
> Since the 3D boxes are calculated using the 2D box and the depth image, there may be errors with the depth when the 2D boxes do not match the objects properly. You can use segmentation mask models [...]

Segmentation didn't help with the stretching of the 3d bounding as the cube gets raised.
I trained a new model to do segmentation masks:
model ([best.pt](https://file.io/WVMdf79cRyMO))

This is the recording with the new model:

https://github.com/user-attachments/assets/1ec86c81-3f99-46fe-8ab9-cccc1146ffda

**Node: I ran into an error that only restarting my computer fixed it and I didn't know how to fix it, you're more knowledgeable than me so you'd probably seen this issue before:**

https://github.com/user-attachments/assets/1415d6b4-465d-4cc6-a7f3-5a4a1692d853

### My YOLO launch files:

**yolov11.launch.py**
```
import os
from launch import LaunchDescription
from launch.substitutions import LaunchConfiguration
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from ament_index_python.packages import get_package_share_directory

def generate_launch_description():

    return LaunchDescription(
        [
            IncludeLaunchDescription(
                PythonLaunchDescriptionSource(
                    os.path.join(
                        get_package_share_directory("yolo_bringup"),
                        "launch",
                        "yolo.launch.py",
                    )
                ),
                launch_arguments={
                    "model": LaunchConfiguration("model", default="/home/iras/best.pt"),
                    "tracker": LaunchConfiguration("tracker", default="bytetrack.yaml"),
                    "device": LaunchConfiguration("device", default="cuda:0"),
                    "enable": LaunchConfiguration("enable", default="True"),
                    "threshold": LaunchConfiguration("threshold", default="0.8"),
                    "input_image_topic": LaunchConfiguration(
                        "input_image_topic", default="/camera/camera/color/image_raw"
                    ),
                    "image_reliability": LaunchConfiguration(
                        "image_reliability", default="1"
                    ),
                    "namespace": LaunchConfiguration("namespace", default="yolo"),
                }.items(),
            ),

            IncludeLaunchDescription(
                PythonLaunchDescriptionSource(
                    os.path.join(
                        get_package_share_directory('realsense2_camera'),
                        'launch',
                        'rs_launch.py',
                    )
                ),
                launch_arguments={
                    'camera_name': 'camera',
                    'initial_reset': 'true',
                    'pointcloud.enable': 'true',
                    'align_depth.enable': 'true',
                }.items()
            )
    
        ]
[... 305 lines cut ...]
            debug_node_cmd,
            yolo_rviz2_node,
        )

    use_tracking = LaunchConfiguration("use_tracking")
    use_tracking_cmd = DeclareLaunchArgument(
        "use_tracking", default_value="True", description="Whether to activate tracking"
    )

    use_3d = LaunchConfiguration("use_3d")
    use_3d_cmd = DeclareLaunchArgument(
        "use_3d", default_value="False", description="Whether to activate 3D detections"
    )
    
    
    return LaunchDescription(
        [
            use_tracking_cmd,
            use_3d_cmd,
            OpaqueFunction(function=run_yolo, args=[use_tracking, use_3d]),
        ]
    )
```

**NOTE: I’ve updated the YOLOv11 and YOLO launch files to include RViz and RealSense, so I no longer need to run them in separate windows each time I use YOLO. This is temporary, as I plan to integrate YOLO without requiring RViz and RealSense to be specified in the YOLO launch file.**

YumTaha, 2024-12-09:
[Screencast from 12-08-2024 11:03:41 PM.webm](https://github.com/user-attachments/assets/08953971-b4e7-4003-bdfc-26aa6d09e0fb)

I looked into 3d node but couldn't figure out exactly how convert bb to 3d method calculates all the dimensions, but this video was interesting.

YumTaha, 2024-12-09:
> @YumTaha your [recording_1](https://file.io/HM7n1AcG3Igu) is deleted.

Try one of these links, I couldn't find a good file sharing system
[recording_1](https://filebin.net/2w60u5jvudlsrrjh) or [recording_1](https://file.io/JJYHBg6V90In) or [recording_1](https://www.filemail.com/d/byuwihknpcmhsqa)
(if you know a better way of sharing files let me know)

> Try to reduce the maximum_detection_threshold. You can try values near your object box size.

Tried from 0.1 to 0.9 values of the `maximum_detection_threshold`, it doesn't change nothing.
I was wondering, since you created this GitHub, did you test it on your computer, and did it work? If so, what do you think could be causing the issue on my end? Specifically, which part of the code might be responsible for the stretching effect to the ground so I can investigate further?

YumTaha, 2024-12-10:
> The RGB camera image topic is missing to reproduce your output fully.

here is a recording of all topics (5s) [recording_alltopics](https://www.filemail.com/d/wqcxxoovrwbjlle)

mgonzs13, 2024-12-10:
No `/tf` topics this time. I need `/tf`, `/tf_static`, your RGB image topic, depth image topic and camera info topic.

YumTaha, 2024-12-10:
here [recording](https://filebin.net/vije7f34kx6yxtt4)

mgonzs13, 2024-12-10:
I've been testing your segmentation model. With 0.05 of `maximum_detection_threshold` seems to work. Here you have the command I am using:

```bash
ros2 launch yolo_bringup yolo.launch.py use_3d:=True input_image_topic:="/camera/camera/color/image_raw" image_reliability:=1 input_depth_topic:="/camera/camera/aligned_depth_to_color/image_raw" depth_image_reliability:=1 input_depth_info_topic:="/camera/camera/aligned_depth_to_color/camera_info" depth_info_reliability:=1 model:="../Downloads/best.pt" target_frame:="camera_link" maximum_detection_threshold:=0.05
```

However, it seems that there are still errors with the depth.

YumTaha, 2024-12-10:
The threshold has helped fix the stretching issues, though not completely. However, it's enough for me to continue working on the rest. Thank you for taking the time every day to help resolve that issue.
Another thing—it's not an issue, just a possible feature—I'm wondering if the segmentation model also includes the color of the boudingbox (instead of just id). Since I use RoboFlow to train my models, I can choose the color for each detected object and train it accordingly. However, it seems that in your original code, a random color is assigned to each boundingbox instead of using the color the model was trained on. If possible, could the model retain that attribute?

mgonzs13, 2024-12-10:
> Another thing—it's not an issue, just a possible feature—I'm wondering if the segmentation model also includes the color of the boudingbox (instead of just id). Since I use RoboFlow to train my [...]

I don't know if the colors are also downloaded when using RoboFlow. Another option is to use the color palette of Ultralytics.

closed 2024-12-13 by YumTaha