mgonzs13/yolo_ros #63: Rviz2 not displaying the camera

prakash-aryan, 2024-12-09:
I have a zed camera and I am running the yolo nodes through docker

[image]
[image]
[image]
[image]
[image]

mgonzs13, 2024-12-10:
@prakash-aryan the `ros2 launch yolo_bringup yolo.launch.py` command is using default topics and params. You have to use your camera topics, for example, `ros2 launch yolo_bringup yolov5.launch.py input_image_topic:=<your topic>`. Take a look at the README to see the other [launch arguments](https://github.com/mgonzs13/yolo_ros?tab=readme-ov-file#parameters).

prakash-aryan, 2024-12-11:
Yes it works now. 
Thanks a lot

I used - > 

` ros2 launch yolo_bringup yolo.launch.py input_image_topic:=/zed_camera/rgb/left_image use_debug:=True device:=cpu`

closed 2024-12-11 by prakash-aryan

Johnny-Hu-406, 2025-03-02:
When executing `ros2 topic echo /yolo/detections`, I can see the following output, and `ros2 topic echo /yolo/tracking` also has similar output.

[image]