mgonzs13/yolo_ros #65: Issue with YOLOV11 Node Output

besma14, 2024-12-19:
Hello,

I am using a Basler camera with ROS2, and its topic is configured as /cam0, /cam1 ... . I launch the YOLO node with the following command using the GPU: ros2 launch yolo_bringup yolov11.launch.py input_image_topic:="/cam1" 
However, I do not receive any detection output on topics like /yolo/dbg_image or /yolo/detections. The camera is functioning correctly and publishing images on /cam0.

Could you please assist me in diagnosing and resolving this issue?

i have ROS2 Humble, ubuntu 22.04, i used the dockerfile 

Thank you in advance for your support.

[image]

mgonzs13, 2024-12-19:
@besma14 can you check the QoS of the camera topics? Both, the publisher and subscriber must have the same QoS.

besma14, 2024-12-20:
hi, I looked at the QoS as you mentioned as you can see in the screenshots and I still don't have an image
[image]
[image]
[image]

besma14, 2024-12-20:
I changed the Qos to RELIABLE but I still don't have my image, please help me 
[image]

[image]

[image]

Phance, 2025-01-02:
@besma14 I also want to use Basler camera to realize YOLO real-time detection, can you help me solve some problems? How can I use the Basler camera to achieve detection in Rviz2

mgonzs13, 2025-01-06:
@Phance you'll have to use the ROS 2 package for your camera (https://github.com/basler/pylon-ros-camera). yolo_ros only does not provide code for cameras.

closed 2025-02-20 by github-actions[bot]