mgonzs13/yolo_ros #67: Visualization problem

racket405, 2024-12-23:
After running colcon, I launched yolo_node with `ros2 launch yolo_bringup yolo.launch.py`，but there was no troubleshootings nor rviz2 launching, like this:
[image] \
How can I solve this, please?
And how can I use the model I trained myself instead of Ultralytics'?

mgonzs13, 2024-12-23:
@racket405 the `ros2 launch yolo_bringup yolo.launch.py` command uses the default values, which includes the default model. You have to set your model as `ros2 launch yolo_bringup yolo.launch.py model:=<YOUR MODEL>`.

On the other hand, in you to use rviz2 you have to run rviz2 ``ros2 run rviz2 rviz2` or `rviz2`.

Phance, 2024-12-24:
> @racket405该`ros2 launch yolo_bringup yolo.launch.py`命令使用默认值，其中包括默认模型。您必须将您的模型设置为`ros2 launch yolo_bringup yolo.launch.py model:=<YOUR MODEL>`。
> 
> 另一方面，要使用 rviz2，您必须运行 rviz2“ros2 run rviz2 rviz2 `or`rviz2”。

[image]
[image]
 I launched yolo_node with ros2 launch yolo_bringup yolo.launch.py model:=yolov8m-pose.pt，and I run rviz2. But it still does not display in Rviz2, and there are no detection indications.
 How can I solve this, please?

racket405, 2024-12-24:
> @racket405 the `ros2 launch yolo_bringup yolo.launch.py` command uses the default values, which includes the default model. You have to set your model as `ros2 launch yolo_bringup yolo.launch.py model:=<YOUR MODEL>`.
> 
> On the other hand, you to use rviz2 you have to run rviz2 ``ros2 run rviz2 rviz2`or`rviz2`.

Thanks for telling me how to load my own model, should I put my model in a specified folder or just replace `<YOUR MODEL>` with the path of model? And sadly, the former question is not solved after I ran rviz2. My hardware and software settings are below:
1. Hardware: Jetson AGX Xavier 64GB, Orbbec_camera Gemini2
2. Software: Ubuntu 22.04, ROS humble
Looking forward to you quick reply!

mgonzs13, 2024-12-24:
If you are not getting detections you should check if the nodes are subscribed to the camera topics and the QoS is correct. Check the [launch params](https://github.com/mgonzs13/yolo_ros?tab=readme-ov-file#parameters) to change both topics and reliability for each topic.

Phance, 2024-12-24:
@racket405 @mgonzs13 
[image]
I noticed that the camera-related topics are present in your published topics, but after running ros2 topic list, I only see the following topics. How did you run it?
How can the camera topics be correctly published, and how can the detection images be displayed in Rviz2?
I sincerely hope to receive your reply. Thank you!

mgonzs13, 2024-12-24:
@Phance the camera topics depend on your camera and you have to run the node or launch for it.

racket405, 2024-12-25:
@Phance I used Orbbec Gemini2 camera, all the topics is set by the manufacturer, I changed nothing. You should download your camera's SDK if it has one.
@mgonzs13 I noticed two differences of the parameters of the code `yolo.launch.py` and I changed it (annotated the original code) but there was still nothing showed in my Rviz2. 
[image]
[image]
And then I ran ` ros2 node info /yolo/yolo_node` . Indeed my camera_image topic was not subscribed, but I'm not sure this is the very node or topic that should subscribe the image topic, can you tell me, please? I will try my Realsense D435i camera to exclude hardware incompatibility.
[image]

racket405, 2024-12-25:
> @Phance I used Orbbec Gemini2 camera, all the topics is set by the manufacturer, I changed nothing. You should download your camera's SDK if it has one. @mgonzs13 I noticed two differences of the parameters of the code `yolo.launch.py` and I changed it (annotated the original code) but there was still nothing showed in my Rviz2. ![Screenshot from 2024-12-24 21-02-24](https://private-user-images.githubusercontent.com/82014331/398498821-471ce028-213c-417d-a8b2-51f384f40f03.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUwOTQwNzAsIm5iZiI6MTczNTA5Mzc3MCwicGF0aCI6Ii84MjAxNDMzMS8zOTg0OTg4MjEtNDcxY2UwMjgtMjEzYy00MTdkLWE4YjItNTFmMzg0ZjQwZjAzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjI1VDAyMjkzMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg4ZmNjOTMzNDg0NGNlMjc2ZDRjMTcyMDJmMjRjYTAxZjY3OTJmOTc3ZjkwYjEwMzhhMTQyNTMzZDZhZGU4YmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.9a3lkXey2ZPMfRp38yMYYevngnwOmwINAXYcuHlad6Q) ![Screenshot from 2024-12-24 21-01-37](https://private-user-images.githubusercontent.com/82014331/398498920-394799d5-db3d-4cdd-a536-77321cabe7ba.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUwOTQwNzAsIm5iZiI6MTczNTA5Mzc3MCwicGF0aCI6Ii84MjAxNDMzMS8zOTg0OTg5MjAtMzk0Nzk5ZDUtZGIzZC00Y2RkLWE1MzYtNzczMjFjYWJlN2JhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjI1VDAyMjkzMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVhZTI5NTJlOGJkMmUyYzFiNjE1MjM5ZjgwZDkxMjZmMjFlMzAyOTkxNjRmZGJmMzA4ZTg5NGFlZTk5YmRiOWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.inkMAG6twY3gvMvZtfjEmaI-IVrVh2YnreTqXNMLUtU) And then I ran ` ros2 node info /yolo/yolo_node` . Indeed my camera_image topic was not subscribed, but I'm not sure this is the very node or topic that should subscribe the image topic, can you tell me, please? I will try my Realsense D435i camera to exclude hardware incompatibility. ![Screenshot from 2024-12-24 21-11-05](https://private-user-images.githubusercontent.com/82014331/398499345-b62ff165-8e65-4b40-9625-013126289191.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUwOTQwNzAsIm5iZiI6MTczNTA5Mzc3MCwicGF0aCI6Ii84MjAxNDMzMS8zOTg0OTkzNDUtYjYyZmYxNjUtOGU2NS00YjQwLTk2MjUtMDEzMTI2Mjg5MTkxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjI1VDAyMjkzMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlhMTI5YzQzYjk2YmFlMTgxZWQyOTBlZGJmNzJhNTQ5YjkyMzNjMzJmN2NlMzBiNGJjODIwNzIzZGIyYTAyNTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.mFFvcRWUwQUlef15nijXA0KgaJyxvIw8nAluOzOIm6k)

I tried my D435 camera, and it showed new problem: CUDA is not available. Now I am installing new Pytorch, but I am concerning that my CUDA and Pytorch version is too high to use this package. My CUDA version is 12.6.68 and its pytorch version is 24.06 [install pytorch on jeston](https://docs.nvidia.com/deeplearning/frameworks/install-pytorch-jetson-platform/index.html#overview__section_z4r_vjd_v2c). Is it okay?

mgonzs13, 2024-12-25:
@racket405 check if torch can use CUDA

```python
>>> import torch
>>> torch.cuda.is_available()
True
>>> torch.cuda.device_count()
1
```

Btw, you should use launch arguments instead of modifying the launch file. Check all [params](https://github.com/mgonzs13/yolo_ros?tab=readme-ov-file#parameters).

```bash
ros2 launch yolo_bringup yolo.launch.py model:=<YOUR MODEL> input_image_topic:=<YOUR TOPIC> image_reliability:=<TOU RELIABILITY>
```

racket405, 2024-12-25:
> @racket405 check if torch can use CUDA
> 
> ```python
> >>> import torch
> >>> torch.cuda.is_available()
> True
> >>> torch.cuda.device_count()
> 1
> ```
> 
> Btw, you should use launch arguments instead of modifying the launch file. Check all [params](https://github.com/mgonzs13/yolo_ros?tab=readme-ov-file#parameters).
> 
> ```shell
> ros2 launch yolo_bringup yolo.launch.py model:=<YOUR MODEL> input_image_topic:=<YOUR TOPIC> image_reliability:=<TOU RELIABILITY>
> ```

Yeah, I tried this, but there were some problems with compatibility of Jetson and Pytorch, my JetPack version is too high, I can't find corresponding torchvision. Anyway, I ran this code on CPU and it worked
[image]
But due to the absence of Pytorch acceleration, the FPS was so low that  I couldn't use it for real-time detection. I will try my best to install compatible pytorch and torchvision so that I can use CUDA. I suppose there will no other problems. Thanks for your instruction.

racket405, 2024-12-26:
Okay, I solved ALL MY PROBLEMS!!! Now I use CUDA to  run this code. Actually there previous failures were due to pytorch and torchvision incompatibility. My jetpack is 6.1, which is respectively high version causing many problems. I recommend use lower version for stability. But I still decide to share my setting steps:
1. Download CUDA on jetson refering to https://blog.csdn.net/pleasescanf/article/details/133189245
2. Download pytorch and torchvision refering to [nvidia jetson pytorch 2.4.0 and torchvision 0.19.0 support](https://forums.developer.nvidia.com/t/jetpack6-1-install-torchvision/310007/9)
NOTICE: you must be sure about that the versions of pytorch, CUDA and torchvision are compatible!
3. run `ros2 run yolo_bringup yolo.launch.py` with params specified
You will find using GPU is way more faster than using CPU

closed 2024-12-26 by racket405

Phance, 2025-01-02:
@racket405 I want to use my camera to implement detection in Rviz2, what parameters do I need to change? How do you do that? May I ask you? We look forward to receiving your reply. Thank you!
[image]

mgonzs13, 2025-01-06:
@Phance, launch your camera, which is not included in this repo since this yolo_ros only provides ROS 2 nodes for YOLO. You'll have to install the ROS 2 packages for your camera, if exit.