mgonzs13/yolo_ros #69: Train models and datasets

Phance, 2025-01-03:
@mgonzs13 Can I create my own data set and then train the model, and finally run the trained model with this?How can I specifically train my own model?I hope you can give me some advice and some starting points. Thank you very much!

mgonzs13, 2025-01-06:
Hi, @Phance to train a custom model you have to use `ultralytics`. Take a look at its docs https://docs.ultralytics.com/es/modes/train/#usage-examples.

Phance, 2025-01-11:
@mgonzs13 Hi ! After training the model with my custom dataset, how can I use this yolo_ros to run and perform detection? What format should I use to save the final model? I am really looking forward to your reply, thank you very much.

mgonzs13, 2025-01-13:
@Phance you can use pt format. Then, pass the file path to the launch using the model argument.

liamcarlos, 2025-01-28:
Hey @Phance did you figure out how to use you custom model set?  I've trained my own model and successfully used it with this package and found it as an incredibly useful tool for my project, so feel free to ask me anything if you need help!

closed 2025-03-14 by github-actions[bot]

Phance, 2025-05-02:
@liamcarlos Hello, will there be lag when running your own model, and can it be detected in real time?

liamcarlos, 2025-05-05:
@Phance Not from my experience..  Is your model lagging?