mgonzs13/yolo_ros #71: Define id

racket405, 2025-01-07:
Previously, I successfully ran `yolo.launch.py` and published `/yolo/detections` topic, but when I ran `ros2 topic echo /yolo/detections`, I found the `id` of feedback is empty. How can I change my settings to show id, this means a lot to me.
[image]

mgonzs13, 2025-01-07:
Hi @racket405, which topic are you showing in your image? The tracking ID data, if tracking is activated, is published in `ros2 topic echo /yolo/tracking`.

racket405, 2025-02-10:
> Hi [@racket405](https://github.com/racket405), which topic are you showing in your image? The tracking ID data, if tracking is activated, is published in `ros2 topic echo /yolo/tracking`.

I didn't show /yolo/tracking, instead, I showed /yolo/detections, how can I activate tracking, please？

mgonzs13, 2025-02-10:
Setting `use_tracking` to `True` should activate tracking. How are you launching yolo? Have you changed anything in the code or the launch files?

racket405, 2025-02-12:
I didn't change anything in the code and used default value of  `use_tracking`. But there was no id shown. By the way, can I change `tracking_node.py` to use different tracker instead of bytetrack?

mgonzs13, 2025-02-12:
Check also if the tracking node is running. You can use other ultralytics trackers like botsort.

racket405, 2025-02-13:
OK，I echoed `/yolo/tracking` instead of `/yolo/detections`, and I got ids. Thanks for your help!

closed 2025-03-29 by github-actions[bot]