mgonzs13/yolo_ros #72: GPU execution failed

Phance, 2025-01-07:
[image]

[yolo_node-1]   File "/home/hance/.local/lib/python3.10/site-packages/ultralytics/engine/predictor.py", line 306, in setup_model
[yolo_node-1]     device=select_device(self.args.device, verbose=verbose),
[yolo_node-1]   File "/home/hance/.local/lib/python3.10/site-packages/ultralytics/utils/torch_utils.py", line 192, in select_device
[yolo_node-1]     raise ValueError(
[yolo_node-1] ValueError: Invalid CUDA 'device=gpu' requested. Use 'device=cpu' or pass valid CUDA device(s) if available, i.e. 'device=0' or 'device=0,1,2,3' for Multi-GPU.
[yolo_node-1] 
[yolo_node-1] torch.cuda.is_available(): False
[yolo_node-1] torch.cuda.device_count(): 0
[yolo_node-1] os.environ['CUDA_VISIBLE_DEVICES']: None
[yolo_node-1] See https://pytorch.org/get-started/locally/ for up-to-date torch install instructions if no CUDA devices are seen by torch.
[yolo_node-1] 
[ERROR] [yolo_node-1]: process has died [pid 5313, exit code 1, cmd '/home/hance/ros2_ws/install/yolo_ros/lib/yolo_ros/yolo_node --ros-args -r __node:=yolo_node -r __ns:=/yolo --params-file /tmp/launch_params_9dnamt_c -r image_raw:=/my_camera/pylon_ros2_camera_node/image_raw'].

@mgonzs13 I want to run YOLO detection using a GPU, but after installing CUDA, it still fails to run. What exactly should I do next? Is there anything else I need to install?

mgonzs13, 2025-01-07:
@Phance, How have you installed CUDA? You have to install a nvidia driver compatible with your torch version, for example, `nvidia-driver-550` and `torch 2.4.0` in Ubuntu 22.

mgonzs13, 2025-01-07:
@Phance check the nvidia drive. You can use the `nvidia-smi` command.

closed 2025-02-21 by github-actions[bot]