mgonzs13/yolo_ros #76: Can't use ros2 launch yolo_bringup yolo.launch.py stuck when starting script. AGX Orin 64gb Developer kit

xXMrPotatoXx, 2025-02-24:
[image]

bhavya-goyal-22, 2025-02-26:
Here's a refined version of your response with better clarity and correctness:

"Please verify the input topic that the node is subscribing to by checking it in RViz or using ros2 topic echo. Also, ensure that the camera topic is correctly set and that data is being published on the topic.

For example, you can check the camera topic with:
`ros2 topic echo /image_raw`

Try launching your file using the following command:
`ros2 launch yolo_bringup yolo.launch.py input_image_topic:=<your_topic>
`

Johnny-Hu-406, 2025-03-02:
Hello, I am facing the same issue.

To simulate the camera input, I am publishing the image to `/camera/rgb/image_raw`, and I can preview it in Rviz2, where it displays correctly.

However, the `/yolo/dbg_image` node still does not produce any output.

[image]

Here is a screenshot of my terminal showing the following image
[image]

I hope to receive your reply and assistance, as I would appreciate your help in understanding this issue. Thank you very much for your time and support.

mgonzs13, 2025-03-02:
Check the QoS of the `/camera/rgb/image_raw` and also check if the detection topics are publishing. The QoS of the publisher and subscriber must be the same.

Johnny-Hu-406, 2025-03-02:
First, I use `ros2 topic info /camera/rgb/image_raw --verbose` to confirm that the QoS settings of the publisher and subscriber nodes appear to be consistent.

[image]

Additionally, I also use `ros2 docker --report` to check the relationship between the publisher node and the subscriber node.
[image]

I’m not sure what other directions I can try to solve this issue. Any suggestions? Thanks!

mgonzs13, 2025-03-02:
@Johnny-Hu-406, try `ros2 topic echo /yolo/detections` and `ros2 topic echo /yolo/tracking`  to check if detection and tracking are getting done.

Johnny-Hu-406, 2025-03-02:
@mgonzs13  When executing `ros2 topic echo /yolo/detections`, I can see the following output, and `ros2 topic echo /yolo/tracking` also has similar output.

[image]

mgonzs13, 2025-03-02:
That means you are getting detections. Next, check the topics of the `/yolo/debug_node`. Also, try `ros2 topic echo /yolo/dbg_image`.

Johnny-Hu-406, 2025-03-02:
First, I tried `ros2 topic echo /yolo/detections` and could see the output, such as coordinates and confidence

After that, I tried `ros2 topic echo /yolo/dbg_image`, but there was no output in the terminal.
Then, the terminal gets stuck on the following screen.
[image]

I believe that `/yolo/debug_node` successfully receives input, but it is unable to output images to the `/yolo/dbg_image` topic. Additionally, `ros2 topic echo /yolo/dbg_image` does not produce a response. I'm not sure how to resolve this issue.

mgonzs13, 2025-03-02:
So, if the problem is in the `/yolo/debug_node`, check the QoS of its topics, include the image and detection topics.

Johnny-Hu-406, 2025-03-02:
I use `ros2 topic info /camera/rgb/image_raw --verbose` to confirm that the QoS settings of the `/camera/rgb/image_raw` and `/yolo/debug_node` nodes appear to be consistent.

[image]

mgonzs13, 2025-03-02:
@Johnny-Hu-406 are you using a depth camera? If not or if you don't want 3D detection, use `use_3d:=False` in your ros2 launch command.

Johnny-Hu-406, 2025-03-03:
@mgonzs13 I'm not entirely sure what caused the issue, but after reinstalling your repository and Docker, the problem was successfully resolved.
Currently, the images from `/yolo/debug_node` are functioning properly.  

I sincerely appreciate your patience and prompt responses—they have been incredibly helpful to me!

closed 2025-04-17 by github-actions[bot]