mgonzs13/yolo_ros #77: With Depth, 3D detection not working

ArghyaChatterjee, 2025-03-09:
Hello,

I am using a zed camera. I have correctly remapped the topics and able to generate the demos correctly for 2D detections but not able to generate the 3D detection results. 

To be specific, I am using `ros2 launch yolo_bringup yolo.launch.py model:=yolov8m-seg.pt use_3d:=False` and this is working.

This command which should also be showing 3D detection `ros2 launch yolo_bringup yolo.launch.py model:=yolov8m-seg.pt` is executing but the node seems to hang and the depth and RGB image coming from zed seem to freeze (froze on rviz and also froze when I do topic echo) and I see no detection in 2D image let alone in the 3D one.

The QoS profile zed has for the depth are:
```
$ ros2 topic info -v /zed/zed_node/left/image_rect_color
Type: sensor_msgs/msg/Image

Publisher count: 1

Node name: zed_node
Node namespace: /zed
Topic type: sensor_msgs/msg/Image
Endpoint type: PUBLISHER
GID: 01.0f.98.94.c7.f4.ac.2e.00.00.00.00.00.00.49.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: rviz
Node namespace: /
Topic type: sensor_msgs/msg/Image
Endpoint type: SUBSCRIPTION
GID: 01.0f.98.94.91.7f.d2.b7.00.00.00.00.00.00.34.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

$ ros2 topic info -v /zed/zed_node/depth/depth_registered
Type: sensor_msgs/msg/Image

Publisher count: 1

Node name: zed_node
Node namespace: /zed
Topic type: sensor_msgs/msg/Image
Endpoint type: PUBLISHER
GID: 01.0f.98.94.c7.f4.ac.2e.00.00.00.00.00.00.79.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: rviz
Node namespace: /
Topic type: sensor_msgs/msg/Image
Endpoint type: SUBSCRIPTION
GID: 01.0f.98.94.91.7f.d2.b7.00.00.00.00.00.00.33.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```
This is same for RGB images as well.

The 2 depth publishing mode that zed has:

mono16:
```
$ ros2 topic echo /zed/zed_node/depth/depth_registered --once
header:
  stamp:
    sec: 1741493413
    nanosec: 735315610
  frame_id: zed_left_camera_optical_frame
height: 376
width: 672
encoding: mono16
is_bigendian: 0
step: 1344
data:
- 32
- 6
- 38
- 6
- 43
- 6
- 48
- 6
- 49
...
```

32FC1:
```
$ ros2 topic echo /zed/zed_node/depth/depth_registered --once
header:
  stamp:
    sec: 1741493657
    nanosec: 734350808
  frame_id: zed_left_camera_optical_frame
height: 376
width: 672
encoding: 32FC1
is_bigendian: 0
step: 2688
data:
- 255
- 255
- 255
- 127
..
```

I tried both but not working. Still have the same issue. Do you have any better idea about why this is happening ?

ArghyaChatterjee, 2025-04-11:
Here is my launch file:

yolo.launch.py:
```
# Copyright (C) 2023 Miguel Ángel González Santamarta

# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.

# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.

# You should have received a copy of the GNU General Public License
# along with this program.  If not, see <https://www.gnu.org/licenses/>.

from launch import LaunchDescription, LaunchContext
from launch.actions import DeclareLaunchArgument, OpaqueFunction
from launch.substitutions import LaunchConfiguration, PythonExpression
from launch_ros.actions import Node
from launch.conditions import IfCondition

def generate_launch_description():

    def run_yolo(context: LaunchContext, use_tracking, use_3d):

        use_tracking = eval(context.perform_substitution(use_tracking))
        use_3d = eval(context.perform_substitution(use_3d))

        model_type = LaunchConfiguration("model_type")
        model_type_cmd = DeclareLaunchArgument(
            "model_type",
            default_value="YOLO",
            choices=["YOLO", "World"],
            description="Model type form Ultralytics (YOLO, World)",
        )

        model = LaunchConfiguration("model")
        model_cmd = DeclareLaunchArgument(
            "model",
            default_value="yolov8m.pt",
            description="Model name or path",
        )

        tracker = LaunchConfiguration("tracker")
        tracker_cmd = DeclareLaunchArgument(
            "tracker",
            default_value="bytetrack.yaml",
            description="Tracker name or path",
        )

        device = LaunchConfiguration("device")
        device_cmd = DeclareLaunchArgument(
            "device",
            default_value="cuda:0",
            description="Device to use (GPU/CPU)",
        )

        enable = LaunchConfiguration("enable")
        enable_cmd = DeclareLaunchArgument(
            "enable",
            default_value="True",
            description="Whether to start YOLO enabled",
        )

        threshold = LaunchConfiguration("threshold")
        threshold_cmd = DeclareLaunchArgument(
[... 253 lines cut ...]
            debug_node_cmd,
        )

    use_tracking = LaunchConfiguration("use_tracking")
    use_tracking_cmd = DeclareLaunchArgument(
        "use_tracking",
        default_value="True",
        description="Whether to activate tracking",
    )

    use_3d = LaunchConfiguration("use_3d")
    use_3d_cmd = DeclareLaunchArgument(
        "use_3d",
        default_value="True",
        description="Whether to activate 3D detections",
    )

    return LaunchDescription(
        [
            use_tracking_cmd,
            use_3d_cmd,
            OpaqueFunction(function=run_yolo, args=[use_tracking, use_3d]),
        ]
    )
```

ArghyaChatterjee, 2025-04-11:
For me, this works only when I give the values like this:
```
ros2 launch yolo_bringup yolo.launch.py \
    model:=yolov8m-pose.pt \
    use_3d:=True \
    input_image_topic:=/zed/zed_node/left/image_rect_color \
    input_depth_topic:=/zed/zed_node/depth/depth_registered \
    input_depth_info_topic:=/zed/zed_node/depth/camera_info \
    target_frame:=zed_left_camera_frame
```

[image]

This looks good. However, a good upgrade will be to add those joint keypoints as lines to visualize an skeleton inside rviz. Here is an example:

[image]

Thanks.

mgonzs13, 2025-04-11:
@ArghyaChatterjee So, the launch is working for you, right?

> This looks good. However, a good upgrade will be to add those joint keypoints as lines to visualize an skeleton inside rviz. Here is an example:

You mean in the 3D version?

closed 2025-05-26 by github-actions[bot]