mgonzs13/yolo_ros #78: Is the YOLO-World model working ??

ArghyaChatterjee, 2025-03-09:
Hello,

I was able to run the 2d detections demos that you have working correctly but not able to run the YOLO-World Model. Here is my `yolo-world.launch.py` file:
```
# Copyright (C) 2023 Miguel Ángel González Santamarta

# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.

# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.

# You should have received a copy of the GNU General Public License
# along with this program.  If not, see <https://www.gnu.org/licenses/>.

import os
from launch import LaunchDescription
from launch.substitutions import LaunchConfiguration
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from ament_index_python.packages import get_package_share_directory

def generate_launch_description():

    return LaunchDescription(
        [
            IncludeLaunchDescription(
                PythonLaunchDescriptionSource(
                    os.path.join(
                        get_package_share_directory("yolo_bringup"),
                        "launch",
                        "yolo.launch.py",
                    )
                ),
                launch_arguments={
                    "model_type": "World",
                    "model": LaunchConfiguration("model", default="yolov8s-worldv2.pt"),
                    "tracker": LaunchConfiguration("tracker", default="bytetrack.yaml"),
                    "device": LaunchConfiguration("device", default="cuda:0"),
                    "enable": LaunchConfiguration("enable", default="True"),
                    "threshold": LaunchConfiguration("threshold", default="0.5"),
                    "input_image_topic": LaunchConfiguration(
                        "input_image_topic", default="/zed/zed_node/left/image_rect_color"
                    ),
                    "image_reliability": LaunchConfiguration(
                        "image_reliability", default="1"
                    ),
                    "namespace": LaunchConfiguration("namespace", default="yolo"),
                }.items(),
            )
        ]
    )

```
And here is how I am running it:
```
$ ros2 launch yolo_bringup yolo-world.launch.py use_3d:=False
[INFO] [launch]: All log files can be found below /home/arghya/.ros/log/2025-03-09-13-28-27-231539-arghya-Pulse-GL66-12UEK-2430261
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [yolo_node-1]: process started with pid [2430273]
[INFO] [tracking_node-2]: process started with pid [2430275]
[INFO] [debug_node-3]: process started with pid [2430277]
[tracking_node-2] [INFO] [1741544909.486636950] [yolo.tracking_node]: [tracking_node] Configuring...
[yolo_node-1] [INFO] [1741544909.522318873] [yolo.yolo_node]: [yolo_node] Configuring...
[debug_node-3] [INFO] [1741544909.531735059] [yolo.debug_node]: [debug_node] Configuring...
[tracking_node-2] [INFO] [1741544909.570578982] [yolo.tracking_node]: [tracking_node] Configured
[tracking_node-2] [INFO] [1741544909.570926452] [yolo.tracking_node]: [tracking_node] Activating...
[tracking_node-2] [INFO] [1741544909.573674398] [yolo.tracking_node]: [tracking_node] Activated
[yolo_node-1] [INFO] [1741544909.591999575] [yolo.yolo_node]: [yolo_node] Configured
[yolo_node-1] [INFO] [1741544909.592330109] [yolo.yolo_node]: [yolo_node] Activating...
[debug_node-3] [INFO] [1741544909.603144718] [yolo.debug_node]: [debug_node] Configured
[debug_node-3] [INFO] [1741544909.603481590] [yolo.debug_node]: [debug_node] Activating...
[debug_node-3] [INFO] [1741544909.606638987] [yolo.debug_node]: [debug_node] Activated
[yolo_node-1] [INFO] [1741544909.632047749] [yolo.yolo_node]: Trying to fuse model...
[yolo_node-1] [INFO] [1741544909.843707665] [yolo.yolo_node]: [yolo_node] Activated

```
The node seems to correctly launching but I don't see any output.

Thanks in Advance.

ArghyaChatterjee, 2025-03-10:
The way I got it to work is by following:
```
$ ros2 launch yolo_bringup yolo.launch.py model:=yolov8s-worldv2.pt use_3d:=False
```
Normally, when I launch it like this:
```
$ ros2 launch yolo_bringup yolo-world.launch.py use_3d:=False
```
It doesn't work.

I have another question. Aren't this YOLO world model supposed to be outputting walls, ceilings and floors as well ?? I don't see those classes when I start the detection node. Is there any particular reason for that ?

mgonzs13, 2025-03-10:
Hey @ArghyaChatterjee, what if you run `ros2 launch yolo_bringup yolo-world..launch.py model:=yolov8s-worldv2.pt use_3d:=False :="/zed/zed_node/left/image_rect_color" "image_reliability:="1"`?

closed 2025-04-24 by github-actions[bot]