mgonzs13/yolo_ros #79: In a custom dataset, certain categories cannot be detected.

Johnny-Hu-406, 2025-03-10:
Hello,

I am working on designing a strawberry-picking robot and currently using simulated images for testing. However, when running object tracking with YOLOv8/YOLOv12 in this ROS package, some category is not detected properly.

What I've Tried : 
1. I confirmed that my dataset is correct.
2. When running direct prediction using Python and YOLO on PC, the results are correct (e.g., Image 1).

[image]

However, when I import the trained weights into the ROS2 package and run inference, the performance drops significantly, sometimes even missing detections (e.g., Image 2).
Command Used
```
ros2 launch yolo_bringup yolov12.launch.py \
imgsz_height:=640 \
imgsz_width:=640 \
use_3d:=False \
input_image_topic:=/rgb \
model:=yolo12m_best.pt \
use_tracking:=False
```

[image]

What could be causing this issue? How can I troubleshoot the problem?

Any help would be greatly appreciated! Thanks!

Johnny-Hu-406, 2025-03-12:
@mgonzs13 
I tried the following command, but the detection performance is still not good:

```bash
ros2 launch yolo_bringup yolov12.launch.py \
  imgsz_height:=640 \
  imgsz_width:=640 \
  use_3d:=False \
  input_image_topic:=/rgb \
  model:=yolo12m_best.pt \
  image_reliability:=0 \
  use_tracking:=False \
  threshold:=0.2 \
  iou:=0.5
```

Johnny-Hu-406, 2025-03-19:
I use this command to launch YOLOv12 following the official YOLO example without ROS 2 and test my image.
```
from ultralytics import YOLO
import cv2
import torch
import matplotlib.pyplot as plt

model = YOLO("runs/detect/train/weights/best.pt") 

results = model("coco/test2017/", save=True, show=False) 
```

Also, I have recorded the topic "/rgb" as a rosbag, but when I play it back, the detection result is still not ideal.

mgonzs13, 2025-03-23:
Hey @Johnny-Hu-406, I've been testing your model and I've found out that the problem is the [conversion](https://github.com/mgonzs13/yolo_ros/blob/a99a101c720afe76a24170c5798eba7ac57f8c3d/yolo_ros/yolo_ros/yolo_node.py#L330) from BGR to RGB. When using BGR I get this result in rviz2:
[image]

Which encoding has you used to train your model, BGR? From this https://github.com/ultralytics/ultralytics/issues/9912 issue, I understand that YOLO models need RGB and if they receive a BGR image, it will be converted. I don't know if YOLOv12 works differently.

Johnny-Hu-406, 2025-04-06:
@mgonzs13 
Thank you for your response. I also believe that the issue lies in the image conversion, which has resulted in poor detection outcomes.

mgonzs13, 2025-04-07:
@Johnny-Hu-406 I've added a `yolo_encoding` param to choose the encoding. The default value is `bgr8`.

closed 2025-05-22 by github-actions[bot]