mgonzs13/yolo_ros #82: /yolo/dgb_image visualizing in Rviz2

changbumLee, 2025-03-15:
Hi. I have managed to run the code and can view it by subscribing to the yolo/dbg_image. But I wanted to ask how can I visualize the image in the Rviz world like what is shown in the demo.

I launched yolo_node with ros2 launch yolo_bringup yolo.launch.py.
I have the input_image_topic remapped to /camera/image_raw. 
The /yolo/detections streams fine, but /yolo/dbg_image shows a 'No Image' error with no detections.

[image]

[image]

[image]

[image]

mgonzs13, 2025-03-15:
Please @changbumLee, check other [similar issues](https://github.com/mgonzs13/yolo_ros/issues?q=Rviz2%20). You should check all topics, detections, tracking, detections_3d. And also check the QoS of your camera topics, and check the README for info about the params of the launch.

changbumLee, 2025-03-16:
There is one different topic among Qos

[image]

[image]

[image]

[image]

mgonzs13, 2025-03-16:
@changbumLee check also the QoS of your camera topics.

changbumLee, 2025-03-16:
this

[image]

mgonzs13, 2025-03-16:
@changbumLee you should check all camera topics. Then, check if the yolo topics are getting messages with echo or hz.

changbumLee, 2025-03-16:
all camera topics are this

[image]

[image]

and getting messages with echo or hz 

-> /yolo/tracking

[image]

[image]

-> /yolo/detections

[image]

[image]

/yolo/dbg_image doesn't show anything, neither echo nor hz

[image]

mgonzs13, 2025-03-16:
Are you using a depth camera? You can try with use_3d set to False.

dyldonahue, 2025-03-26:
I also seem to be having this issue. Some details for my setup

- using a realsense, and have explicitly set use_3d to false
- using default model param
- have tried with device=cuda:0, device=cpu, and without editing this parameter

the yolo/detections topic publishes labels as expected, but the debug image topic does not publish anything, and thus shows NO_IMAGE in rviz when trying to visualize this topic

mgonzs13, 2025-03-27:
@dyldonahue, which QoS have the nodes (camera, yolo, debug)? Which command are you using to launch yolo?

ErikParkerrr, 2025-05-05:
Also having this issue

mgonzs13, 2025-05-07:
So, @ErikParkerrr, have you checked all topics, subscriptions and publications, and the QoS?

closed 2025-06-22 by github-actions[bot]

mgonzs13, 2025-10-06:
Hi @dorroddorrod, have you checked that all topics are set correctly for all nores, including the QoS?