mgonzs13/yolo_ros #83: README use_3d is Wrong

robohase, 2025-03-17:
Hi Miguel. I love your yolo project for ros2.
there was a little trouble for using this project

The README states that the use_3d parameter defaults to False, but in yolo.launch.py it's set to True. I think you should change parameter or fix README

nxnen8282, 2025-03-17:
Our system attempted to address this issue:

Fix inconsistency between README and launch file where use_3d parameter had different default values. Changed launch file to match README's default value of False.

You can examine the suggested modifications here: https://github.com/nxnen8282/mgonzs13-yolo_ros-83/commit/924b1d806d3890be50aa0da0c131142454f29fe3.

> [!CAUTION]
> Advisory: This solution was generated using automated tools, and we strongly recommend reviewing the code thoroughly before implementation. The solution may be partial - please consider it as inspiration for your own implementation.

---

http://missionfixinternet.com

mgonzs13, 2025-03-17:
@robohase sorry for the inconsistency. It is fixed in 9296b6eafec9bc1fcb1d3e8836aca0092269b0a3

closed 2025-04-05 by robohase