mgonzs13/yolo_ros #85: use_3d:=True  is giving error : Could not transform: "base_link.

hemsinghb, 2025-03-31:
Hi, I'm getting this problem. continuously...on the execution of 

"$  ros2 launch yolo_bringup yolo.launch.py use_3d:=True"

"[detect_3d_node-3] [ERROR] [1743398721.291918889] [yolo.detect_3d_node]: Could not transform: "base_link" passed to lookupTransform argument target_frame does not exist."

however    without "use_3d":=true " ]   is working fine.   i.e. 
"$  ros2 launch yolo_bringup yolo.launch.py " 
 

my system is ros2 humble on orin jetpack6.2. I use realsense camera. D435. Please guide further. 
 Thank you,
 hems.

mgonzs13, 2025-03-31:
@hemsinghb you need to set the `target_frame` to your base frame. The default value is`base_link`, and you are getting the error due to the fact that you don't have a frame with that name.

hemsinghb, 2025-04-01:
Thank you. It's sorted out by running another script that has 'base_link'.