mgonzs13/yolo_ros #87: Visual inspection

Phance, 2025-04-17:
Using some official pre-trained models for training, the obtained ".pt" model can be loaded for detection, but when I was training the model, I added the attention mechanism for small innovation, and the obtained ".pt" model could not be loaded, and there was no detection image in Rviz2, and "No Image" was displayed, what is the reason for this?

mgonzs13, 2025-04-17:
@Phance have you tried your new model with bare Ultralytics? Your new model architecture with attention may not be supported.

Phance, 2025-04-17:
@mgonzs13 What does it currently support? Also, when I run the previous model, there will be stuttering, discontinuity, and the real-time performance is not very good, do you know how to solve this problem? Maybe it's because I'm running on a dev board and only have the CPU?

closed 2025-06-01 by github-actions[bot]