mgonzs13/yolo_ros #88: Custom Model stuck at tracking node when launching.

ZGregorio, 2025-04-22:
So I'm trying to run my custom model that I trained using YOLOv5.

I use the following code:
`ros2 launch yolo_bringup yolov5.launch.py 
use_3d:=True 
model:=/home/ws/src/yolo_ros/yolo_bringup/exp21/exp21/weights/best.pt 
device:=CPU`

and it gets stuck like so:

`[INFO] [launch]: All log files can be found below /home/zejj/.ros/log/2025-04-22-13-57-47-646822-raspberrypi-660718
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [yolo_node-1]: process started with pid [660721]
[INFO] [tracking_node-2]: process started with pid [660722]
[INFO] [detect_3d_node-3]: process started with pid [660723]
[INFO] [debug_node-4]: process started with pid [660724]
[detect_3d_node-3] [INFO] [1745330269.801735943] [yolo.detect_3d_node]: [detect_3d_node] Configuring...
[detect_3d_node-3] [INFO] [1745330269.827801997] [yolo.detect_3d_node]: [detect_3d_node] Configured
[detect_3d_node-3] [INFO] [1745330269.840849357] [yolo.detect_3d_node]: [detect_3d_node] Activating...
[detect_3d_node-3] [INFO] [1745330269.852016682] [yolo.detect_3d_node]: [detect_3d_node] Activated
[yolo_node-1] [INFO] [1745330274.368321266] [yolo.yolo_node]: [yolo_node] Configuring...
[yolo_node-1] [INFO] [1745330274.588651506] [yolo.yolo_node]: [yolo_node] Configured
[yolo_node-1] [INFO] [1745330274.590181987] [yolo.yolo_node]: [yolo_node] Activating...
[debug_node-4] [INFO] [1745330274.823910495] [yolo.debug_node]: [debug_node] Configuring...
[debug_node-4] [INFO] [1745330275.012628167] [yolo.debug_node]: [debug_node] Configured
[debug_node-4] [INFO] [1745330275.013351556] [yolo.debug_node]: [debug_node] Activating...
[debug_node-4] [INFO] [1745330275.021403346] [yolo.debug_node]: [debug_node] Activated
[tracking_node-2] [INFO] [1745330275.378635761] [yolo.tracking_node]: [tracking_node] Configuring...
[tracking_node-2] [INFO] [1745330275.510896957] [yolo.tracking_node]: [tracking_node] Configured
[tracking_node-2] [INFO] [1745330275.511509234] [yolo.tracking_node]: [tracking_node] Activating...
[tracking_node-2] [INFO] [1745330275.516256231] [yolo.tracking_node]: [tracking_node] Activated`

what am I doing wrong? sorry if its an obvious fix

mgonzs13, 2025-04-23:
The yolo is not activating. Can you try to load your model with Ultralytics?

```python
from ultralytics import YOLO
yolo = YOLO(/home/ws/src/yolo_ros/yolo_bringup/exp21/exp21/weights/best.pt)
```

ZGregorio, 2025-04-23:
I tried modifying yolov5.launch.py in the launch folder of my yolo_bringup directory.

I added the yolo as instructed which can be seen as follows:

from ultralytics import YOLO

`yolo = YOLO("/home/ws/src/yolo_ros/yolo_bringup/weights/best.pt")`
`def generate_launch_description():`

    return LaunchDescription(
        [
            IncludeLaunchDescription(
                PythonLaunchDescriptionSource(
                    os.path.join(
                        get_package_share_directory("yolo_bringup"),
                        "launch",
                        "yolo.launch.py",
                    )
                ),
                launch_arguments={
                    "model": LaunchConfiguration("model", default=yolo),
                    "tracker": LaunchConfiguration("tracker", default="bytetrack.yaml"),
                    "device": LaunchConfiguration("device", default="cuda:0"),
                    "enable": LaunchConfiguration("enable", default="True"),
                    "threshold": LaunchConfiguration("threshold", default="0.5"),
                    "input_image_topic": LaunchConfiguration(
                        "input_image_topic", default="/camera/rgb/image_raw"
                    ),
                    "image_reliability": LaunchConfiguration(
                        "image_reliability", default="1"
                    ),
                    "namespace": LaunchConfiguration("namespace", default="yolo"),
                }.items(),
            )
        ]
    )`

with this code, the default model is now my own custom model.

now after i build this package and tried launch the node, i obtained an error saying that our model is being launched as a yolov8 and it is not compatible with our weight (YOLOv5).

the error i obtained are as follow:

zejj@raspberrypi:/home/ws $ ros2 launch yolo_bringup yolov5.launch.py input_image_topic:=/depth_camera/image_raw device:=cpu
[INFO] [launch]: All log files can be found below /home/zejj/.ros/log/2025-04-23-13-47-43-486819-raspberrypi-63793
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [py]:
 - TypeError: ERROR ❌️ /home/ws/src/yolo_ros/yolo_bringup/weights/best.pt appears to be an Ultralytics YOLOv5 model originally trained with https://github.com/ultralytics/yolov5.
This model is NOT forwards compatible with YOLOv8 at https://github.com/ultralytics/ultralytics.
Recommend fixes are to train a new model using the latest 'ultralytics' package or to run a command with an official Ultralytics model, i.e. 'yolo predict model=yolo11n.pt'
 - InvalidFrontendLaunchFileError: The launch file may have a syntax error, or its format is unknown

now i tried using the following code without using from ultralytics import YOLO, which loads YOLOv8 by default.

`yolo = "/home/ws/src/yolo_ros/yolo_bringup/weights/best.pt"`

but i can't seem to get past 

[tracking_node-2] [INFO] [1745416724.324858114] [yolo.tracking_node]: [tracking_node] Configuring...
[tracking_node-2] [INFO] [1745416724.550668481] [yolo.tracking_node]: [tracking_node] Configured
[tracking_node-2] [INFO] [1745416724.551369184] [yolo.tracking_node]: [tracking_node] Activating...
[tracking_node-2] [INFO] [1745416724.557937179] [yolo.tracking_node]: [tracking_node] Activated

i tried tracing this problem.
i cant reach the logger that says "Trying to fuse model ..."

the following code is a snipper from the yolo_node.py which is inside yolo_bringup/yolo_ros/

`
def on_activate(self, state: LifecycleState) -> TransitionCallbackReturn: 
        self.get_logger().info(f"[{self.get_name()}] Activating...")

        try:
            self.yolo = self.type_to_model[self.model_type](self.model)
        except FileNotFoundError:
            self.get_logger().error(f"Model file '{self.model}' does not exists")
            return TransitionCallbackReturn.ERROR

        try:
            self.get_logger().info("Trying to fuse model...")
            self.yolo.fuse()
        except TypeError as e:
            self.get_logger().warn(f"Error while fuse: {e}")

        self._enable_srv = self.create_service(SetBool, "enable", self.enable_cb)

        if isinstance(self.yolo, YOLOWorld):
            self._set_classes_srv = self.create_service(
                SetClasses, "set_classes", self.set_classes_cb
            )

        self._sub = self.create_subscription(
            Image, "image_raw", self.image_cb, self.image_qos_profile
        )
`

how might i be able to solve this problem?

ZGregorio, 2025-04-23:
> I tried modifying yolov5.launch.py in the launch folder of my yolo_bringup directory.
>
> I added the yolo as instructed which can be seen as follows:
>
> from ultralytics import YOLO
>
> `yolo = [...]

i accidentally pressed close issue can you reopen it? T_T
im so sorry

ZGregorio, 2025-04-23:
I ran the code and it produced this error:

TypeError: ERROR ❌️ /home/zejj/strawberry-picking-rpi/src/yolo_ros/yolo_bringup/exp21/exp21/weights/best.pt appears to be an Ultralytics YOLOv5 model originally trained with https://github.com/ultralytics/yolov5.
This model is NOT forwards compatible with YOLOv8 at https://github.com/ultralytics/ultralytics.
Recommend fixes are to train a new model using the latest 'ultralytics' package or to run a command with an official Ultralytics model, i.e. 'yolo predict model=yolo11n.pt'

Im assuming once I retrain my custom model it should work? Thank you btw :)

closed 2025-06-07 by github-actions[bot]