mgonzs13/yolo_ros #90: 实时性

Phance, 2025-05-02:
@mgonzs13 Hello, I ran my self-trained model and ran the instructions as follows:
ros2 launch yolo_bringup yolo.launch.py input_image_topic:=/my_camera/pylon_ros2_camera_node/image_raw use_debug:=True device:=cpu
However, my detection speed is very slow, and sometimes there is no feedback from the detection, so please ask if this is because of the camera or because there is only the CPU. My camera is a Basler industrial camera, and the whole system runs on an AlxBoard board, which is developed by Intel and only has a CPU.

mgonzs13, 2025-05-02:
@Phance, CPU is the most likely problem here. What is the rate of Yolo detection?

Phance, 2025-05-03:
@mgonzs13 

[image]|
This is the rate at which I run the detection effect, is there a recommended board if I want to use a GPU? How did you develop the assay? Using Nvidia Jeston's board?

Phance, 2025-05-04:
@mgonzs13 Which camera is the best choice for ROS2?

mgonzs13, 2025-05-04:
I don't know which could be the best camera for ROS 2, I have used webcams, xtions, relasense, zed and more. Using CUDA will speed up the inference, but I don't know the impact of using a Jetson.

closed 2025-06-18 by github-actions[bot]