mgonzs13/yolo_ros #96: About rclpy.lifecycle module

Shua1ge, 2025-07-09:
My ROS2 Galactic Env donot have rclpy.lifecycle this module,could you teach me how to introoduce this module to my ROS2 Env,pls

Shua1ge, 2025-07-10:
I used the four pieces of code provided in  https://github.com/ros2/rclpy/blob/humble/rclpy/rclpy/lifecycle/node.py ,I placed them in:
/opt/ros/galactic/lib/python3.8/site-packages/rclpy/lifecycle and re colcon build and run yolo,a new problem has emerged,Could U please advice me on how to solve，thanks a million:

[debug_node-3] Traceback (most recent call last):
[debug_node-3]   File "/home/t1an/yolo_ros2/install/yolo_ros/lib/yolo_ros/debug_node", line 11, in <module>
[debug_node-3]     load_entry_point('yolo-ros==0.0.0', 'console_scripts', 'debug_node')()
[debug_node-3]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 490, in load_entry_point
[debug_node-3]     return get_distribution(dist).load_entry_point(group, name)
[debug_node-3]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
[debug_node-3]     return ep.load()
[debug_node-3]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2445, in load
[debug_node-3]     return self.resolve()
[debug_node-3]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2451, in resolve
[debug_node-3]     module = __import__(self.module_name, fromlist=['__name__'], level=0)
[debug_node-3]   File "/home/t1an/yolo_ros2/install/yolo_ros/lib/python3.8/site-packages/yolo_ros/debug_node.py", line 28, in <module>
[debug_node-3]     from rclpy.lifecycle import LifecycleNode
[debug_node-3]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/lifecycle/__init__.py", line 15, in <module>
[debug_node-3]     from .managed_entity import ManagedEntity
[debug_node-3]   File "/opt/ros/galactic/lib/python3.8/site-packages/rclpy/lifecycle/managed_entity.py", line 21, in <module>
[debug_node-3]     TransitionCallbackReturn = _rclpy.TransitionCallbackReturnType
[debug_node-3] AttributeError: module 'rclpy._rclpy_pybind11' has no attribute 'TransitionCallbackReturnType'
[ERROR] [debug_node-3]: process has died [pid 11107, exit code 1, cmd '/home/t1an/yolo_ros2/install/yolo_ros/lib/yolo_ros/debug_node --ros-args -r __node:=debug_node -r __ns:=/yolo --params-file /tmp/launch_params_tn5u75oy -r image_raw:=/camera/color/image_raw -r detections:=tracking'].
[ERROR] [yolo_node-1]: process has died [pid 11103, exit code 1, cmd '/home/t1an/yolo_ros2/install/yolo_ros/lib/yolo_ros/yolo_node --ros-args -r __node:=yolo_node -r __ns:=/yolo --params-file /tmp/launch_params_icwi06v2 -r image_raw:=/camera/color/image_raw'].

mgonzs13, 2025-07-17:
@Shua1ge It seems there are more incompatibilities. What about using a previous version of this repo? Or a newer ROS 2 distro?

Shua1ge, 2025-07-19:
I havent had the experience of using Docker in Nvidia Xavier NX yet,I will first find out the feasibility of making Docker with ros2 humble on it.Thank U for giving me such a wonderful suggestion! ,i will make some attempts